Koenkk/zigbee2mqtt #3079: expose device capabilities

rmalchow, 2020-03-07:
hi,

i am playing with zigbee2mqtt for a bit. i have one thing that i really miss, which is to expose the channels and events of a device in a slightly more generic way. i can see now, with "homeasistant" on, there already is some information, but is very much tailored to the needs of homeassistant. 

what i am playing with is to deal with all of those on a lower level, so instead of dealing with a "light" that has "color temperature" and "brightness", i would like to treat it as a bunch of values i can set, increment or decrement within a given range. so, all logic would start from very basic components, e.g. fixed value, timer, debouncer, cron expression ... and then, for displaying things, also different blocks for combining values. 

although it seems that the herdsman converters do actually have (almost) all data necessary for this, i am not even entirely sure if this kind of approach  is possible in all cases. the data structure for this could look something like this:

```
    "abcdefh112233zzz" : {
		"device_info" : {},
		"channels" : [
	    	{
	    		"brightness" : {
	    			"location": "zigbee2mqtt/abcdefh112233xxx:brightness",
		    		"mode" : "rw",
		    		"family" : "light/brightness",
					"type" : "range",
					"min" : 0,
					"max" : 255
				}
			},
			{
	    		"motion_detected" : {
	    			"location": "zigbee2mqtt/abcdefh112233xxx:click",
		    		"mode" : "r",
		    		"family" : "sensor/motion",
					"type" : "event",
					"values" : [ "on", "off" ]
				}
	    	},
			{
	    		"temperature" : {
	    			"location": "zigbee2mqtt/abcdefh112233xxx:temp",
		    		"mode" : "r",
		    		"family" : "sensor/temperature",
					"type" : "range",
					"min" : 0,
					"max" : 40,
					"unit" : "c"
				}
	    	}
		]
	}
```

and configuration on top of this might look something like this:

[image]

sjorge, 2020-03-07:
Might be interesting to query these via `zigbee2mqtt/bridge/capability` and send it a friendly_name or device_id and then have it publish the result. I think zigbee2mqttassistant would probably also love that.

Koenkk, 2020-03-07:
While I like the idea, I have 2 problems with it:
- How usable will this be? (for how many people?)
- It adds a lot of extra maintenance

sjorge, 2020-03-07:
Can't we figure this out based on the endpoints in the db?

rmalchow, 2020-03-08:
hi,
i dont think this is immediately useful. but, like @sjorge said, this is what i was wondering: what data is actually there. i had a look at some of the config data, and it seems a lot of things are there anyways ... one thing i am kind of missing is ranges (min/max), with some of the data hiding within functions.

what i am experimenting with is to put mqtt more at the center. e.g. i have a small piece that links my nuki doorlocks to mqtt. that's very easy to do. i have another piece that listens for mqtt messages and triggers scripts based on that. these scripts in turn send mqtt messages. i have another piece, which is basically a number pad, and sends mqtt messages as well (this isnt actually hardware, it's just a simple web page, with a small backend that translates into mqtt. so .. if i enter the correct pin, it sends a message `"foo/pinpad_1" -> { "pin" : "123456" }` and then a script listens to these messages, checks the pin, then sends another message `"locks/nuki_01" -> { "action" : "unlatch" }`.

state is held in retained messages, and is available to the script as a big map, so it can, for example, check the values in `system/clock/currentTime` to see what time it is. 

i believe that normalizing on a lower level could actually work (or, rather ... i am trying to figure out if thiis is the case), and it would decouple things quite nicely - and one could still "denormalize" stuff back in a UI layer. 

but as of now, these are my own experiments, and i don't expect anyone to put a lot of effort into this ... as mentioned above, i was just wondering how much of that information is available, and how easy it would be to expose it (whatever actually DOES exist) in, for example zigbee2mqtt/config 

.rm

closed 2020-05-15 by stale[bot]