Koenkk/zigbee2mqtt #3081: aqara warer sensor not reporting status

kovaga, 2020-03-07:
# Bug Report
## What happened
value_json.water_leak is not reported anymore.

## What did you expect to happen
previously ""JSONPATH:$.water_leak"" true or false was reported.

## How to reproduce it (minimal and precise)
run zigbee2mqtt

## Debug Info
zigbee2mqtt version 1.11.0 (commit #31e5678)
CC26X2R1 Coordinator firmware version: '{"type":"zStack3x0","meta":{"transportrev":2,"product":1,"majorrel":2,"minorrel":7,"maintrel":1,"revision":20191106}}'

Have upgraded from cc2530 to cc26x2r1 v3.x and noticed that all of my aqara water leak sensors are not reporting water_leak status anymore.

info  2020-03-07 15:01:41: MQTT publish: topic 'zigbee2mqtt/sens_water3', payload '{"battery":100,"voltage":3005,"linkquality":81,"last_seen":1583589701082}'
info  2020-03-07 15:03:43: MQTT publish: topic 'zigbee2mqtt/sens_water4', payload '{"battery":100,"voltage":3025,"linkquality":78,"last_seen":1583589823124}'
info  2020-03-07 15:43:04: MQTT publish: topic 'zigbee2mqtt/sens_water2', payload '{"battery":100,"voltage":3005,"linkquality":75,"last_seen":1583592184513}'
info  2020-03-07 15:46:32: MQTT publish: topic 'zigbee2mqtt/sens_water1', payload '{"battery":100,"voltage":3025,"linkquality":75,"last_seen":1583592392360}'

having looked through the openhab logs, i noticed that some other fields are missing as well at other sensors.

i.e.
smoke sensor lacks sensitivity - Invalid path '$.sensitivity' in '{"battery":100,"voltage":3075,"smoke_density":0,"linkquality":75,"last_seen":1583589585806,"smoke":false}'

Is there a way to get back the reporting of those default values?

Koenkk, 2020-03-08:
Did you already try putting them in water (to trigger them once)?

kovaga, 2020-03-09:
Oh yeah, it did appear and is now reporting, once they were triggered.
I think for consistency reasons it would make sense to report the "false" flag even if the sensor hasnt been triggered.

closed 2020-03-09 by kovaga