Koenkk/zigbee2mqtt #3084: Removing a device (normal or force) does not remove retained state.json

sjorge, 2020-03-07:
#  Bug Report
## What happened
I removed a device, the retained state of the device was not removed from `state.json`.

## What did you expect to happen
On device removal the state in `state.json` should also be removed.

## How to reproduce it (minimal and precise)
- pair a device
- set retain to true for the device
- wait for the device to have an entry in `state.json`
- remove the device
- check `state.json` for device

## Debug Info
zigbee2mqtt version: dev
CC253X firmware version: n/a

Koenkk, 2020-03-07:
I think it is removed from the state, but the `state.json` is not updated immediately: every 5 minutes or when stopping Zigbee2mqtt (which I think is good behaviour).

sjorge, 2020-03-07:
Hmm, I had a lot info left in state.json for devices that i had removed days ago.

sjorge, 2020-03-07:
```
{
    "20861": {
        "color": {
            "x": 0.458,
            "y": 0.41
        },
        "brightness": 254,
        "state": "ON",
        "color_temp": 120
    },
    "20862": {
        "state": "ON",
        "brightness": 255,
        "color": {
            "x": 0.458,
            "y": 0.41
        },
        "color_temp": 366
    },
    "0x0017880104259333": {
        "state": "ON",
        "linkquality": 31,
        "brightness": 255,
        "color": {
            "x": 0.458,
            "y": 0.41
        },
        "color_temp": 366
    },
    "0x90fd9ffffee77fcf": {
        "color": {
            "x": 0.4597,
            "y": 0.4085
        },
        "linkquality": 57,
        "brightness": 254,
        "state": "OFF",
        "color_temp": 370,
        "update_available": false
    },
    "0x000d6ffffe8e8d4f": {
        "linkquality": 31,
        "update_available": false
    },
    "0x00158d0001ffaffc": {
        "linkquality": 34,
        "occupied_heating_setpoint": 22,
        "local_temperature": 21,
        "unoccupied_heating_setpoint": 16,
        "pi_heating_demand": 33,
        "eurotronic_error_status": 0,
        "current_heating_setpoint": 22,
        "eurotronic_host_flags": {
            "mirror_display": false,
            "boost": false,
            "window_open": false,
            "child_protection": true
        },
        "system_mode": "auto",
        "eurotronic_system_mode": 129,
        "battery": 65,
        "local_temperature_calibration": -0.6
    },
    "0x00158d0004148f89": {
        "illuminance": 0,
        "linkquality": 31,
        "occupancy": false,
        "battery": 100,
        "voltage": 3015,
[... 53 lines cut ...]
        "local_temperature": 21.5,
        "pi_heating_demand": 53,
        "boost": "Down",
        "occupied_heating_setpoint": 23.5
    }
}
```

```
{"id":1,"type":"Coordinator","ieeeAddr":"0x00124b001938a7e5","nwkAddr":0,"manufId":0,"epList":[1,2,3,4,5,6,8,11,12,110],"endpoints":{"1":{"profId":260,"epId":1,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[]},"2":{"profId":257,"epId":2,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[]},"3":{"profId":261,"epId":3,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[]},"4":{"profId":263,"epId":4,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[]},"5":{"profId":264,"epId":5,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[]},"6":{"profId":265,"epId":6,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[]},"8":{"profId":260,"epId":8,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[]},"11":{"profId":260,"epId":11,"devId":1024,"inClusterList":[],"outClusterList":[1280],"clusters":{},"binds":[]},"12":{"profId":49246,"epId":12,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[]},"110":{"profId":260,"epId":110,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[]}},"interviewCompleted":false,"meta":{}}
{"id":2,"type":"Router","ieeeAddr":"0x000d6ffffe8e8d4f","nwkAddr":47816,"manufId":4476,"manufName":"IKEA of Sweden","powerSource":"Mains (single phase)","modelId":"TRADFRI signal repeater","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":8,"inClusterList":[0,3,9,2821,4096,64636],"outClusterList":[25,32,4096],"clusters":{"genBasic":{"attributes":{"modelId":"TRADFRI signal repeater","manufacturerName":"IKEA of Sweden","powerSource":1,"zclVersion":3,"appVersion":33,"stackVersion":98,"hwVersion":1,"dateCode":"20190318","swBuildId":"2.2.005"}}},"binds":[{"cluster":0,"type":"endpoint","deviceIeeeAddress":"0x00124b001938a7e5","endpointID":1}]},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[33],"outClusterList":[33],"clusters":{},"binds":[]}},"appVersion":32,"stackVersion":98,"hwVersion":1,"dateCode":"20190318","swBuildId":"2.2.005","zclVersion":3,"interviewCompleted":true,"meta":{"reporting":1,"configured":2},"lastSeen":1583527646173}
{"id":3,"type":"Router","ieeeAddr":"0x000d6ffffe197fe9","nwkAddr":36186,"manufId":4476,"manufName":"IKEA of Sweden","powerSource":"Mains (single phase)","modelId":"TRADFRI bulb E27 CWS opal 600lm","epList":[1],"endpoints":{"1":{"profId":49246,"epId":1,"devId":512,"inClusterList":[0,3,4,5,6,8,768,2821,4096],"outClusterList":[5,25,32,4096],"clusters":{"genBasic":{"attributes":{"modelId":"TRADFRI bulb E27 CWS opal 600lm","manufacturerName":"IKEA of Sweden","powerSource":1,"zclVersion":1,"appVersion":17,"stackVersion":87,"hwVersion":1,"dateCode":"20180410","swBuildId":"1.3.009"}},"genLevelCtrl":{"attributes":{"currentLevel":254}},"genOnOff":{"attributes":{"onOff":1}},"lightingColorCtrl":{"attributes":{"currentX":30015,"currentY":26869,"colorMode":1}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b001938a7e5","endpointID":1},{"cluster":8,"type":"endpoint","deviceIeeeAddress":"0x00124b001938a7e5","endpointID":1},{"cluster":768,"type":"endpoint","deviceIeeeAddress":"0x00124b001938a7e5","endpointID":1}]}},"appVersion":17,"stackVersion":87,"hwVersion":1,"dateCode":"20180410","swBuildId":"1.3.009","zclVersion":1,"interviewCompleted":true,"meta":{"reporting":1},"lastSeen":1583527614579}
{"id":4,"type":"Router","ieeeAddr":"0x0017880104259333","nwkAddr":27321,"manufId":4107,"manufName":"Philips","powerSource":"Mains (single phase)","modelId":"LCT012","epList":[11,242],"endpoints":{"11":{"profId":49246,"epId":11,"devId":528," [... cut ...] "genPowerCfg":{"attributes":{"batteryVoltage":31}},"msIlluminanceMeasurement":{"attributes":{"measuredValue":4772}}},"binds":[{"cluster":1,"type":"endpoint","deviceIeeeAddress":"0x00124b001938a7e5","endpointID":1},{"cluster":1024,"type":"endpoint","deviceIeeeAddress":"0x00124b001938a7e5","endpointID":1}]}},"appVersion":21,"stackVersion":2,"hwVersion":1,"dateCode":"20191118","swBuildId":"2019\u0000www.","zclVersion":3,"interviewCompleted":true,"meta":{"configured":1},"lastSeen":1583527280152}
```

sjorge, 2020-03-07:
`0x086bd7fffe7ecd67`: is the wiser trv I was playing with that still doesn't work due to the undefined error, removed it again a few days ago.
`0x90fd9ffffee77fcf`: one of the tradfri bulbs I was swapping out with due to the ping failures

And 2 more that I had already removed manually while zigbee2mqtt was stopped.

Koenkk, 2020-03-09:
I cannot reproduce this: https://pastebin.com/xCkgcEkU

sjorge, 2020-03-09:
Lets close it for now and I will reopen if I hit the issue again.

closed 2020-03-09 by sjorge