Koenkk/zigbee2mqtt #3086: New auto discovery message after rename device

svrooij, 2020-03-08:
# Bug Report

This isn't a bug report but I think it's a feature request.

## What happened

1. Device renamed with publish to `zigbee2mqtt/bridge/config/rename_last`
2. No new auto discovery message

## What did you expect to happen

A new auto discovery message, since the topics for this device changed, there needs to come an updated home assistant auto discovery message. So all apps using the auto discovery know the new topics.

## How to reproduce it (minimal and precise)

1. Subscribe to `homeassistant/#`
2. Add new device (for `rename_last` to work).
2. Rename a device by sending `new_name` to `zigbee2mqtt/bridge/config/rename_last`

## Debug Info
zigbee2mqtt version: 1.11.0
CC253X firmware version: latest

sjorge, 2020-03-08:
This is fixed in the latest dev version I think:
https://github.com/Koenkk/zigbee2mqtt/commit/67b4bcf81550a2c6e87f8fe46d48ba913294738d

svrooij, 2020-03-08:
Any idea when this will be in production?

I was testing my new diyhue integration and noticed this.

sjorge, 2020-03-08:
@svrooij not sure, it depends on when @Koenkk does another release  Iguess.

closed 2020-03-08 by svrooij

sjorge, 2020-03-08:
Not sure if you are using docker or not... but:
https://hub.docker.com/layers/koenkk/zigbee2mqtt/latest-dev/images/sha256-7c1219cc27d50e78c49b4813cf407727f46dd378433ed1c0a2ea495b4f8d1ad9?context=explore

Should have the fix, or if you are running it manually installed you can just switch to the dev branch and run npm ci again before restarting.

svrooij, 2020-03-08:
I’ll try to change to dev tomorrow, thank for the help!!