Koenkk/zigbee2mqtt #3088: I want to add my new device, Tuya curtain motor(TS0601)

cchalong, 2020-03-08:
# Bug Report
## What happened
Recently I bought tuya zigbee curtain motor(see below link)
https://ko.aliexpress.com/item/4000463256008.html?spm=a2g12.12010612.8148356.5.35a5312b29uwEn

I'm trying to connect it to z2m, but not easy for me.
Could you help me?

I succeeded in paring, and triggered some actions and then I add the device to devices.js

when I triggered some action, the log is as follows.

1. Received Zigbee message from '0xec1bbdfffec3d049', type 'attributeReport', cluster 'genBasic', data '{"65506":21,"appVersion":82}' from endpoint 1 with groupID 0

2. Received Zigbee message from '0xec1bbdfffec3d049', type 'raw', cluster 'manuSpecificTuyaDimmer', data '{"type":"Buffer","data":[9,9,2,0,3,7,4,0,1,1]}' from endpoint 1 with groupID 0

3. Received Zigbee message from '0xec1bbdfffec3d049', type 'read', cluster 'genTime', data '["localTime"]' from endpoint 1 with groupID 0

and when motor gets started spin, i got the log(#2), and when I make motor spin in a reverse direction, i got the data : [9,9,2,0,3,7,4,0,1,0]. I guess that the data shows the function of "open" or "close"

unfortunately, i don't have remote controller(RF), so I couldn't test "stop" or "position" function.

and the device that I added to devices.js is as below.

 {                                                                                                                                           
        zigbeeModel: ['TS0601'],                                                                                                           
        model: 'YS-MT750',                                                                                           
        vendor: 'TuYa',                                                                                                                         
        description: 'Tuya curtain motor',                                                                        
        supports: 'open, close, stop, position',                                                                                           
        fromZigbee: [fz.tuya_switch, fz.ZNCLDJ11LM_ZNCLDJ12LM_curtain_options_output, fz.ignore_basic_report, fz.ignore_time_read, fz.cover_position_tilt, fz.ZNCLDJ11LM_ZNCLDJ12LM_curtain_analog_output],
        toZigbee: [],                                                                                                                      
    },          

Actually, I don't have the ability to new converter to fromZigbee.js, so I just use the existing ones. I searched for the converters which match with cluster and type shown on my log and I added some converters from Xiaomi curtain motor.

Currently, the message of "No converter available" disappeared. but I still can't add this device in my HA.

Your help and feedback would be much appreciated.

## What did you expect to happen

## How to reproduce it (minimal and precise)

## Debug Info
zigbee2mqtt version:
CC253X firmware version:

cchalong, 2020-03-08:
And I try to get more log. I'm not sure what data means(why it changes)

zigbee2mqtt:debug 2020-03-09 00:39:37: Received Zigbee message from '0xec1bbdfffec3d049', type 'raw', cluster 'manuSpecificTuyaDimmer', data '{"type":"Buffer","data":[9,4,2,0,3,7,4,0,1,0]}' from endpoint 1 with groupID 0
zigbee2mqtt:info  2020-03-09 00:39:37: MQTT publish: topic 'zigbee2mqtt/0xec1bbdfffec3d049', payload '{"undefined":"OFF","linkquality":63}'
zigbee2mqtt:debug 2020-03-09 00:40:25: Received Zigbee message from '0xec1bbdfffec3d049', type 'raw', cluster 'manuSpecificTuyaDimmer', data '{"type":"Buffer","data":[9,0,2,0,1,7,4,0,1,1]}' from endpoint 1 with groupID 0
zigbee2mqtt:info  2020-03-09 00:40:25: MQTT publish: topic 'zigbee2mqtt/0xec1bbdfffec3d049', payload '{"undefined":"ON","linkquality":23}'
zigbee2mqtt:debug 2020-03-09 00:45:24: Received Zigbee message from '0xec1bbdfffec3d049', type 'raw', cluster 'manuSpecificTuyaDimmer', data '{"type":"Buffer","data":[9,7,2,0,4,3,2,0,4,0,100,0,50]}' from endpoint 1 with groupID 0
zigbee2mqtt:info  2020-03-09 00:45:24: MQTT publish: topic 'zigbee2mqtt/0xec1bbdfffec3d049', payload '{"undefined":"ON","linkquality":63,"state_l3":"OFF"}'

zigbee2mqtt:debug 2020-03-09 00:40:28: Received Zigbee message from '0xec1bbdfffec3d049', type 'attributeReport', cluster 'genBasic', data '{"65506":21,"appVersion":255}' from endpoint 1 with groupID 0
zigbee2mqtt:debug 2020-03-09 00:40:30: Received Zigbee message from '0xec1bbdfffec3d049', type 'attributeReport', cluster 'genBasic', data '{"65506":21,"appVersion":82}' from endpoint 1 with groupID 0
zigbee2mqtt:debug 2020-03-09 00:40:30: Received Zigbee message from '0xec1bbdfffec3d049', type 'attributeReport', cluster 'genBasic', data '{"65503":"\u0000\u0000\u0000\u0000\u0005\u0000\u0000\u0000\u0000\u0005"}' from endpoint 1 with groupID 0

Koenkk, 2020-03-09:
Can you try with:

```js
{
    zigbeeModel: ['TS0601'],
    model: 'YS-MT750',
    vendor: 'TuYa',
    description: 'Tuya curtain motor',
    supports: 'open, close, stop, position',
    fromZigbee: [],
    toZigbee: [tz.cover_state],
},
```

and control the curtain by sending to `zigbee2mqtt/0xec1bbdfffec3d049/set` payload `{"state": "OPEN"}`, `{"state": "CLOSE"}`, `{"state": "STOP"}`

cchalong, 2020-03-09:
thanks for your comment, 
actually I'm beginner of HA, so I currently don't know how to send the payload. I'm searching for how to do and it can take some time, haha. I'll be back shortly 
or if there is a simple way, please let me be informed.

cchalong, 2020-03-09:
when I input in developer tools->Mqtt->publish a packet with zigbee2mqtt/0xec1bbdfffec3d049/set in topic and "state": "OPEN" in payload, 
I got the following log.

I'm not sure that what I did is exactly what you wanted me to do

zigbee2mqtt:debug 2020-03-10 04:52:18: Received MQTT message on 'zigbee2mqtt/0xec1bbdfffec3d049/set' with data '"state": "OPEN"'
zigbee2mqtt:debug 2020-03-10 04:52:18: Publishing 'set' 'state' to 'tuya_curtain_motor'
zigbee2mqtt:error 2020-03-10 04:52:18: Publish 'set' 'state' to 'tuya_curtain_motor' failed: 'Error: Cluster 'closuresWindowCovering' has no command 'undefined''
zigbee2mqtt:debug 2020-03-10 04:52:18: Error: Cluster 'closuresWindowCovering' has no command 'undefined'
    at Object.getCommand (/zigbee2mqtt-1.11.0/node_modules/zigbee-herdsman/dist/zcl/utils.js:106:19)
    at Endpoint.<anonymous> (/zigbee2mqtt-1.11.0/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:323:37)
    at Generator.next (<anonymous>)
    at /zigbee2mqtt-1.11.0/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:8:71
    at new Promise (<anonymous>)
    at __awaiter (/zigbee2mqtt-1.11.0/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:4:12)
    at Endpoint.command (/zigbee2mqtt-1.11.0/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:321:16)
    at Object.convertSet (/zigbee2mqtt-1.11.0/node_modules/zigbee-herdsman-converters/converters/toZigbee.js:201:26)
    at EntityPublish.onMQTTMessage (/zigbee2mqtt-1.11.0/lib/extension/entityPublish.js:173:52)
    at Controller.callExtensionMethod (/zigbee2mqtt-1.11.0/lib/controller.js:329:44)
zigbee2mqtt:info  2020-03-10 04:52:18: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"zigbee_publish_error","message":"Publish 'set' 'state' to 'tuya_curtain_motor' failed: 'Error: Cluster 'closuresWindowCovering' has no command 'undefined''","meta":{"friendly_name":"tuya_curtain_motor"}}'

Koenkk, 2020-03-09:
The payload should be `{"state": "OPEN"}` (with the `{` brackets).

cchalong, 2020-03-09:
and then, error msg disappeared but there is still no response from the device.

zigbee2mqtt:debug 2020-03-10 06:00:14: Received MQTT message on 'zigbee2mqtt/tuya_curtain_motor/set' with data '{"state": "OPEN"}'
zigbee2mqtt:debug 2020-03-10 06:00:14: Publishing 'set' 'state' to 'tuya_curtain_motor'
zigbee2mqtt:debug 2020-03-10 06:00:32: Received Zigbee message from 'tuya_curtain_motor', type 'attributeReport', cluster 'genBasic', data '{"65506":21,"appVersion":82}' from endpoint 1 with groupID 0
zigbee2mqtt:debug 2020-03-10 06:00:32: No converter available for 'YS-MT750' with cluster 'genBasic' and type 'attributeReport' and data '{"65506":21,"appVersion":82}'
zigbee2mqtt:debug 2020-03-10 06:00:43: Received MQTT message on 'zigbee2mqtt/tuya_curtain_motor/set' with data '{"state": "CLOSE"}'
zigbee2mqtt:debug 2020-03-10 06:00:43: Publishing 'set' 'state' to 'tuya_curtain_motor'

Koenkk, 2020-03-10:
In that case this device uses non-standard Zigbee commands, the only way to found out what to send is by sniffing the traffic when controlling it via the original gateway.

cchalong, 2020-03-10:
Okay I'm discussing it with manufacturer. And what gateway I need. Probably it will take some time to get the gateway. Anyway if any ways to sniff the traffic, please guide me. Thanks for your continuing help^^

jeholsa, 2020-05-13:
Any luck with this? I have the same Tuya YS-MT750 and try to get it connected with RaspBee and Deconz to Home Assistant. Is this motor and Zigbee interface is the same as https://www.zigbee2mqtt.io/devices/owvfni3.html or https://github.com/Koenkk/zigbee-herdsman-converters/issues/1159 ?

cchalong, 2020-05-26:
> Koenkk/zigbee-herdsman-converters#1159

is it paired with the products as you linked?

DomiStyle, 2020-05-26:
Support for the TS0601 was added with https://github.com/Koenkk/zigbee-herdsman-converters/pull/1211 so it should work in the latest version.

Maybe these two can be merged @Koenkk?
https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/devices.js#L1158
https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/devices.js#L1170

Koenkk, 2020-05-26:
@DomiStyle aren't these different devices? (pictures on https://www.zigbee2mqtt.io/information/supported_devices.html#tuya are different)

Koenkk, 2020-05-26:
@DomiStyle thanks for your response. It indeed makes sense to merge them so done this.

Assuming this issue can be closed as the device is supported now.

closed 2020-05-26 by Koenkk