Koenkk/zigbee2mqtt #3090: Xiaomi WXCJKG13LM and WXCJKG13LM bound with ZNCZ02LM automatically

didiergithub, 2020-03-08:
# Bug Report
## What happened
I very recently purchase a CC2531 and a few Xiaomi/Aqara devices and it is all new to me.

I successfully paired 2 x ZNCZ02LM (Power Plug), 1 WXCJKG13LM (Aqara Opple switch 6 buttons) and 2 x WXCJKG12LM (Aqara Opple switch 4 buttons).

The button 1 and 2 of all 3 switches turn on and off both power plugs at the same time by default, something I don't really want as I want to control these actions via NodeRed.

I tried both the ZStack 1.2 source route and the default firmware. Same behaviour.

## What did you expect to happen
I expected that after pairing the switches and the power plugs would not be bound by default.

## How to reproduce it (minimal and precise)

## Debug Info
zigbee2mqtt version: 1.11.0
CC253X firmware version: CC2531_DEFAULT_20190608

Koenkk, 2020-03-09:
Can you share your `database.db` file?

didiergithub, 2020-03-09:
Good evening,

Here it is:
[database.zip](https://github.com/Koenkk/zigbee2mqtt/files/4308448/database.zip)

Thank you for looking into this.

Koenkk, 2020-03-09:
Thanks, I see whats going wrong, when starting zigbee2mqtt, do you see `Failed to configure .....` in the logs? (can take up to 60 seconds to show).

didiergithub, 2020-03-09:
Here is what I see:

[zigbee2mqtt debug 1.txt](https://github.com/Koenkk/zigbee2mqtt/files/4308524/zigbee2mqtt.debug.1.txt)

Bedankt

Koenkk, 2020-03-10:
Could it be that you have a custom zigbee-herdsman-converters? (did you ever try to add support for a new device or something similar?).

To make sure please provide the content of `/opt/zigbee2mqtt/node_modules/zigbee-herdsman-converters/devices.js`.

didiergithub, 2020-03-10:
Hi Koen,

I did play with the device.js as I had some missing converters for some functions.

Here are the device.js as well as the fromzigbee.js:

[Device and fromzigbee.zip](https://github.com/Koenkk/zigbee2mqtt/files/4314029/Device.and.fromzigbee.zip)

Thanks
Regards
Didier

didiergithub, 2020-03-10:
I just started to use the latest devices.js I found on your github (It looks like it was posted less than an hour ago).

That doesn't make a difference though.

Do I have to re-pair the devices maybe?

Koenkk, 2020-03-10:
yes please remove it first (https://www.zigbee2mqtt.io/information/mqtt_topics_and_message_structure.html#zigbee2mqttbridgeconfigremove) and pair it (and please provide the log of that).

didiergithub, 2020-03-10:
I struggles a bit but at the end I got all three switches to work. I had to re-pair them a few times.

They even support double and triple click on all button which wasn't the case before.

I guess it was the devices.js file then.

Zeer bedankt voor u hulp.

Cheers

Didier

Koenkk, 2020-03-10:
Mooi, bedankt en veel plezier :smile: (I assume this can be closed now).

closed 2020-03-10 by Koenkk

didiergithub, 2020-03-11:
Hi Koen,

Yes of course it can be closed.

Thank you again for your great support.

Cheers
Didier