Koenkk/zigbee2mqtt #3092: Bug report: "Entity is non-numeric", "Unknown" for sensor with Home Assistant when restart

dzungpv, 2020-03-09:
I use hassio (Now: Home Assistant), every-time HA restart, zigbee2mqtt (add-on) still running on the docker image. But temperature/humidity/light sensor show "Entity is non-numeric", "Unknown" in the UI  and it only fix next sensor value update or restart zigbee2mqtt add-ons.

Affect devices: WSDCGQ01LM, WSDCGQ11LM, RTCGQ11LM (maybe all Xiaomi sensor)

## Debug Info
zigbee2mqtt version: 1.11.0
CC1352P-2 firmware version:  20191106/20200229
HA version: 0.106.5
MQTT server:  Mosquitto broker add-on 5.1

Koenkk, 2020-03-09:
Could it be because the values are not published at that point yet?

dzungpv, 2020-03-10:
> 
> 
> Could it be because the values are not published at that point yet?

Yes, whenever HA restart

Koenkk, 2020-03-10:
Could you set `retain: true` for these sensors in the Zigbee2mqtt config and check if it solves the issue?

dzungpv, 2020-03-11:
> 
> 
> Could you set `retain: true` for these sensors in the Zigbee2mqtt config and check if it solves the issue?

Yes, it fixed. So default all sensor, "retain: true" better apply for all of them.

closed 2020-05-17 by stale[bot]