Koenkk/zigbee2mqtt #3093: Messed up CC1352p-2 firmware upgrade - fixable?

jarrah31, 2020-03-09:
So... I attempted to flash the CC1352P-2 board on my MacBook via a Windows 10 VM. It initially connected ok and prompted me to update the debugger firmware. I selected Yes and then it appeared to disconnect and reconnect, during which time the usual VMware prompt to choose which machine to connect the USB device to appeared. I chose Windows, but by then the firmware update process failed, along with a message saying don't disconnect the board during an upgrade... Yep, I know....

Before it failed the LED lit up on the board when plugging in the USB cable, but now I have no light.

Does anyone know if it's possible to recover from this please? Do I need to connect via serial?

TIA

sjorge, 2020-03-09:
I think you can just try again, but I think you can't use a VM as it will reset the USB device so from the host perspective it gets unplugged and replugged.

You might want to have a look at https://github.com/JelmerT/cc2538-bsl, I think this should be able to flash the CC1352 also.

nanobra1n, 2020-03-10:
Is there a reason why are you not using the macOS version of [UNIFLASH](http://www.ti.com/tool/download/UNIFLASH)?

jarrah31, 2020-03-10:
> Is there a reason why are you not using the macOS version of [UNIFLASH](http://www.ti.com/tool/download/UNIFLASH)?

Ahh I didn’t realise, good to know for future, thanks.

@sjorge thanks for the link, i’ll give that a go. The only problem could be that the device is no longer recognised by Windows when plugged in, but i’ll see what Linux or MacOS sees later today.

jarrah31, 2020-03-10:
I also opened a thread on the TI forum and luckily the following post has provided an answer:
https://e2e.ti.com/support/wireless-connectivity/sub-1-ghz/f/156/p/887212/3281522#3281522

I will need to purchase another CC1352P-2 to fix the broken one, but that's ok as I'll use it as a router to increase range.

CodeFinder2, 2020-03-10:
So, then you should close this issue... (I guess),

closed 2020-03-11 by jarrah31