Koenkk/zigbee2mqtt #3095: Trust 202 unsupported but previously working

mousemat86, 2020-03-09:
I recently updated firmware version zStack12 20190608 and zigbee2mqtt version 1.11.0 and the device will no longer pair. Could any one shed some light on why it's unsupported? 

Previously running zigbee2mqtt version 1.6.0 - not sure of previous firmware version. 

`zigbee2mqtt:info  2020-03-07 10:14:54: Device '0x00158d0001314742' joined
zigbee2mqtt:info  2020-03-07 10:14:54: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_connected","message":{"friendly_name":"0x00158d0001314742"}}'
zigbee2mqtt:info  2020-03-07 10:14:54: Starting interview of '0x00158d0001314742'
zigbee2mqtt:info  2020-03-07 10:14:54: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_started","meta":{"friendly_name":"0x00158d0001314742"}}'
zigbee2mqtt:warn  2020-03-07 10:14:57: Received message from unsupported device with Zigbee model 'undefined'
zigbee2mqtt:warn  2020-03-07 10:14:57: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
zigbee2mqtt:warn  2020-03-07 10:14:58: Received message from unsupported device with Zigbee model 'undefined'
zigbee2mqtt:warn  2020-03-07 10:14:58: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
zigbee2mqtt:warn  2020-03-07 10:15:00: Received message from unsupported device with Zigbee model 'undefined'
zigbee2mqtt:warn  2020-03-07 10:15:00: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
zigbee2mqtt:warn  2020-03-07 10:15:01: Received message from unsupported device with Zigbee model 'undefined'
zigbee2mqtt:warn  2020-03-07 10:15:01: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
zigbee2mqtt:warn  2020-03-07 10:15:02: Received message from unsupported device with Zigbee model 'undefined'
zigbee2mqtt:warn  2020-03-07 10:15:02: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
zigbee2mqtt:warn  2020-03-07 10:15:02: Received message from unsupported device with Zigbee model 'undefined'
zigbee2mqtt:warn  2020-03-07 10:15:02: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
zigbee2mqtt:warn  2020-03-07 10:15:05: Received message from unsupported device with Zigbee model 'undefined'
zigbee2mqtt:warn  2020-03-07 10:15:05: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
zigbee2mqtt:error 2020-03-07 10:15:25: Failed to interview '0x00158d0001314742', device has not successfully been paired
zigbee2mqtt:info  2020-03-07 10:15:25: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_failed","meta":{"friendly_name":"0x00158d0001314742"}}'
`

_Originally posted by @mousemat86 in https://github.com/Koenkk/zigbee2mqtt/issues/635#issuecomment-596071493_

Koenkk, 2020-03-10:
It seems that the interview fails, please provide the herdsman debug logging when pairing the device.

To enable herdsman debug logging, see https://www.zigbee2mqtt.io/information/debug.html#zigbee-herdsman-debug-logging

james-fry, 2020-03-11:
I have one of these and I found pairing to be extremely challenging - with many repeats of reset and pair cycle.

mousemat86, 2020-03-11:
OK during the process of retrieving a debug log it paired. Thanks for the suggestion. 

@james-fry could you assit me with setting up dimming and using action_groups for different bulbs - any guidance would be appriciated tomplums@gmail.com

closed 2020-03-11 by mousemat86

james-fry, 2020-03-11:
> OK during the process of retrieving a debug log it paired. Thanks for the suggestion.
> 
> @james-fry could you assit me with setting up dimming and using action_groups for different bulbs - any guidance would be very much appreciated if you have time [tomplums@gmail.com](mailto:tomplums@gmail.com)

I use node-red and i posted the flow I used for this in another issue:
https://github.com/Koenkk/zigbee2mqtt/issues/635#issuecomment-526774720

Actually I made some improvements since then and this is my current flow:

[image]

`[{"id":"871af3aa.f2eef","type":"switch","z":"3da2c349.0ebdcc","name":"On, off, up, down","property":"payload.action","propertyType":"msg","rules":[{"t":"eq","v":"on","vt":"str"},{"t":"eq","v":"off","vt":"str"},{"t":"eq","v":"up-press","vt":"str"},{"t":"eq","v":"down-press","vt":"str"}],"checkall":"true","repair":true,"outputs":4,"x":990,"y":1240,"wires":[["4ee0c9c1.0fd358"],["f202f877.ff4ae8"],["78a74e7c.8701b"],["e8618271.79e67"]]},{"id":"4ee0c9c1.0fd358","type":"api-call-service","z":"3da2c349.0ebdcc","name":"Turn On James Bedside Light","server":"b8197728.8ac978","version":1,"debugenabled":false,"service_domain":"light","service":"turn_on","entityId":"light.0x00178801024996be_light","data":"","dataType":"json","mergecontext":"","output_location":"","output_location_type":"none","mustacheAltTags":false,"x":1410,"y":1160,"wires":[[]]},{"id":"f202f877.ff4ae8","type":"api-call-service","z":"3da2c349.0ebdcc","name":"Turn Off James Bedside Lights","server":"b8197728.8ac978","version":1,"debugenabled":false,"service_domain":"light","service":"turn_off","entityId":"light.0x00178801024996be_light","data":"","dataType":"json","mergecontext":"","output_location":"","output_location_type":"none","mustacheAltTags":false,"x":1410,"y":1220,"wires":[[]]},{"id":"f6063295.d2b4a","type":"function","z":"3da2c349.0ebdcc","name":"Set brightness in payload +32","func":"var brightness = msg.data.attributes.brightness;\n\nif (brightness < 32) {\n   brightness = 32;\n}  else if (brightness >= 32 && brightness < 64) {\n   brightness = 64;\n}  else if (brightness >= 64 && brightness < 96) {\n   brightness = 96;\n}  else if (brightness >= 96 && brightness < 128) {\n   brightness = 128;\n}  else if (brightness >= 128 && brightness < 160) {\n   brightness = 160;\n}  else if (brightness >= 160 && brightness < 192) {\n   brightness = 192;\n}  else if (brightness >= 192 && brightness < 224) {\n   brightness = 224;\n}  else if (brightness >= 224) {\n   brightness = 255;\n} \n\nmsg.payload = {\n domain: 'light',\n service: 'turn_on',\n data: { entity_id: 'light.0x00178801024996be_light', brightness: brightness}\n};\nreturn msg;","outputs":1,"noerr":0,"x":1830,"y":1280,"wires":[["367d8265.bc17ee"]]},{"id":"367d8265.bc17ee","type":"api-call-service","z":"3da2c349.0ebdcc","name":"Set James Bedside Brightness","server":"b8197728.8ac978","version":"1","debugenabled":false,"service_domain":"light","service":"turn_on","entityId":"light.0x00178801024996be_light","data":"","dataType":"json","mergecontext":"","output_location":"payload","output_location_type":"msg","mustacheAltTags":false,"x":2150,"y":1280,"wires":[[]]},{"id":"35c42d05.81b5e2","type":"mqtt in","z":"3da2c349.0ebdcc","name":"","topic":"zigbee2mqtt_cc1352p/Trust remote","qos":"2","datatype":"auto","broker":"a9c1a622.a79798","x":220,"y":1460,"wires":[["2090a197.5d9bae"]]},{"id":"2090a197.5d9bae","type":"json","z":"3da2c349.0ebdcc","name":"","property":"payload","action":"","pretty":false,"x":550,"y":1460,"wires":[["9a1a1a7c.393cc8"]]},{"id":"9a1a1a7c.393cc8","type":"switch","z":"3da2c349.0ebdcc","name":"Action Group","property":"payload.action_group","propertyType":"msg","rules":[{"t":"eq","v":"150","vt":"str"},{"t":"eq","v":"145","vt":"str"},{"t":"eq","v":"146","vt":"str"},{"t":"eq","v":"147","vt":"str"},{"t":"eq","v":"148","vt":"str"},{"t":"eq","v":"149","vt":"str"}],"checkall":"true","repair":false,"outputs":6,"x":750,"y":1460,"wires":[["871af3aa.f2eef"],["2751481a.365828"],["5625935.1b3176c"],[],[],[]]},{"id":"bd59907f.b0f92","type":"api-call-service","z":"3da2c349.0ebdcc","name":"Turn On SmartMi Fan","server":"b8197728.8ac978","version":1,"service_domain":"fan","service":"turn_on","entityId":"fan.xiaomi_smart_fan_2","data":"","dataType":"json","mergecontext":"","output_location":"","output_location_type":"none","mustacheAltTags":false,"x":1380,"y":1640,"wires":[[]]},{"id":"b756e137.6a91c","type":"api-call-service","z":"3da2c349.0ebdcc","name":"Turn Off SmartMi Fan","server":"b8197728.8ac978","version":1,"service_domain":"fan","service":"turn_off","entityId":"fan.xiaomi_smart_fan_2","data":"","dataType":"json","mergecontext":"","output_location":"","output_location_type":"none","mustacheAltTags":false,"x":1380,"y":1700,"wires":[[]]},{"id":"5625935.1b3176c","type":"switch","z":"3da2c349.0ebdcc","name":"On, off, up, down","property":"payload.action","propertyType":"msg","rules":[{"t":"eq","v":"on","vt":"str"},{"t":"eq","v":"off","vt":"str"},{"t":"eq","v":"up-press","vt":"str"},{"t":"eq","v":"down-press","vt":"str"}],"checkall":"true","repair":true,"outputs":4,"x":990,"y":1720,"wires":[["bd59907f.b0f92"],["b756e137.6a91c"],["e0de85db.47b3f8"],["a0d4d7d3.e7fda8"]]},{"id":"2751481a.365828","type":"switch","z":"3da2c349.0ebdcc","name":"On, off, up, down","property":"payload.action","propertyType":"msg","rules":[{"t":"eq","v":"on","vt":"str"},{"t":"eq","v":"off","vt":"str"},{"t":"eq","v":"up-press","vt":"str"},{"t":"eq","v":"down-press","vt":"str"}],"checkal [... cut ...] "legacy":false,"addon":true,"rejectUnauthorizedCerts":true,"ha_boolean":"y|yes|true|on|home|open","connectionDelay":true},{"id":"a9c1a622.a79798","type":"mqtt-broker","z":"","name":"Mosquitto","broker":"192.168.1.10","port":"1883","tls":"","clientid":"NodeRed","usetls":false,"compatmode":false,"keepalive":"300","cleansession":true,"birthTopic":"","birthQos":"0","birthPayload":"","closeTopic":"","closeQos":"0","closePayload":"","willTopic":"/unxpected_disconnect","willQos":"0","willPayload":""}]`

mousemat86, 2020-03-12:
Brilliant thanks @james-fry I actually setup your other flow last night and it was working very well. I'll give this a go. Appreciate it