Koenkk/zigbee2mqtt #3096: cannot update color_temp when bulb is in off state

wro03302, 2020-03-10:
# Bug Report
Hi, 
thanks for zigbee2mqtt - it is great and still better and better!
I have tiny issue but it is a bit problematic in my setup.
I do not allow to change color temperature, it is done automaticaly according to schedule.
when schedule is adjusting temperature - not all of bulbs are in ON state, those that are in off state "reject" update of color_temperature.

## What happened
When setting color temp to 250 from 451 it is still 451
using attributes mode (openhab)

`zigbee2mqtt/urZbiuro1/set/color_temp 250 <- setting color_temp 250
zigbee2mqtt/urZbiuro1/state OFF
zigbee2mqtt/urZbiuro1/color_temp 250 <- looks that is accepted
zigbee2mqtt/urZbiuro1/brightness 224
zigbee2mqtt/urZbiuro1/linkquality 31
zigbee2mqtt/urZbiuro1/update_available false
zigbee2mqtt/urZbiuro1/state OFF
zigbee2mqtt/urZbiuro1/color_temp 451 <- reporting old value...
zigbee2mqtt/urZbiuro1/brightness 224
zigbee2mqtt/urZbiuro1/linkquality 31
zigbee2mqtt/urZbiuro1/update_available false
`
## What did you expect to happen

I would expect that bulbs that are avaliable, will receive an update.
when bulb is ON, everything is working as expected.

## How to reproduce it (minimal and precise)
```
$ mosquitto_pub -t 'zigbee2mqtt/urZbiuro1/set/state' -m on;
$ mosquitto_pub -t 'zigbee2mqtt/urZbiuro1/set/color_temp' -m 451;
$ mosquitto_pub -t 'zigbee2mqtt/urZbiuro1/set/state' -m off;
$ mosquitto_pub -t 'zigbee2mqtt/urZbiuro1/set/color_temp' -m 250;

```
in second window monitor what is happening in mosquitto_sub
## Debug Info
`zigbee2mqtt/bridge/config {"version":"1.11.0","commit":"e140cf5","coordinator":{"type":"zStack12","meta":{"transportrev":2,"product":0,"majorrel":2,"minorrel":6,"maintrel":3,"revision":20190619}},"log_level":"info","permit_join":false}
`
zigbee2mqtt version:
CC253X firmware version:

sjorge, 2020-03-10:
It depends on the bulb, my CWS Trådfri bulbs take color updates and my WS Trådfri takes color_temp updates even when off.

But my Hue, Innr, and Osram bulbs do not.

wro03302, 2020-03-10:
Hmm I have Tradfrii bulbs....

sjorge, 2020-03-10:
Are they CWS or WS trådfri's? The CWS ones don't accept color_temp only color.

wro03302, 2020-03-10:
My test was done with WS (White Spectrum)

sjorge, 2020-03-10:
Interesting, I don't have mine paired but I could have sworn it used to work for them also.

closed 2020-05-16 by stale[bot]