Koenkk/zigbee2mqtt #3097: Heiman HS1SAs won't join anymore: Failed to interview / Failed to configure

SaturnusDJ, 2020-03-10:
Z-Stack_Home_1.2.
Zigbee2MQTT 1.11.0.

3x Osram Smart+ as router.

```
12:04:33: Device '0x00158d0004c0a6e5' joined
12:04:33: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_connected","message":{"friendly_name":"0x00158d0004c0a6e5"}}'
12:04:33: Starting interview of '0x00158d0004c0a6e5'
12:04:33: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_started","meta":{"friendly_name":"0x00158d0004c0a6e5"}}'
12:05:15: Failed to interview '0x00158d0004c0a6e5', device has not successfully been paired
12:05:15: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_failed","meta":{"friendly_name":"0x00158d0004c0a6e5"}}'
12:05:15: Configuring '0x00158d0004c0a6e5'
12:05:25: Failed to configure '0x00158d0004c0a6e5', attempt 1 (Error: Bind 0x00158d0004c0a6e5/1 genPowerCfg from '0x00124b001cd47ff6/1' failed (Error: AREQ - ZDO - bindRsp after 10000ms)
    at Endpoint.<anonymous> (/app/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:235:23)
    at Generator.throw (<anonymous>)
    at rejected (/app/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:6:65))
```

Strange thing: before the Osrams were paired, the smoke detector paired with the coordinator successfully (as the first ever device). Is it possible that these Osrams defunct in their supposed router functionality and on top of that 'ruin' the network?
Now I started from scratch with the Osrams first and the smoke detectors still do not pair, and even trying to pair at millimeters close to the coordinator does not work.

Osram firmware version was `swBuildId":"V1.04.12"`

```12:30:57: Finished update of 'light_sebas_bed', from {"softwareBuildID":"V1.04.12","dateCode":"20140331DEOS****"}' to '{"softwareBuildID":"V1.05.09","dateCode":"20140331DEOS****"}'```

CodeFinder2, 2020-03-10:
As far was I know, you should pair the (end) devices at their designated position as close as possible to let the pick the "best" (mostly closest) router nearby. So, no need to just pair it with the coordinator although devices will - sooner or later - pick a new router once another fails to work (but this only happens if the router breaks completely, AFAIK, so that it's better to do what I wrote first).

I haven't had any issues "pairing through" OSRAM plugs. Nonetheless, all my experiences are based on the old firmware of the plugs (v1.04.12).

I had this strange "join interval" some time ago too (and to be clear: i am not talking about the time when some automation in HA automatically disables joining).

Note that there are different firmwares and that you can also try reflashing. People have reported that it might also help.

SaturnusDJ, 2020-03-12:
I paired with coordinator directly to get everything working. After positioning the devices they picked a router.
Maybe 'issue 2' is the only thing that is happening and that being the reason I could not pair via a router. Will have a look at this when pairing again.

closed 2020-03-12 by SaturnusDJ