Koenkk/zigbee2mqtt #3098: Ikea e14 unable to OTA update: device did not request any image blocks

mr2c12, 2020-03-10:
Z2m v1.9
Coordinator: CC26X2R1

Device: IKEA TRADFRI LED bulb E14 600 lumen, dimmable, white spectrum, opal white (LED1733G7)

Any attempt to update its firmware fails with error: "device did not request any image blocks"

I already tried to:
- turn off and on the device
- place the bulb really close to the coordinator
- update other devices (e.g. an ikea signal extender) works

Interesting part of the log:

> zigbee2mqtt:info  2020-03-10 18:46:07: Update of '0x000d6ffffe14a719' at 0%
> zigbee2mqtt:info  2020-03-10 18:46:07: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"ota_update","message":"Update of '0x000d6ffffe14a719' at 0%","meta":{"status":"update_progress","device":"0x000d6ffffe14a719","progress":0}}'
> (node:547) UnhandledPromiseRejectionWarning: Error: AREQ - AF - dataConfirm after 3000ms
>     at Timeout.object.timer.setTimeout [as _onTimeout] (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/utils/waitress.js:44:24)
>     at ontimeout (timers.js:436:11)
>     at tryOnTimeout (timers.js:300:5)
>     at listOnTimeout (timers.js:263:5)
>     at Timer.processTimers (timers.js:223:10)
> (node:547) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). (rejection id: 1)
> (node:547) [DEP0018] DeprecationWarning: Unhandled promise rejections are deprecated. In the future, promise rejections that are not handled will terminate the Node.js process with a non-zero exit code.
> zigbee2mqtt:error 2020-03-10 18:47:13: Update of '0x000d6ffffe14a719' failed (Timeout: device did not request any image blocks)

Can you please help me?

mr2c12, 2020-05-09:
Fixed ("sort of") when I run z2m on a more powerful machine!

closed 2020-05-09 by mr2c12

Tonio16, 2021-06-09:
Hello

What do you mean with a more powerfull machine? cpu ?

Antoine

JoachimCanu, 2021-12-06:
Bonjour,

Une solution à ce problème ?

Merci

mipsou, 2024-04-19:
Hi,
Same trouble with green HA
`Zigbee2MQTT:info  2024-04-19 04:22:02: MQTT publish: topic 'zigbee2mqtt/TRADFRI signal repeater 1', payload '{"linkquality":110,"update":{"installed_version":570447409,"latest_version":587753009,"state":"available"},"update_available":true}'
Zigbee2MQTT:info  2024-04-19 04:22:02: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Update of 'TRADFRI signal repeater 1' failed (OTA: Timeout, device did not request any image blocks)","meta":{"device":"TRADFRI signal repeater 1","status":"update_failed"},"type":"ota_update"}'
Zigbee2MQTT:info  2024-04-19 04:22:02: MQTT publish: topic 'zigbee2mqtt/bridge/response/device/ota_update/update', payload '{"data":{"id":"TRADFRI signal repeater 1"},"error":"Update of 'TRADFRI signal repeater 1' failed (OTA: Timeout, device did not request any image blocks)","status":"error","transaction":"g777v-1"}'
Zigbee2MQTT:error 2024-04-19 04:22:02: Update of 'TRADFRI signal repeater 1' failed (OTA: Timeout, device did not request any image blocks)`