Koenkk/zigbee2mqtt #3103: modelId

MaxRei-dev, 2020-03-11:
Hi,
running 1.11.0. 
Can somone confirm that there is no modelId in the json ?
thanks

MaxRei-dev, 2020-03-12:
I mean when a device is reporting..
Thanks

MaxRei-dev, 2020-03-16:
no one ?

sjorge, 2020-03-16:
I don't think there every was, aside from in the output published to `zigee2mqtt/bridge/devices`

MaxRei-dev, 2020-03-17:
thanks. i can confirm that it was in older versions but not now. but anyways i will do things different. (i was using this to group the battery level of devices. eg lumi.sens, lumi.sensor_smoke,... and the react different to baettery level. eg. Smoke sensor -> acoustic alarm so i dont forget to change the battery, Motion sensor -> just an push notification and so on)

closed 2020-03-18 by MaxRei-dev