Koenkk/zigbee2mqtt #3104: Entity '...' is unknown after debugging - LED1836G9

bromosky, 2020-03-11:
Hey there,

wanted to add E1743 Dimmer from IKEA to my Network, after adding two LED1836G9 from IKEA. (Had to update to 1.11.0 of herdsman) Everything worked out, till i tried to add these 2 Dimmers.
Reflashed the CC2531, just to make sure, it's fine. (It is). Can't connect two dimmers.
So I tried the two IKEA bulbs again(On/off). One is working fine, the other one says 
`Entity '0xccccccfffeab3455' is unknown`. 
So I looked into the configuration.yaml.
`  '0xccccccfffeab3455':
    friendly_name: '0xccccccfffeab3455'
    retains: false
`
Back to the Database: there is no entry for the adress. I had 9 devices, but now, it only shows 8 id's.
So the database seems to "forgot" my one bulb.

What I did try:
-rejoining the Bulb (after deleting it from configuration.yaml and without it)
-come up with some creative ideas for the database -> didn't let herdsman start :- )
-removing the bulb ->
`zigbee2mqtt:debug 2020-03-11 16:58:33: Received MQTT message on 'zigbee2mqtt/bridge/config/remove' with data '0xccccccfffeab3455'
zigbee2mqtt:error 2020-03-11 16:58:33: Cannot remove, device '0xccccccfffeab3455' does not exist
zigbee2mqtt:info  2020-03-11 16:58:33: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_removed_failed","message":"0xccccccfffeab3455"}'
`

zigbee2mqtt version: 1.11.0
CC2531 firmware version: newest (reflashed today)

So, how to get the second bulb running again?
Maybe, just maybe, how to add these two dimmer?

I hope, someone will come up with any idea. I will be grateful!
Max :)

Koenkk, 2020-03-11:
So the current problem is that `0xccccccfffeab3455` doesn't pair? Make sure to follow https://www.zigbee2mqtt.io/information/FAQ.html#why-does-my-device-not-or-fail-to-pair

bromosky, 2020-03-11:
Hey,
thanks for the fast reply.
The bulb was added before. After unplugging, reflashing the CC2531, the `0xccccccfffeab3455` isn't in the database anymore. Everything works fine, except this one bulb, which shows now, that there is no Entity for it.

Permit-join is true, no RPi4, bulb isn't paired anymore (factory reset), limit isn't reached (8+1 devices), distance bulb<>coordinator was 10cm, unplugging didn't work, after reflashing, the bulb didn't show up in database no more (initial failure)
->so nothing from the FAQ worked for me. How is it possible, that a fully functioning bulb disappeared from the database.db? That's why the bulb has no Entity, because it isn't registered in the DB, but in the configuration.yaml? 

Max :)

Koenkk, 2020-03-11:
I have no clue for now, do you still have debug logs of this? (Or even better can you reproduce it?)

bromosky, 2020-03-11:
Log level is just info, because i haven't had any problems before this whatsoever.

Adding:

`Mar 11 13:25:30 FhemMaxPi npm[2159]: zigbee2mqtt:info  2020-03-11 13:25:30: Device '0xccccccfffeab3455' joined
Mar 11 13:25:30 FhemMaxPi npm[2159]: zigbee2mqtt:info  2020-03-11 13:25:30: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_connected","message":{"friendly_name":"0xccccccfffeab3455"}}'
Mar 11 13:25:30 FhemMaxPi npm[2159]: zigbee2mqtt:info  2020-03-11 13:25:30: Starting interview of '0xccccccfffeab3455'
Mar 11 13:25:30 FhemMaxPi npm[2159]: zigbee2mqtt:info  2020-03-11 13:25:30: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_started","meta":{"friendly_name":"0xccccccfffeab3455"}}'
Mar 11 13:25:41 FhemMaxPi npm[2159]: zigbee2mqtt:info  2020-03-11 13:25:41: Successfully interviewed '0xccccccfffeab3455', device has successfully been paired
Mar 11 13:25:41 FhemMaxPi npm[2159]: zigbee2mqtt:info  2020-03-11 13:25:41: Device '0xccccccfffeab3455' is supported, identified as: IKEA TRADFRI LED bulb E26/E27 806 lumen, dimmable, warm white (LED1836G9)
Mar 11 13:25:41 FhemMaxPi npm[2159]: zigbee2mqtt:info  2020-03-11 13:25:41: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_successful","meta":{"friendly_name":"0xccccccfffeab3455","model":"LED1836G9","vendor":"IKEA","description":"TRADFRI LED bulb E26/E27 806 lumen, dimmable, warm white","supported":true}}'
Mar 11 13:27:53 FhemMaxPi npm[2159]: zigbee2mqtt:info  2020-03-11 13:27:53: MQTT publish: topic 'zigbee2mqtt/0xd0cf5efffe1f5a5f', payload '{"color_temp":254,"state":"OFF","brightness":185,"linkquality":94,"color_mode":2,"color":{"x":0.4599,"y":0.4106}}'
Mar 11 13:28:21 FhemMaxPi npm[2159]: zigbee2mqtt:info  2020-03-11 13:28:21: MQTT publish: topic 'zigbee2mqtt/0xccccccfffeab3455', payload '{"update_available":false}'
Mar 11 13:32:10 FhemMaxPi npm[2159]: zigbee2mqtt:info  2020-03-11 13:32:10: MQTT publish: topic 'zigbee2mqtt/0xccccccfffeab3455', payload '{"update_available":false,"state":"ON","brightness":125}'`

Leaving network
`Mar 11 14:52:45 FhemMaxPi npm[2159]: zigbee2mqtt:warn  2020-03-11 14:52:45: Device '0xccccccfffeab3455' left the network
Mar 11 14:52:45 FhemMaxPi npm[2159]: zigbee2mqtt:info  2020-03-11 14:52:45: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_removed","message":"left_network","meta":{"friendly_name":"0xccccccfffeab3455"}}'
Mar 11 14:52:46 FhemMaxPi npm[2159]: zigbee2mqtt:warn  2020-03-11 14:52:46: Device '0xccccccfffeab3455' left the network
Mar 11 14:52:46 FhemMaxPi npm[2159]: zigbee2mqtt:info  2020-03-11 14:52:46: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_removed","message":"left_network","meta":{"friendly_name":"0xccccccfffeab3455"}}'`

I didn't triggered the leaving. It did it on it's own. Last command was 5 minutes before the leaving. That's quite interesting. So the bulb might be the problem here. But why doesn't a factory reset fixes this? Do you see any information, which could help you?

bromosky, 2020-03-12:
Update:
Drove to IKEA. Got a new set. Added the switches, so adding something to the network works just fine.
Tried with the new bulb, again. It doesn't show up available to pair or something similar. So, it wasn't the bulb, but I am unable to add this type of bulb. 

LED1836G9 won't be recognised by CC2531 after adding two, the second one left the network. Now I'm unable to add any LED1836G9 again.

manju-rn, 2020-03-16:
Well, I never had any issues adding, just need to ensure that during the pairing process CC2531 is sufficiently close.  If you do not have the CC2531 with antenna, suggest you order that, it drastically increases the range and the pairing process is quite smoother as well.

closed 2020-05-22 by stale[bot]