Koenkk/zigbee2mqtt #3105: 'lastSeen' vs 'last_seen' confusion, can we make this clearer?

backcountrymountains, 2020-03-11:
So zigbee2mqtt/bridge/config/devices has 'lastSeen' , which, with my devices, shows the time my devices were added to z2m or, occasionally, when the device was actually last seen. I'm not sure what information is used to update the 'lastSeen' information.

If I add 'last_seen' to the configuration.yaml and individually change the state of the device, then the topic for each of my devices contains an accurate 'last_seen' time.

However, if I use zigbee groups as configured in groups.yaml, the 'last_seen' information does not update in the topic for each individual device nor does it appear in the topic of the group.

Is it possible to do something to make the difference between 'lastSeen' and 'last_seen' clearer? 

Can the zigbee2mqtt/bridge/config/devices be updated to use the 'last_seen' instead of 'lastSeen'?

Can groups incorporate 'last_seen' or can individual devices be updated with 'last_seen' when the containing group is utilized?

Is this an issue specific to me?

Thanks! z2m is great!

Version of Zigbee2Mqtt: 1.11.0

Koenkk, 2020-03-11:
- lastSeen = last_seen
- Last seen is updated when a message from a device is received
- Groups don't send messages, so they don't have a lastSeen

backcountrymountains, 2020-03-11:
In my experience lastSeen != last_seen
For me:
```
zigbee2mqtt/bridge/config/devices
  {
    "ieeeAddr": "0xccccccfffee9af1e",
    "type": "Router",
    "networkAddress": 26633,
    "model": "A9A19A60WESDZ02",
    "friendly_name": "0xccccccfffee9af1e",
    "manufacturerID": 4688,
    "manufacturerName": "The Home Depot",
    "powerSource": "Mains (single phase)",
    "modelID": "Ecosmart-ZBT-A19-CCT-Bulb",
    "hardwareVersion": 1,
    "softwareBuildID": "2.3",
    "dateCode": "20190514-346",
    "lastSeen": 1583953912555
  },
```
Which is Wednesday, March 11, 2020 1:11:52.555 PM GMT-06:00 DST
vs
```
zigbee2mqtt/0x7cb03eaa00a5c039
{
  "state": "OFF",
  "color_temp": 170,
  "brightness": 254,
  "last_seen": 1583952111970
}
```
Which is Wednesday, March 11, 2020 12:41:51.970 PM GMT-06:00 DST

I set up 'availability_timeout: 300' since writing the issue, and it seems that the zigbee2mqtt/bridge/config/devices topic is updating lastSeen based on queries from the controller, but the device specific topic seems to only be updating last_seen when the device itself is turned on directly.

Not a big deal, I suppose

Koenkk, 2020-03-12:
After `zigbee2mqtt/0xccccccfffee9af1e` can you directly do a `zigbee2mqtt/bridge/config/devices` and see if the lastSeen time equals? Please provide the complete log of this.

backcountrymountains, 2020-03-18:
My CC2531 seems to have irreperably crashed. Sadly the ground pad has separated from the board so I can't reflash it to recover :`(

closed 2020-05-24 by stale[bot]