Koenkk/zigbee2mqtt #3106: iCasa ICZB-KPD18S support

dresende, 2020-03-11:
This is a supported device and I was able to add it to my CC pen. The problem is that I don't receive any button tap information. I tried to follow the controller manual to try to pair a scene button to a light (without having an actual light, just to see if zigbee2mqtt would say "ok, button paired") but it doesn't seem to work.

Does anyone have any information about how to use this wall controller?

Koenkk, 2020-03-11:
@rutgerhendriks initially added support for this device, could you help with this? Does this device still work OK for you in zigbee2mqtt 1.11.0?

closed 2020-05-17 by stale[bot]

rutgerhendriks, 2020-11-07:
Hi, sorry for the late reply, I'm currently running zigbee2mqtt v1.15.0  and that is working fine for the KPD device. I use the scene buttons in home assistant automations just fine

dresende, 2020-11-09:
Thank you for responding. I archived the buttons and used an Aqara 6-button instead, but will look again with a newer zigbee2mqtt version as buttons are always useful 😄

sepo83, 2022-11-13:
I have the same issue here. I can pair my device and zigbee2mqtt shows `linkquality` and `battery` with reasonable values.

However, it does not react to button-actions. When I turn on heardsmen-zigbee debug logging I get the following message when pushing a button (e.g. S2):
```
2022-11-13T09:00:00.174Z zigbee-herdsman:controller:log Received 'zcl' data '{"frame":{"Header":{"frameControl":{"frameType":1,"manufacturerSpecific":false,"direction":0,"disableDefaultResponse":false,"reservedBits":0},"transactionSequenceNumber":127,"manufacturerCode":null,"commandIdentifier":5},"Payload":{"groupid":0,"sceneid":2},"Command":{"ID":5,"parameters":[{"name":"groupid","type":33},{"name":"sceneid","type":32}],"name":"recall"}},"address":62838,"endpoint":1,"linkquality":255,"groupID":15873,"wasBroadcast":true,"destinationEndpoint":53}'

2022-11-13T09:00:00.175Z zigbee-herdsman:controller:log 'zcl' data is from unknown device with address '62838', skipping...
```

Btw, the device works fine with Phoscon.