Koenkk/zigbee2mqtt #3107: Failed to call 'DeviceReceive' 'onZigbeeEvent'

ccorderod, 2020-03-11:
# Bug Report
## What happened
Using Z2M dev, cc1352-P2 firmware dev 20200306 on pi4.

Log is full of:
`error 2020-03-11 19:20:59: Failed to call 'DeviceReceive' 'onZigbeeEvent' (TypeError: Cannot read property 'debounce' of null
    at DeviceReceive.onZigbeeEvent (/opt/zigbee2mqtt/lib/extension/deviceReceive.js:119:82)
    at Controller.callExtensionMethod (/opt/zigbee2mqtt/lib/controller.js:340:44)
    at Controller.onZigbeeEvent (/opt/zigbee2mqtt/lib/controller.js:230:14)
    at Zigbee.emit (events.js:311:20)
    at Controller.<anonymous> (/opt/zigbee2mqtt/lib/zigbee.js:57:52)
    at Controller.emit (events.js:311:20)
    at Controller.<anonymous> (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/controller/controller.js:466:22)
    at Generator.next (<anonymous>)
    at /opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/controller/controller.js:8:71
    at new Promise (<anonymous>))
`

No ZB messages are proccesed by Z2M.

## What did you expect to happen

## How to reproduce it (minimal and precise)
Rebooting pi or disconnecting/reconnecting the controller does not fix the problem
## Debug Info
zigbee2mqtt version: dev - latest commit
CC253X firmware version: 20200306

Koenkk, 2020-03-11:
Could you provide the debug logging of this and your `configuration.yaml`?

To enable debug logging set in `configuration.yaml`:
```yaml
advanced:
  log_level: debug
```

ccorderod, 2020-03-11:
Sure. May I send both to you by PM through discord?

Koenkk, 2020-03-11:
yes

ccorderod, 2020-03-11:
Done. Log, config, devices & groups

Koenkk, 2020-03-11:
In your `configuration.yaml` `device_options: null` is set which is invalid. Either remove it completely from `configuration.yaml` or change it to `device_options: {}`. 

I've added an additional check so Zigbee2mqtt refuses to start now when this is set:
```
> zigbee2mqtt@1.11.0-dev start /Users/Koenkk/Git/zigbee2mqtt
> node index.js

zigbee2mqtt:info  2020-03-11 20:16:55: Logging to console and directory: '/Users/Koenkk/Git/zigbee2mqtt/data/log/2020-03-11.20-16-55' filename: log.txt
zigbee2mqtt:error 2020-03-11 20:16:55: Refusing to start, configuration.yaml is not valid, found the following errors:
zigbee2mqtt:error 2020-03-11 20:16:55:   - device_options should be object
zigbee2mqtt:error 2020-03-11 20:16:55: If you don't know how to solve this, read https://www.zigbee2mqtt.io/configuration/configuration.html
```

Thanks for reporting!

closed 2020-03-11 by Koenkk

LukeHandle, 2020-04-14:
Weirdly also had the same issue (upgrading from 1.8.0).

The doc URL might need an adjustment to https://www.zigbee2mqtt.io/information/configuration.html ?

Also, the doc there shows the following which still seems to throw the error:
```
# Optional: Device specific options
device_options:
  # See 'Device specific configuration' below
```

Deleting the key per your advice works fine though.

Koenkk, 2020-04-14:
Updated the docs, thanks!