Koenkk/zigbee2mqtt #3108: Add new life control devices

Iv711, 2020-03-11:
Please add support for
Life Control water leak sensor: (https://lifecontrol.ru/devices/water-sensor/)
`        zigbeeModel: ['Leak_Sensor'],
        model: 'Leak_Sensor',
        vendor: 'Nexturn',
        description: 'LifeControl water leak switch',
        supports: 'water leak',
        fromZigbee: [fz.nexturn_leak_sensor, fz.legacy_battery],
        toZigbee: [],
    },  `

`    nexturn_leak_sensor: {
        cluster: 'ssIasZone',
        type: 'commandStatusChangeNotification',
        convert: (model, msg, publish, options, meta) => {
            return {water_leak: msg.data.zonestatus === 1};        
        },`

And Life Control door sensor: (https://lifecontrol.ru/devices/door-alarm/)
`    {
        zigbeeModel: ['Door_Sensor'],
        model: 'Door_Sensor',
        vendor: 'Nexturn',
        description: 'LifeControl door sensor',
        supports: 'contact',
        fromZigbee: [fz.nexturn_door_sensor, fz.legacy_battery],
        toZigbee: [],
    },`

`    nexturn_door_sensor: {
        cluster: 'ssIasZone',
        type: 'commandStatusChangeNotification',
        convert: (model, msg, publish, options, meta) => {
            return {contact: msg.data['zonestatus'] === 0};            
        },`

Thx!

Iv711, 2020-03-11:
Oh.. I`m sorry ;)
https://pastebin.com/7ZAby3kZ

Iv711, 2020-03-11:
And... Life Control RGB Led Bulb https://lifecontrol.ru/devices/smart-lamp/
A try this code, and it worked for me

https://pastebin.com/CGT99AdP

True, if you turn off the power to turn on the lamp, then in HA the status will hang - OFF
Although the logs show that when the lamp is turned on, it reports:
Device 'lc_rgb_lamp01' announced itself
And that's it = (

Koenkk, 2020-03-12:
- I've added the devices, please check the latest zigbee2mqtt dev branch in a few hours to check if everything works OK
- To receive state changes when lamps turn on/off on repower use https://www.zigbee2mqtt.io/information/availability.html

closed 2020-03-12 by Koenkk

Iv711, 2020-03-13:
2Koenkk - the door sensor does not display correctly;)
[image]

At the same time, when the sensor is activated, the readings linkquality change, but the readings of the sensor itself do not change.

`"capabilities": null,
                "config_entry_id": "efe0...",
                "device_class": "moisture",
                "device_id": "2ba...",
                "disabled_by": null,
                "entity_id": "binary_sensor.lc_door01_water_leak",
                "icon": null,
                "name": null,
                "original_icon": null,
                "original_name": "lc_door01_water_leak",
                "platform": "mqtt",
                "supported_features": 0,
                "unique_id": "0x...4_water_leak_zigbee2mqtt",
                "unit_of_measurement": null`

Iv711, 2020-03-13:
rgb_lamp - don`t show linkquality.
And - waterleak and door sensor does not send % battery. Moreover, it stopped working after re-binding. When the sensors were connected and there was information about them in the database,% was shown.

Iv711, 2020-03-13:
[image]

Koenkk, 2020-03-14:
- Can you provide the zigbee2mqtt debug log when putting the water leak sensor in AND out of water?

To enable debug logging set in `configuration.yaml`:
```yaml
advanced:
  log_level: debug
```

- Did battery work with your original configuration? (https://github.com/Koenkk/zigbee2mqtt/issues/3108#issue-579494197)
- It's normal for bulbs not the send linkquality because they don't send messages normally. For this check the networkmap.

Iv711, 2020-03-14:
> Did battery work with your original configuration? (#3108 (comment))
Yes
> It's normal for bulbs not the send linkquality because they don't send messages normally. For this check the networkmap.
Its ok)
> Can you provide the zigbee2mqtt debug log when putting the water leak sensor in AND out of water?
Of course I can, but with the leakage sensor everything is fine, it works (except for the battery sensor). The problem is precisely the window / door sensor - it is displayed as a leak sensor, and not as a contact.
Door sensor log:
https://pastebin.com/yiSHm8XG
As far as I can judge the problem in MQTT Discovery message:
https://pastebin.com/t0ahrZt3

Koenkk, 2020-03-15:
Yes it was indeed discovered incorrectly, fixed in latest dev.