Koenkk/zigbee2mqtt #3109: Shenzhen Homa HLC610-Z is reporting as being unsupported

bedfelllow, 2020-03-11:
# Bug Report
## What happened
Shenzhen Homa HLC610-Z is being reported as being unsupported:

> zigbee2mqtt:warn  2020-03-11 20:15:15: Received message from unsupported device with Zigbee model 'HOMA0019'
zigbee2mqtt:warn  2020-03-11 20:15:15: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.

## What did you expect to happen
For it to be supported as it is listed as a support device: https://www.zigbee2mqtt.io/devices/HLC610-Z.html
## How to reproduce it (minimal and precise)
Put zigbee2mqtt into device join mode and plug in Shenzhen Homa HLC610-Z
## Debug Info
zigbee2mqtt version: 1.8.0
CC253X firmware version: 20190608

bedfelllow, 2020-03-11:
A restart of zigbee2mqtt reports:

> zigbee2mqtt:info  2020-03-11 21:03:28: 0x680ae2fffeb18442 (0x680ae2fffeb18442): Not supported (Router)

Seems that is is not being recognised as a Shenzhen Homa HLC610-Z

sjorge, 2020-03-11:
Can you post the database.db entry for that device?

bedfelllow, 2020-03-11:
> Can you post the database.db entry for that device?

Where would I find this?

sjorge, 2020-03-11:
> Where would I find this?
```
grep 80ae2fffeb18442 data/database.db
```

For me it's `/opt/zigbee2mqtt/data/database.db` but the full path is probably different for you.

bedfelllow, 2020-03-11:
> > Where would I find this?
> 
> ```
> grep 80ae2fffeb18442 data/database.db
> ```
> 
> For me it's `/opt/zigbee2mqtt/data/database.db` but the full path is probably different for you.

If I ever work out how to copy from nano and paste into here, I'll let you know :+1:

bedfelllow, 2020-03-11:
`{"id":6,"type":"Router","ieeeAddr":"0x680ae2fffeb18442","nwkAddr":22149,"manufId":34457,"manufName":"ShenZhen_Homa","powerSource":"Unknown","modelId":"HOMA0019","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":257,"inClusterList":[0,3,4,5,6,8,4096,65518],"outClusterList":[25,4096,65518],"clusters":{"genBasic":{"attributes":{"modelId":"HOMA0019","manufacturerName":"ShenZhen_Homa","powerSource":0,"zclVersion":3,"appVersion":1,"stackVersion":3,"hwVersion":2,"dateCode":"20190905","swBuildId":"3.0.0"}}},"binds":[]},"242":{"profId":41440,"epId":242,"devId":102,"inClusterList":[33],"outClusterList":[33],"clusters":{},"binds":[]}},"appVersion":1,"stackVersion":3,"hwVersion":2,"dateCode":"20190905","swBuildId":"3.0.0","zclVersion":3,"interviewCompleted":true,"meta":{}}`

sjorge, 2020-03-11:
I think it is a different version of the device, the one you linked is:
```
    {
        zigbeeModel: ['HOMA1002'],
        model: 'HLC610-Z',
        vendor: 'Shenzhen Homa',
        description: 'Wireless dimmable controller',
        extend: generic.light_onoff_brightness,
    },
```

But according to your database entry for the device the model is: `HOMA0019`

If you feel up to it, you could edit devices.js and try replacing the above with the following:

```
    {
        zigbeeModel: ['HOMA1002', 'HOMA0019'],
        model: 'HLC610-Z',
        vendor: 'Shenzhen Homa',
        description: 'Wireless dimmable controller',
        extend: generic.light_onoff_brightness,
    },
```

bedfelllow, 2020-03-11:
> I think it is a different version of the device, the one you linked is:
>
> ```
>     {
>         zigbeeModel: ['HOMA1002'],
>         model: 'HLC610-Z',
>         vendor: 'Shenzhen Homa',
>         description: [...]

I can't find devices.js

bedfelllow, 2020-03-11:
Thanks @sjorge, I'll give it a go tomorrow

bedfelllow, 2020-03-12:
@sjorge, 

>     {
>         zigbeeModel: ['HOMA1002', 'HOMA0019'],
>         model: 'HLC610-Z',
>         vendor: 'Shenzhen Homa',
>         description: 'Wireless dimmable controller',
>         extend: generic.light_onoff_brightness,
>     },

Thsi worked, thanks. I'll look at trying to get it added to the project.

Koenkk, 2020-03-12:
@bedfelllow added it, will be supported in the latest zigbee2mqtt dev branch in a few hours.

@sjorge thanks for helping out!

closed 2020-03-12 by Koenkk