Koenkk/zigbee2mqtt #3111: WXCJKG13LM badly paired/configured

lolorc, 2020-03-12:
zigbee2mqtt version 1.11.0-dev (commit #8e9729c)

I'm unable to use any of my WXCJKG13LM devices on my main z2m instance (cc2652 on rpi3)
The devices are paired and configured but the devices don't send any message (report attributes) to the coordinator.
Using the same z2m version on my notebook with a cc2531 the devices work correctly

looking at the database.db I can see the entry for the same opple sensor is not the same:
cc2652:
```
{
   "zclVersion" : 3,
   "swBuildId" : "2019\u0000www.",
   "interviewCompleted" : true,
   "epList" : [
      1,
      2,
      3,
      4,
      5,
      6
   ],
   "powerSource" : "Battery",
   "id" : 60,
   "stackVersion" : 2,
   "dateCode" : "20190730",
   "ieeeAddr" : "0x04cf8cdf02134657",
   "nwkAddr" : 53361,
   "endpoints" : {
      "4" : {
         "devId" : 259,
         "epId" : 4,
         "profId" : 260,
         "binds" : [],
         "clusters" : {},
         "outClusterList" : [
            6
         ],
         "inClusterList" : [
            18,
            3
         ]
      },
      "6" : {
         "epId" : 6,
         "devId" : 259,
         "profId" : 260,
         "outClusterList" : [
            6
         ],
         "binds" : [],
         "clusters" : {},
         "inClusterList" : [
            18,
            3
         ]
      },
      "3" : {
         "devId" : 259,
         "profId" : 260,
         "epId" : 3,
         "binds" : [],
         "inClusterList" : [
            18,
            3
         ],
         "clusters" : {},
         "outClusterList" : [
            6
         ]
      },
      "2" : {
         "epId" : 2,
         "devId" : 259,
         "profId" : 260,
         "clusters" : {},
         "binds" : [],
         "outClusterList" : [
            6,
            3
         ],
         "inClusterList" : [
            3
         ]
      },
      "1" : {
         "devId" : 261,
         "epId" : 1,
         "profId" : 260,
         "clusters" : {
            "genBasic" : {
               "attributes" : {
                  "dateCode" : "20190730",
                  "manufacturerName" : "LUMI",
                  "appVersion" : 17,
                  "modelId" : "lumi.remote.b686opcn01",
                  "hwVersion" : 1,
                  "swBuildId" : "2019\u0000www.",
                  "zclVersion" : 3,
                  "powerSource" : 3,
                  "stackVersion" : 2
               }
            }
         },
         "binds" : [],
         "inClusterList" : [
            0,
            3,
            1
         ],
         "outClusterList" : [
            3,
            6,
            8,
            768
         ]
      },
      "5" : {
         "epId" : 5,
         "devId" : 259,
         "profId" : 260,
         "binds" : [],
         "clusters" : {},
         "outClusterList" : [
            6
         ],
         "inClusterList" : [
            18,
            3
         ]
      }
   },
   "manufId" : 4447,
   "meta" : {
      "configured" : 1
   },
   "appVersion" : 17,
   "type" : "EndDevice",
   "modelId" : "lumi.remote.b686opcn01",
   "hwVersion" : 1,
   "lastSeen" : 1583826502434,
   "manufName" : "LUMI"
}
```

cc2531:
```
{
   "lastSeen" : 1583826185502,
   "epList" : [
      1,
      2,
      3,
      4,
      5,
      6
   ],
   "nwkAddr" : 18886,
   "type" : "EndDevice",
   "meta" : {
      "configured" : 1
   },
   "interviewCompleted" : true,
   "endpoints" : {
      "5" : {
         "epId" : 5,
         "clusters" : {
            "genMultistateInput" : {
               "attributes" : {
                  "presentValue" : 1
               }
            }
         },
         "outClusterList" : [],
         "inClusterList" : [],
         "binds" : []
      },
      "1" : {
         "binds" : [],
         "clusters" : {
            "genBasic" : {
               "attributes" : {
                  "manufacturerName" : "LUMI",
                  "modelId" : "lumi.remote.b686opcn01"
               }
            },
            "aqaraOpple" : {
               "attributes" : {
                  "247" : {
                     "data" : [
                        1,
                        33,
                        249,
                        11,
                        3,
                        40,
                        25,
                        4,
                        33,
                        168,
                        1,
                        5,
                        33,
                        94,
                        1,
                        6,
                        36,
                        2,
                        0,
                        0,
                        0,
                        0,
                        8,
                        33,
                        17,
                        1,
                        10,
                        33,
                        0,
                        0,
                        12,
                        32,
                        1,
                        100,
                        16,
                        0
                     ],
                     "type" : "Buffer"
                  }
               }
            },
            "genMultistateInput" : {
               "attributes" : {
                  "presentValue" : 1
               }
            }
         },
         "outClusterList" : [
            3,
            6,
            8,
            768
         ],
         "devId" : 261,
         "epId" : 1,
         "profId" : 260,
         "inClusterList" : [
            0,
            3,
            1
         ]
      },
      "4" : {
         "outClusterList" : [],
         "clusters" : {
            "genMultistateInput" : {
               "attributes" : {
                  "presentValue" : 1
               }
            }
         },
         "epId" : 4,
         "inClusterList" : [],
         "binds" : []
      },
      "2" : {
         "epId" : 2,
         "outClusterList" : [],
         "clusters" : {},
         "inClusterList" : [],
         "binds" : []
      },
      "6" : {
         "binds" : [],
         "inClusterList" : [],
         "epId" : 6,
         "outClusterList" : [],
         "clusters" : {}
      },
      "3" : {
         "binds" : [],
         "epId" : 3,
         "outClusterList" : [],
         "clusters" : {
            "genMultistateInput" : {
               "attributes" : {
                  "presentValue" : 1
               }
            }
         },
         "inClusterList" : []
      }
   },
   "manufName" : "LUMI",
   "ieeeAddr" : "0x04cf8cdf02134657",
   "manufId" : 4151,
   "id" : 8,
   "powerSource" : "Battery",
   "modelId" : "lumi.remote.b686opcn01"
}
```
 z2m commit #407d329 (cc2652), z2m commit #4d54e50 (cc2531), so same herdsman* 
manufId (xioami on cc2652, eurotronic on cc2531) is not the same and endpoints are not the same

lolorc, 2020-03-12:
logs, we can see the binding process is not the same.
cc2652:
```
Mar 10 17:11:19   zigbee-herdsman:controller:log Device '0x04cf8cdf02134657' joined +8s
Mar 10 17:11:19   zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +7ms
Mar 10 17:11:19   zigbee-herdsman:controller:log Device '0x04cf8cdf02134657' accepted by handler +3ms
Mar 10 17:11:19   zigbee-herdsman:controller:log New device '0x04cf8cdf02134657' joined +1ms
Mar 10 17:11:19   zigbee-herdsman:controller:log Creating device '0x04cf8cdf02134657' +0ms
Mar 10 17:11:19 zigbee2mqtt:info  2020-03-10 17:11:19: Device '0x04cf8cdf02134657' joined
Mar 10 17:11:19 zigbee2mqtt:info  2020-03-10 17:11:19: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_connected","message":{"friendly_name":"0x04cf8cdf02134657"}}'
Mar 10 17:11:19   zigbee-herdsman:controller:log Interview '0x04cf8cdf02134657' start +28ms
Mar 10 17:11:19 zigbee2mqtt:info  2020-03-10 17:11:19: Starting interview of '0x04cf8cdf02134657'
Mar 10 17:11:19 zigbee2mqtt:info  2020-03-10 17:11:19: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_started","meta":{"friendly_name":"0x04cf8cdf02134657"}}'
Mar 10 17:11:19   zigbee-herdsman:controller:device Interview - start device '0x04cf8cdf02134657' +0ms
Mar 10 17:11:19   zigbee-herdsman:adapter:zStack:znp:SREQ --> ZDO - nodeDescReq - {"dstaddr":25284,"nwkaddrofinterest":25284} +9s
Mar 10 17:11:19   zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,4,37,2,196,98,196,98,35] +9s
Mar 10 17:11:19   zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,101,2,0,102] +58ms
Mar 10 17:11:19   zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,101,2,0,102] +0ms
Mar 10 17:11:19   zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 5 - 2 - [0] - 102 +1ms
Mar 10 17:11:19   zigbee-herdsman:adapter:zStack:znp:SRSP <-- ZDO - nodeDescReq - {"status":0} +9s
Mar 10 17:11:19   zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +1ms
Mar 10 17:11:19   zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,13,69,193,196,98,196,98] +140ms
Mar 10 17:11:19   zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,13,69,193,196,98,196,98] +1ms
Mar 10 17:11:19   zigbee-herdsman:adapter:zStack:unpi:parser <-- [189,179,119,60,223,140,207,4,132,208] +0ms
Mar 10 17:11:19   zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,13,69,193,196,98,196,98,189,179,119,60,223,140,207,4,132,208] +1ms
Mar 10 17:11:19   zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 13 - 2 - 5 - 193 - [196,98,196,98,189,179,119,60,223,140,207,4,132] - 208 +0ms
Mar 10 17:11:19   zigbee-herdsman:adapter:zStack:znp:AREQ <-- ZDO - endDeviceAnnceInd - {"srcaddr":25284,"nwkaddr":25284,"ieeeaddr":"0x04cf8cdf02134657","capabilities":132} +208ms
Mar 10 17:11:19   zigbee-herdsman:controller:log Device announce '0x04cf8cdf02134657' +173ms
Mar 10 17:11:19 zigbee2mqtt:debug 2020-03-10 17:11:19: Device '0x04cf8cdf02134657' announced itself
Mar 10 17:11:19   zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +4ms
Mar 10 17:11:20   zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,18,69,130,196,98,0,196] +387ms
Mar 10 17:11:20   zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,18,69,130,196,98,0,196] +1ms
Mar 10 17:11:20   zigbee-herdsman:adapter:zStack:unpi:parser <-- [98,2,64,132,95,17,127,100] +0ms
Mar 10 17:11:20   zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,18,69,130,196,98,0,196,98,2,64,132,95,17,127,100] +1ms
Mar 10 17:11:20   zigbee-herdsman:adapter:zStack:unpi:parser <-- [0,0,44,100,0,0,14] +1ms
Mar 10 17:11:20   zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,18,69,130,196,98,0,196,98,2,64,132,95,17,127,100,0,0,44,100,0,0,14] +0ms
Mar 10 17:11:20   zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 18 - 2 - 5 - 130 - [196,98,0,196,98,2,64,132,95,17,127,100,0,0,44,100,0,0] - 14 +0ms
Mar 10 17:11:20   zigbee-herdsman:adapter:zStack:znp:AREQ <-- ZDO - nodeDescRsp - {"srcaddr":25284,"status":0,"nwkaddr":25284,"logicaltype_cmplxdescavai_userdescavai":2,"apsflags_freqband":64,"maccapflags":132,"manufacturercode":4447,"maxbuffersize":127,"maxintransfersize":100,"servermask":11264,"maxouttransfersize":100,"descriptorcap":0} +394ms
Mar 10 17:11:20   zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +1ms
Mar 10 17:11:20   zigbee-herdsman:controller:device Interview - got node descriptor for device '0x04cf8cdf02134657' +567ms
Mar 10 17:11:20   zigbee-herdsman:adapter:zStack:znp:SREQ --> ZDO - activeEpReq - {"dstaddr":25284,"nwkaddrofinterest":25284} +558ms
Mar 10 17:11:20   zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,4,37,5,196,98,196,98,36] +556ms
Mar 10 17:11:20   zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,101,5,0,97] +15ms
Mar 10 17:11:20   zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,101,5,0,97] +1ms
Mar 10 17:11:20   zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 5 - 5 - [0] - 97 +0ms
Mar 10 17:11:20   zigbee-herdsman:adapter:zStack:znp:SRSP <-- ZDO - activeEpReq - {"status":0} +554ms
Mar 10 17:11:20   zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +0ms
Mar 10 17:11:20   zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,12,69,133,196,98,0,196] +362ms
Mar 10 17:11:20   zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,12,69,133,196,98,0,196] +1ms
[... cut ...]
Mar 10 17:11:22   zigbee-herdsman:controller:log Received 'zcl' data '{"frame":{"Header":{"frameControl":{"frameType":0,"manufacturerSpecific":true,"direction":1,"disableDefaultResponse":true},"transactionSequenceNumber":7,"manufacturerCode":4447,"commandIdentifier":4},"Payload":[{"status":0}]},"address":25284,"endpoint":1,"linkquality":123,"groupID":0}' +177ms
Mar 10 17:11:22   zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +6ms
Mar 10 17:11:22 zigbee2mqtt:info  2020-03-10 17:11:22: Successfully configured '0x04cf8cdf02134657'
```

lolorc, 2020-03-12:
on the cc2530 list of active endpoints is only 1
it then fails to read endpoint 1:
```
2020-03-10T16:16:56.651Z zigbee-herdsman:controller:device Failed to read attributes from endpoint '1' (Error: Read 0x04cf8cdf02134657/1 genBasic(["modelId","manufacturerName"], {"timeout":6000,"manufacturerCode":null,"disableDefaultResponse":true}) failed (Error: Timeout - 16166 - 1 - 2 - 0 - 1 after 6000ms))
2020-03-10T16:16:56.651Z zigbee-herdsman:controller:device Interview procedure failed but got modelID matching 'lumi..*', assuming interview succeeded
2020-03-10T16:16:56.653Z zigbee-herdsman:controller:device Interview - completed for device '0x04cf8cdf02134657' because of quirks ('Error: Interview failed because could not read attributes ('0x04cf8cdf02134657')')
2020-03-10T16:16:56.654Z zigbee-herdsman:controller:log Succesfully interviewed '0x04cf8cdf02134657'
```
the other endpoints are created when the device communicates, ex:
```
2020-03-10T16:17:01.443Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,28,68,129,0,0,18,0,38,63,5,1,0,118,0,207,15,3,0,0,8,24,0,10,85,0,33,1,0,38,63,29,8]
2020-03-10T16:17:01.444Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,28,68,129,0,0,18,0,38,63,5,1,0,118,0,207,15,3,0,0,8,24,0,10,85,0,33,1,0,38,63,29,8]
2020-03-10T16:17:01.444Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 28 - 2 - 4 - 129 - [0,0,18,0,38,63,5,1,0,118,0,207,15,3,0,0,8,24,0,10,85,0,33,1,0,38,63,29] - 8
2020-03-10T16:17:01.444Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - incomingMsg - {"groupid":0,"clusterid":18,"srcaddr":16166,"srcendpoint":5,"dstendpoint":1,"wasbroadcast":0,"linkquality":118,"securityuse":0,"timestamp":200655,"transseqnumber":0,"len":8,"data":{"type":"Buffer","data":[24,0,10,85,0,33,1,0]}}
2020-03-10T16:17:01.445Z zigbee-herdsman:controller:log Received 'zcl' data '{"frame":{"Header":{"frameControl":{"frameType":0,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":true},"transactionSequenceNumber":0,"manufacturerCode":null,"commandIdentifier":10},"Payload":[{"attrId":85,"dataType":33,"attrData":1}]},"address":16166,"endpoint":5,"linkquality":118,"groupID":0}'
2020-03-10T16:17:01.445Z zigbee-herdsman:controller:log 'zcl' data is from unknown endpoint '5' from device with network address '16166', creating it...
2020-03-10T16:17:01.447Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
zigbee2mqtt:debug 2020-03-10 17:17:01: Received Zigbee message from '0x04cf8cdf02134657', type 'attributeReport', cluster 'genMultistateInput', data '{"presentValue":1}' from endpoint 5 with groupID 0
zigbee2mqtt:info  2020-03-10 17:17:01: MQTT publish: topic 'z2m_shiva/0x04cf8cdf02134657', payload '{"action":"button_5_single","linkquality":118}'
```

```
zigbee2mqtt:info  2020-03-10 17:16:29: Device '0x04cf8cdf02134657' joined
zigbee2mqtt:info  2020-03-10 17:16:29: MQTT publish: topic 'z2m_shiva/bridge/log', payload '{"type":"device_connected","message":{"friendly_name":"0x04cf8cdf02134657"}}'
2020-03-10T16:16:29.461Z zigbee-herdsman:controller:log Interview '0x04cf8cdf02134657' start
zigbee2mqtt:info  2020-03-10 17:16:29: Starting interview of '0x04cf8cdf02134657'
zigbee2mqtt:info  2020-03-10 17:16:29: MQTT publish: topic 'z2m_shiva/bridge/log', payload '{"type":"pairing","message":"interview_started","meta":{"friendly_name":"0x04cf8cdf02134657"}}'
2020-03-10T16:16:29.462Z zigbee-herdsman:controller:device Interview - start device '0x04cf8cdf02134657'
2020-03-10T16:16:29.463Z zigbee-herdsman:adapter:zStack:znp:SREQ --> ZDO - nodeDescReq - {"dstaddr":16166,"nwkaddrofinterest":16166}
2020-03-10T16:16:29.463Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,4,37,2,38,63,38,63,35]
2020-03-10T16:16:30.068Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,101,2,0,102]
2020-03-10T16:16:30.069Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,101,2,0,102]
2020-03-10T16:16:30.069Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 5 - 2 - [0] - 102
2020-03-10T16:16:30.069Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- ZDO - nodeDescReq - {"status":0}
2020-03-10T16:16:30.069Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2020-03-10T16:16:30.077Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,13,69,193,38,63,38,63,189,179,119,60,223,140,207,4,132,208]
2020-03-10T16:16:30.077Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,13,69,193,38,63,38,63,189,179,119,60,223,140,207,4,132,208]
2020-03-10T16:16:30.077Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 13 - 2 - 5 - 193 - [38,63,38,63,189,179,119,60,223,140,207,4,132] - 208
2020-03-10T16:16:30.077Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- ZDO - endDeviceAnnceInd - {"srcaddr":16166,"nwkaddr":16166,"ieeeaddr":"0x04cf8cdf02134657","capabilities":132}
2020-03-10T16:16:30.078Z zigbee-herdsman:controller:log Device announce '0x04cf8cdf02134657'
zigbee2mqtt:debug 2020-03-10 17:16:30: Device '0x04cf8cdf02134657' announced itself
2020-03-10T16:16:30.079Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2020-03-10T16:16:32.343Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,13,69,193,38,63,38,63,189,179,119,60,223,140,207,4,132,208]
2020-03-10T16:16:32.343Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,13,69,193,38,63,38,63,189,179,119,60,223,140,207,4,132,208]
[... cut ...] ata":{"type":"Buffer","data":[28,95,17,3,4,0]}}
2020-03-10T16:16:57.616Z zigbee-herdsman:controller:log Received 'zcl' data '{"frame":{"Header":{"frameControl":{"frameType":0,"manufacturerSpecific":true,"direction":1,"disableDefaultResponse":true},"transactionSequenceNumber":3,"manufacturerCode":4447,"commandIdentifier":4},"Payload":[{"status":0}]},"address":16166,"endpoint":1,"linkquality":149,"groupID":0}'
2020-03-10T16:16:57.616Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
```

lolorc, 2020-03-12:
the thing I don't understand is why the device doesn't send any "report attributes", it sends ieee 802.15.4 data request (0x04) but they're not related to any device action as far as I recall.

lolorc, 2020-03-13:
nope, nothing, I'll try to rebind on cc2652 with the database entry from cc2531, but there's definitely something wrong here

lolorc, 2020-03-14:
@Koenkk any idea what could be wrong ?

Koenkk, 2020-03-15:
Looks that there are connectivity issues, please try pairing the device close to the coordinator.

lolorc, 2020-03-15:
@Koenkk as you could have guessed, that's the first thing I've done, pairing 5cm from the coordinator, pairing 5cm from the coordinator with all routers turned off.
If you confirm I won't loose my 40+ pairings by backuping the database.db, coordinator_backup.json and possibly state.json, then I will try again without any paired device. (that was the state the cc2531 was in when I tried)

it should be noted the cc2531 pairing doesn't work correctly, it only get one endpoint, wrong manufacturer id, but the device works as expected, it reports to the coordinator.

with the cc2652, the pairing looks ok, but the device stop sending any report after the "configuring"

I can't tell if the issue with my cc2652 is related to cc2652 irself or with my actual network/setup.

Koenkk, 2020-03-16:
Could you sniff the traffic when pairing the device with the CC2652R and controlling it?

lolorc, 2020-03-22:
@Koenkk in what format would like it ? public text packet dump with private data removed/replaced or the pcap I would share only share with you

Koenkk, 2020-03-22:
pcap

lolorc, 2020-03-22:
@Koenkk would you mind sharing a mail address ? ;-)

Koenkk, 2020-03-23:
@lolorc koenkanters94@gmail.com

lolorc, 2020-03-26:
fixed by upgrading cc2652r firmware (20200312)

Koenkk, 2020-03-26:
Great that this firmware fixed.

I've also fixed the manufacturerID issue you mentioned in the mail.

With the latest dev could you repair the device and confirm that this is fixed?

lolorc, 2020-03-27:
@Koenkk yes, confirmed, the manufId is now also correct, closing it now, Thanks a lot !

closed 2020-03-27 by lolorc