Koenkk/zigbee2mqtt #3112: Zigbee network settles in poor configuration

vdkeybus, 2020-03-12:
After I re-flashed my CC2531, it re-established contact with the previously connected network. 

It is located at the edge of the network. Previously, it had at least 2 lamps to connect to the network. Turning either one off did not make the network unreachable.

However, now it depends on only 1 lamp. If that one is shut down, the network becomes completely unreachabe (error 205).

Is there a way to force re-configuration or to provide hints on how to mesh the network ?

svrooij, 2020-03-12:
I would just buy some IKEA lights and screw them in a socket without a wired switch. Then use the dimmer in the same package to control the light. I have 30+ devices, a lot of battery powered, but also a lot of lights. My network is really stable.

Ikea also has some range extenders, they act as a router when plugged in to a socket (and provide an addition usb connection for charging some usb device as a bonus)

vdkeybus, 2020-03-12:
How would putting the lights there improve the network ? Would you register them or is the idea to not register them and just put a couple of them here and there and hope they'll start relaying messages ?

My problem is that the physical links between the various nodes are not the best. For some reason the zigbee2mqtt map isn't working anymore so I cannot check lqi.

It would be great if the network could somehow reorganize itself.

I think the initial problem comes from me reflashing and testing the CC2531 briefly in a different room.

Thanks,

svrooij, 2020-03-12:
You can have the lights join the zigbee network, all powered devices act as a router in the mesh. By adding more mains powered devices your network will get better.

The devices will have to be joined for it to work.

vdkeybus, 2020-03-13:
>  By adding more mains powered devices your network will get better.
>
And that is exactly what *not* seems to be happening.

I seem to have found a solution by removing power to all lamps, and
gradually turn them on one at a time (not re-pairing). It's still non
perfect though.

closed 2020-05-20 by stale[bot]