Koenkk/zigbee2mqtt #3113: Siterwell GS361 smart thermostatic radiator valve. Which version to better to buy for z2m?

BTopbas, 2020-03-12:
Hello friends; 
I want to buy cheap good designed smart trv. So i look at alibaba and some Chinese products. So i founded red-dot awarded cheap trv.  I contact seller and ask some questions. Seller says 2 different ZigBee versions. One of TuYa ZigBee and other ZigBee 3.0 version. TuYa ZigBee need's TuYa gateway and other version using the company's own(siterwell) gateway and own family well application. 
Seller says siterwell member of zigbee alliance. 
Here is reddot link: https://www.red-dot.org/project/siterwell-gs361-39193/
Here is product link: https://www.china-siter.com/product/194.html

I want to buy and trying to add support.
Any recommendations to which version will be easy to adding support? TuYa ZigBee or ZigBee 3.0
I am thinking to will buy ZigBee 3.0 version. 

And one more question. If i will buy ZigBee 3.0 version without gateway. Can i add support without sniffing communication? Or i need to buy gateway and sniff communication?

sjorge, 2020-03-12:
I think the ZB3 would probably have a higher chance of working without odd issues.

If it follows the hvacThermostat cluster it should not be to hard to get going, but you won’t really know before you get the device.

I wonder if they are easy to get in Europe, I am not super happy with my Schneider or Eurotronic TRVs, the later being the better if the two.

netweaver1970, 2020-03-13:
I have 5 trv's, Tuya gateway versions (without the gateway) coming to Netherlands, arriving today or tomorrow if not corona impacted.

I’m happy to get one over to the Eindhoven area (I work in Veghel so can easily drop off down there) for Koen to try to get it supported. 

They look pretty neat, let’s hope they can be made to work. 

@Kkanters, would you be interested to get one in your hands? PM me for direct contact. Btw I work also at one of your previous employers 😁

sjorge, 2020-03-13:
@netweaver1970 where did you order from? I might go for one of the ZB3 ones to test and play with.

netweaver1970, 2020-03-13:
Straight from factory I think : https://siterlink.en.alibaba.com/?spm=a2756.trade-order-list.0.0.38b976e9thNNyI&tracelog=from_orderlist_company

21USD/piece + 30 USD shipping using DHL Express. Delivery today.

netweaver1970, 2020-03-13:
@sjorge, if I receive them this afternoon you could pick one up this evening (or over the weekend)  in Merelbeke to play with too ... Anything to get them supported will help me (and the whole community too, I guess)

sjorge, 2020-03-13:
I send them an request for one of the ZB 3.0 ones, lets see if they reply.

@netweaver1970 if you get one, a good start is probably trying to pair it and grabbing the database.db entry.

netweaver1970, 2020-03-13:
The DHL delivery was delayed. Will now arrive only on Monday, in the office. But i have now idea what working arrangements will be on Monday due to COVID-19. To be confirmed over the weekend. So don't know when I can pick-up and test. Will keep all updated.

BTopbas, 2020-03-23:
@netweaver1970 Any news?

sjorge, 2020-03-23:
I have not gotten a reply yet, so I don't think they are interested in selling a single generic ZB 3.0 unit.

netweaver1970, 2020-03-23:
I could only pick up the 5 thermostats this morning from Veghel (normally my office). 

And my HA install is in my Belgium house, where I can’t go back to, at least if I still want see my Rotterdam gf, since I’ll not be allowed to travel back to NL. Prisoner here...

So unless Belgium relaxes the travel rules, I can’t go back to try things out. I’ll try to set up an alternative HA for testing this in Rotterdam. I do have a cc2530 spare here but no programmer/debugger. Because that one is my Belgium house too. So I ordered a new one. 

Big mess. But we’re still healthy, let’s keep it that way. 
To be continued. Some day. Patience is a virtue currently

netweaver1970, 2020-04-01:
Good news, support for the (tuya) version is on the way : https://github.com/Koenkk/zigbee2mqtt/pull/3259#pullrequestreview-385669478

netweaver1970, 2020-04-02:
> I'm really curious how it requests heat from the boiler

It doesn’t, it works as a normal but remote controllable TRV. No feedback
to the boiler, you’ll have to automate that yourself somehow using the
specific boiler controls. Look in the PR, it has the possible commands in
it. Simple but effective for the base use case.

If you want TRVs controlling the boiler (almost) directly, look at OpenTRV.
But that’s whole different solution.

sjorge, 2020-04-02:
@netweaver1970 does this look like the same device?
https://www.amazon.de/-/en/dp/B08357D6LJ/ref=sr_1_2?dchild=1&keywords=zigbee+trv+tuya&qid=1585865690&sr=8-2

I never heard back from siterwell about ordering a few for testing.

netweaver1970, 2020-04-03:
Is identical.
Read the full experience story here:

https://community.home-assistant.io/t/i-am-totally-noob-with-zigbee-and-tuya-and-i-need-advice-for-smart-thermostatic-radiator-valves/163607/11

kirovilya, 2020-04-11:
Please check with my revisions.

netweaver1970, 2020-04-11:
Thanks for your extensive Tuya trv support update. Will existing trv's need removing and re-pairing to take care of new functionality? Anywhere a description of all extra new option?

kirovilya, 2020-04-12:
> Will existing trv's need removing and re-pairing

no, just use :)
but PR not merged yet...

Koenkk, 2020-04-12:
@kirovilya merged one minute ago :)

sjorge, 2020-05-05:
I managed to pick up one of the white label ones from essentials... but it came without any instructions.

How do I get it into pairing mode?

sjorge, 2020-05-05:
I think I got it to pair, will try to write up some instructions once I figure out what exactly I did.

As mentioned its ta white label essentials one but it looks like they are not 100% compatible?

```
mosquitto_pub -h localhost -p 1883 -t zigbee2mqtt/0x086bd7fffed12adc/set/window_detection -m 'OFF'
```
Resulted in:
```
zigbee-herdsman-converters:siterwell_gs361: NOT RECOGNIZED DP #1041 with data {"type":"Buffer","data":[0]}
```

But the command did seem to work as the window icon disappeared.

Ciubas, 2020-05-16:
How can I update file devices.js ?

[image]
[image]

Ciubas, 2020-05-18:
I find the problem - I dont know how, but the repo link in Addons deleted, so I dinn't have aupdates! Now its ok :D

And its working! :D

New problem is the ON/AUTO/HEAT working well to you? My work only in auto, HEAT dont work - go to OFF after few seconds....

sjorge, 2020-05-18:
I was using one of the essentials ones, I ended up returning it.
- only off and auto worked for me
- window detect could be set
- valve detect didn't seem to do anything, no error though
- position didn't get update (shame as I need that for my homekit integration)

It was very very slow to respond compared to my eurotronic one.
Also the display was very hard to read and the controls didn't work very well.

netweaver1970, 2020-05-18:
Some feedback here too. I have 5 of the Siterwell ones.

> I was using one of the essentials ones, I ended up returning it.
> 
> * only off and auto worked for me
> * window detect could be set
> * valve detect didn't seem to do anything, no error though
> * position didn't get update (shame as I need that for my homekit integration)
> 
> It was very very slow to respond compared to my eurotronic one.
> Also the display was very hard to read and the controls didn't work very well.
> 
> I think it's up to the vendor to select what features they enable.

The Siterwell one supports these 3 heating modes from the factory:
'Automatic', 'manual' and 'anti-freeze'. Maybe the mapping towards HomeAssistant modes is not 100% corresponding ? 

ValveDetection is a more advanced feature, useful or not I don't know. I also never saw anything with or without. But without knowing what it should do it's hard to test for it.

Position update/feedback I didn't test, I'm using Google Home. Also having some issues there but I think that's more in the HA ->Google Home integration together with the Valve->HA integration. Hoping to find some time to focus on that with the Z2M TRV support dev + HA Google Home integration dev. to try to get it better. Maybe it's just my (mis)understanding how it should work.
I'll try the feedback from TRV to HA out next time I'm back in that house.

The response speed was very acceptable (1-10s) for me, as heating is not needed to be immediate (eg. as lights should be on immediately after movement detection ). Thermal inertia is anyway way higher than the small time to wait for the valve to respond.

What I think is more cumbersome, is the valve operating sound. When the valve is operating (changing position), it makes a chirping sound, seemingly amplified by a resonating pipe/radiator system. And in a quiet room it's rather obvious. In the beginning I was really surprised/annoyed by it, now I got used to it. The same TRV valve with a manual/conventional wax knob is silent, so it's not the TRV as such. The internal motor must be sending vibrations to the pipe/radiator system.

Screen is a bit dim, but I seldom do local adjustments.

I was happily surprised by the accuracy of the temperature measurements of the valve, I have a longish style living room with full glass on the north'ish side (blame previous owners) and I can see, using the valve temp. sensors and an in-the-middle Xiaomi temp/humidy sensor clearly the temp gradient I was feeling/expecting. That gave me a good feeling, at least for accuracy.
Both bathroom sensors (valve + Xiaomi Temp/humidy) fully agree too, 

For the ones wanting to see the UserManual (not that exciting but at least original) :
http://www.smart9.com.cn/wp-content/uploads/2019/10/Siterwell-ZigBee-Radiator-Thermostat-GS361A-H04-User-Manual.pdf

In all honesty, as I could only install it when Belgium started to get hotter, I haven't seen the behavior over the winter. But in the few more chill days, it all worked fine. 

Next step is a true multi-zone heating system with scheduled, full predictive PID control/room with extension towards Bulex boiler eBus control, But all of that combined doesn't exist yet in HA. So it has to be pieced together. My MVP will be a lot less, I fear.
And I have too many other ongoing projects to spend time on it with summer coming up :)

my 5 cents.

sjorge, 2020-05-18:
Thanks for the manually, it looks the same as the german one I scanned... but I can read this one actually. I think the essentials one also had the antifreeze, I could just not make heads or tails from it.

And yes, I agree the moter is rather noisy, compared to the eurotronic and eberly units I have for testing.

netweaver1970, 2020-05-18:
> I'm going to try and grab the Viessman also listed if I can find it somewhere. It looks like it uses a different LCD style but overal still the same.

You mean this one: https://www.badexo.de/viessmann-vicare-heizkoerper-thermostat ?

sjorge, 2020-05-18:
Yep

closed 2020-07-25 by stale[bot]

messismore, 2020-09-12:
Did you end up buying the Viessmann TRV @sjorge ? If so, do you mind to share your experience with it? Really unhappy with the Wiser I bought…