Koenkk/zigbee2mqtt #3114: xiaomi vacuum cleaner gen 1

Riseryn, 2020-03-12:
Hello,
I have installed zigbee2mqtt and mosquitto.
I don't use home assistant.
I have an Ikea tradfri outlet that i can succesfully manage with zigbee2mqtt.
I also have a xiaomi rockrobo gen 1 but i can't find a way to use it with zigbee2mqtt.
Does someone can help me ?
Thanks
# Bug Report
## What happened
Impossible to connect to zigbee2mqtt
## What did you expect to happen
control the vacuum by mqtt
## How to reproduce it (minimal and precise)

## Debug Info
zigbee2mqtt version: latest
CC253X firmware version: don't know how to find it

MaxRei-dev, 2020-03-12:
Hi,
the Vacuum is WLAN based. You could use Valetudo or for example nodered to control it via mqtt.
Greets max

Riseryn, 2020-03-12:
Hi 
I have valetudo re installed. But i can't find a clue about how to use mqtt with it.
Do you know where i can find informations about it, my google search does'nt  give me any result i can use( or understand)
thanks

Koenkk, 2020-03-13:
Closing as this is not related to Zigbee2mqtt

closed 2020-03-13 by Koenkk