Koenkk/zigbee2mqtt #3115: reporting setup only when permit_join is true?

middelink, 2020-03-13:
# Bug Report
## What happened
Recently one of my Tradfri bulbs got disconnected from the power, after turning power back on, it became part of the network and HA was able to switch the light. However as the light is also in a group with the remote control, I noticed it was not reporting status changes as the other lights in the group did.

Powering the light shows it announces itself to the network, but nothing much happens
```
zigbee2mqtt:debug 2020-03-12 22:49:21: Device 'TRADFRI Woonkamer Kastlamp' announced itself
<silence till other devices report in>
```

However, when `permit_join` is set, it works:
```
zigbee2mqtt:debug 2020-03-13 00:29:28: Setup reporting for '0x000b57fffe488dc4' - 1 - genOnOff
zigbee2mqtt:info  2020-03-13 00:29:29: Successfully setup reporting for '0x000b57fffe488dc4' - 1 - genOnOff
zigbee2mqtt:debug 2020-03-13 00:29:29: Setup reporting for '0x000b57fffe488dc4' - 1 - genLevelCtrl
zigbee2mqtt:info  2020-03-13 00:29:29: Successfully setup reporting for '0x000b57fffe488dc4' - 1 - genLevelCtrl
zigbee2mqtt:debug 2020-03-13 00:29:29: Setup reporting for '0x000b57fffe488dc4' - 1 - lightingColorCtrl
zigbee2mqtt:info  2020-03-13 00:29:29: Successfully setup reporting for '0x000b57fffe488dc4' - 1 - lightingColorCtrl
```
I expect this to be a programming edge case as normally reporting is setup when the device is paired, therefore permit_join is true. However, as reporting does not seem to be persistent on TRADFRI (or at least this model) it needs to be re-setup when the bulb returns to the network.

## What did you expect to happen
Reporting to be setup on a returning bulb, even when permit_join is false.

## How to reproduce it (minimal and precise)
Power off the LED1546G12 for some time so it loses its settings
Power back on

## Debug Info
zigbee2mqtt version: 1.11.0
CC253X firmware version: 20190608

sjorge, 2020-03-13:
Before the the latest dev I also had a lot of routing issues to my trådfri bulbs.... this also sometimes resulted in them not being reachable after the announce... so that might be happening here too...

Can you try with current dev.

middelink, 2020-03-14:
This has nothing to do with routing issues, if it was, it would not have immediately worked when I enabled permit_join. Also, if there were routing errors, I would have gotten "no ack" messages in the log.

IMO This is a simple case of a software issue where too much code is put in the pairing code instead of the reconfiguration phase.

h3x4d3c1m4l, 2020-03-15:
I think I noticed the same issue. Reporting did not work properly for a few bulbs. And Z2M did not seem to configure them, even after a reboot of the bulbs and Z2M itself.

After I set `permit_join` to `true` and restarted Z2M, it seemed it had reconfigurered the bulbs as suddenly reporting started to work.

Koenkk, 2020-03-16:
I'm not able to reproduce this on the latest dev.

`permit_join: true`

```
zigbee2mqtt:info  2020-03-16 18:23:43: Logging to console and directory: '/Users/Koenkk/Git/zigbee2mqtt/data/log/2020-03-16.18-23-43' filename: log.txt
zigbee2mqtt:info  2020-03-16 18:23:44: Starting zigbee2mqtt version 1.11.0-dev (commit #27b3fd2)
zigbee2mqtt:info  2020-03-16 18:23:44: Starting zigbee-herdsman...
zigbee2mqtt:info  2020-03-16 18:23:45: zigbee-herdsman started
zigbee2mqtt:info  2020-03-16 18:23:45: Coordinator firmware version: '{"type":"zStack12","meta":{"transportrev":2,"product":0,"majorrel":2,"minorrel":6,"maintrel":3,"revision":20190619}}'
zigbee2mqtt:info  2020-03-16 18:23:45: Currently 1 devices are joined:
zigbee2mqtt:info  2020-03-16 18:23:45: ikea_bulb (0x90fd9ffffe6494fc): LED1624G9 - IKEA TRADFRI LED bulb E14/E26/E27 600 lumen, dimmable, color, opal white (Router)
zigbee2mqtt:warn  2020-03-16 18:23:45: `permit_join` set to  `true` in configuration.yaml.
zigbee2mqtt:warn  2020-03-16 18:23:45: Allowing new devices to join.
zigbee2mqtt:warn  2020-03-16 18:23:45: Set `permit_join` to `false` once you joined all devices.
zigbee2mqtt:info  2020-03-16 18:23:45: Zigbee: allowing new devices to join.
zigbee2mqtt:info  2020-03-16 18:23:45: Connecting to MQTT server at mqtt://localhost
zigbee2mqtt:info  2020-03-16 18:23:45: Connected to MQTT server
zigbee2mqtt:info  2020-03-16 18:23:45: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload 'online'
zigbee2mqtt:info  2020-03-16 18:23:45: MQTT publish: topic 'zigbee2mqtt/ikea_bulb', payload '{"color":{"x":0.458,"y":0.41},"linkquality":157,"brightness":254}'
zigbee2mqtt:info  2020-03-16 18:23:45: MQTT publish: topic 'zigbee2mqtt/bridge/config', payload '{"version":"1.11.0-dev","commit":"27b3fd2","coordinator":{"type":"zStack12","meta":{"transportrev":2,"product":0,"majorrel":2,"minorrel":6,"maintrel":3,"revision":20190619}},"log_level":"info","permit_join":true}'
zigbee2mqtt:info  2020-03-16 18:23:47: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_announced","message":"announce","meta":{"friendly_name":"ikea_bulb"}}'
zigbee2mqtt:info  2020-03-16 18:23:47: Successfully setup reporting for '0x90fd9ffffe6494fc' - 1 - genOnOff
zigbee2mqtt:info  2020-03-16 18:23:48: Successfully setup reporting for '0x90fd9ffffe6494fc' - 1 - genLevelCtrl
zigbee2mqtt:info  2020-03-16 18:23:48: Successfully setup reporting for '0x90fd9ffffe6494fc' - 1 - lightingColorCtrl
```

`permit_join: false`

```
zigbee2mqtt:info  2020-03-16 18:24:19: Logging to console and directory: '/Users/Koenkk/Git/zigbee2mqtt/data/log/2020-03-16.18-24-19' filename: log.txt
zigbee2mqtt:info  2020-03-16 18:24:19: Starting zigbee2mqtt version 1.11.0-dev (commit #27b3fd2)
zigbee2mqtt:info  2020-03-16 18:24:19: Starting zigbee-herdsman...
zigbee2mqtt:info  2020-03-16 18:24:21: zigbee-herdsman started
zigbee2mqtt:info  2020-03-16 18:24:21: Coordinator firmware version: '{"type":"zStack12","meta":{"transportrev":2,"product":0,"majorrel":2,"minorrel":6,"maintrel":3,"revision":20190619}}'
zigbee2mqtt:info  2020-03-16 18:24:21: Currently 1 devices are joined:
zigbee2mqtt:info  2020-03-16 18:24:21: ikea_bulb (0x90fd9ffffe6494fc): LED1624G9 - IKEA TRADFRI LED bulb E14/E26/E27 600 lumen, dimmable, color, opal white (Router)
zigbee2mqtt:info  2020-03-16 18:24:21: Zigbee: disabling joining new devices.
zigbee2mqtt:info  2020-03-16 18:24:21: Connecting to MQTT server at mqtt://localhost
zigbee2mqtt:info  2020-03-16 18:24:21: Connected to MQTT server
zigbee2mqtt:info  2020-03-16 18:24:21: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload 'online'
zigbee2mqtt:info  2020-03-16 18:24:21: MQTT publish: topic 'zigbee2mqtt/ikea_bulb', payload '{"color":{"x":0.458,"y":0.41},"linkquality":157,"brightness":254}'
zigbee2mqtt:info  2020-03-16 18:24:21: MQTT publish: topic 'zigbee2mqtt/bridge/config', payload '{"version":"1.11.0-dev","commit":"27b3fd2","coordinator":{"type":"zStack12","meta":{"transportrev":2,"product":0,"majorrel":2,"minorrel":6,"maintrel":3,"revision":20190619}},"log_level":"info","permit_join":false}'
zigbee2mqtt:info  2020-03-16 18:24:24: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_announced","message":"announce","meta":{"friendly_name":"ikea_bulb"}}'
zigbee2mqtt:info  2020-03-16 18:24:24: Successfully setup reporting for '0x90fd9ffffe6494fc' - 1 - genOnOff
zigbee2mqtt:info  2020-03-16 18:24:24: Successfully setup reporting for '0x90fd9ffffe6494fc' - 1 - genLevelCtrl
zigbee2mqtt:info  2020-03-16 18:24:25: Successfully setup reporting for '0x90fd9ffffe6494fc' - 1 - lightingColorCtrl
```

middelink, 2020-03-20:
Actually, I have been playing with it a bit more with 1.11.0. It seems to be related how long the z2m instance has been running. Directly after a restart it is able to setup the reporting, but if it has been on for say a day (not quite sure where the deadline is, and is probably depending on something I have no idea about) it stops updating the reporting.

However, 20 hours ago I switched to 1.12.0 and 20 hours in, I am able to physically switch the bulb off/on and reporting gets setup correctly. So consider this accidentally fixed?

```
zigbee2mqtt:debug 2020-03-20 09:22:41: Device 'TRADFRI Kantoor Bureau Links' announced itself
zigbee2mqtt:info  2020-03-20 09:22:41: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_announced","message":"announce","meta":{"friendly_name":"TRADFRI Kantoor Bureau Links"}}'
zigbee2mqtt:debug 2020-03-20 09:22:41: Setup reporting for '0x000b57fffed6e84b' - 1 - genOnOff
zigbee2mqtt:info  2020-03-20 09:22:42: Successfully setup reporting for '0x000b57fffed6e84b' - 1 - genOnOff
zigbee2mqtt:debug 2020-03-20 09:22:42: Setup reporting for '0x000b57fffed6e84b' - 1 - genLevelCtrl
zigbee2mqtt:info  2020-03-20 09:22:42: Successfully setup reporting for '0x000b57fffed6e84b' - 1 - genLevelCtrl
zigbee2mqtt:debug 2020-03-20 09:22:45: Received Zigbee message from 'TRADFRI Kantoor Bureau Links', type 'attributeReport', cluster 'genLevelCtrl', data '{"currentLevel":254}' from endpoint 1 with groupID 0
zigbee2mqtt:info  2020-03-20 09:22:45: MQTT publish: topic 'zigbee2mqtt/TRADFRI Kantoor Bureau Links', payload '{"state":"OFF","linkquality":126,"brightness":254,"update_available":false}'
```

closed 2020-03-20 by middelink

sjorge, 2020-03-20:
Might be related to why joining fails when enAbled after a few days.

~ sjorge