Koenkk/zigbee2mqtt #3116: Sinope DM2500ZB - interview fails

bvanheu, 2020-03-13:
# Bug Report
## What happened

Unable to interview with a Sinope dimmer switch DM2500ZB.
https://www.sinopetech.com/en/product/smart-dimmer-600-w-zigbee/

## What did you expect to happen

Pairing successful.

## How to reproduce it (minimal and precise)

I press both button on the switch to trigger the pairing. Pairing is succesful from the device POV (green light on the device). Then the fail to interview happens. I can then disconnect the device by pressing both button for 10 sec.

Tried 4 times:
1. try pairing normally
2. unpaired the device, then tried again the pairing
3. power cycled the CC2531 then try the pairing
4. power-cycled the device then try the pairing

I have 7 Sinope devices successfully paired.

## Debug Info
zigbee2mqtt version: zigbee2mqtt@1.10.0 (commit #8df6be8)
CC253X firmware version: zStack12 - 20190608
(based on: Coordinator firmware version: '{"type":"zStack12","meta":{"transportrev":2,"product":0,"majorrel":2,"minorrel":6,"maintrel":3,"revision":20190608}}')
Logs: https://pastebin.com/4dfh5mfh

The device gets added to 'database.db':
```
{"id":8,"type":"Router","ieeeAddr":"0x500b91400001e5b0","nwkAddr":20491,"manufId":4508,"epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":260,"inClusterList":[0,3,4,5,6,8,2821,65281],"outClusterList":  [3,4,25],"clusters":{},"binds":[]}},"interviewCompleted":false,"meta":{},"lastSeen":1584064878141}
```

sjorge, 2020-03-13:
Looks like the device doesn't respond to the reads during the pairning, looks to be a mains powered device thoguh... which is odd.

But I guess you could try the trick for when pairing battery devices, were you try to keep it awake by clicking the button.

bvanheu, 2020-03-21:
Thanks for the suggestion, I tried to interact with the device while pairing but it didn't help.

I bought a second CC2531 to do more extensive testing with the DM2500ZB.

closed 2020-05-30 by stale[bot]

bvanheu, 2020-06-26:
For the record, i've just tested a different CC2531 device (with an antenna) with the latest version of zigbee2mqtt and it just worked out of the box.

I've submitted a PR to add support for on/off + brightness of the dimmer:
https://github.com/Koenkk/zigbee-herdsman-converters/pull/1350