Koenkk/zigbee2mqtt #3117: UnhandledPromiseRejectionWarning: Error: SRSP - ZDO - mgmtPermitJoinReq after 6000ms

multiholle, 2020-03-13:
# Bug Report
I'm trying to setup zigbee2mqtt with Home Assistant. I successfully installed the add on and it starts without errors.
## What happened
I tried to pair an [Ikea switch](https://www.zigbee2mqtt.io/devices/E1743.html). The pairing fails with an error message. (see log below). The error is thrown once the pairing seems to start.

<details><summary>log</summary>

```
[s6-init] making user provided files available at /var/run/s6/etc...exited 0.
[s6-init] ensuring user provided files have correct perms...exited 0.
[fix-attrs.d] applying ownership & permissions fixes...
[fix-attrs.d] done.
[cont-init.d] executing container initialization scripts...
[cont-init.d] done.
[services.d] starting services
[services.d] done.
./run.sh: line 17: [Info] Configuration backup found in /share/zigbee2mqtt/.configuration.yaml.bk. Skipping config backup.: No such file or directory
[Info] Socat is DISABLED and not started
2020-03-13T11:20:26: PM2 log: Launching in no daemon mode
2020-03-13T11:20:27: PM2 log: App [npm:0] starting in -fork mode-
2020-03-13T11:20:27: PM2 log: App [npm:0] online
> zigbee2mqtt@1.11.0 start /zigbee2mqtt-1.11.0
> node index.js
zigbee2mqtt:info  2020-03-13 11:20:38: Logging to console and directory: '/share/zigbee2mqtt/log/2020-03-13.11-20-38' filename: log.txt
zigbee2mqtt:info  2020-03-13 11:20:39: Starting zigbee2mqtt version 1.11.0 (commit #unknown)
zigbee2mqtt:info  2020-03-13 11:20:39: Starting zigbee-herdsman...
zigbee2mqtt:info  2020-03-13 11:20:45: zigbee-herdsman started
zigbee2mqtt:info  2020-03-13 11:20:45: Coordinator firmware version: '{"type":"zStack12","meta":{"transportrev":2,"product":0,"majorrel":2,"minorrel":6,"maintrel":3,"revision":20190619}}'
zigbee2mqtt:info  2020-03-13 11:20:45: Currently 0 devices are joined:
zigbee2mqtt:warn  2020-03-13 11:20:45: `permit_join` set to  `true` in configuration.yaml.
zigbee2mqtt:warn  2020-03-13 11:20:45: Allowing new devices to join.
zigbee2mqtt:warn  2020-03-13 11:20:45: Set `permit_join` to `false` once you joined all devices.
zigbee2mqtt:info  2020-03-13 11:20:45: Zigbee: allowing new devices to join.
zigbee2mqtt:info  2020-03-13 11:20:45: Connecting to MQTT server at mqtt://core-mosquitto
zigbee2mqtt:info  2020-03-13 11:20:46: Connected to MQTT server
zigbee2mqtt:info  2020-03-13 11:20:46: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload 'online'
zigbee2mqtt:info  2020-03-13 11:20:46: MQTT publish: topic 'zigbee2mqtt/bridge/config', payload '{"version":"1.11.0","commit":"unknown","coordinator":{"type":"zStack12","meta":{"transportrev":2,"product":0,"majorrel":2,"minorrel":6,"maintrel":3,"revision":20190619}},"log_level":"info","permit_join":true}'
(node:226) UnhandledPromiseRejectionWarning: Error: SRSP - ZDO - mgmtPermitJoinReq after 6000ms
    at Timeout._onTimeout (/zigbee2mqtt-1.11.0/node_modules/zigbee-herdsman/dist/utils/waitress.js:44:24)
    at listOnTimeout (internal/timers.js:531:17)
    at processTimers (internal/timers.js:475:7)
(node:226) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). (rejection id: 1)
(node:226) [DEP0018] DeprecationWarning: Unhandled promise rejections are deprecated. In the future, promise rejections that are not handled will terminate the Node.js process with a non-zero exit code.
```
</details>

## Debug Info
zigbee2mqtt version: 1.11.0
CC253X firmware version: Z-Stack-firmware/coordinator/Z-Stack_Home_1.2

backcountrymountains, 2020-03-18:
I also have this problem. From what I've read, one has to re-flash the cc2531 to recover from this error. However, you said that you have done so by trying different firmware. Perhaps there is some other issue with 1.11.0 that needs to get resolved. 

My cc2531 completely stops responding after the error and there was no amount of rebooting or resetting that was able to bring it back to life.

Just bumping to commiserate, I guess.

Koenkk, 2020-03-18:
@multiholle could you switch to the edge addon and set `zigbee_herdsman_debug: true` in the configuration and provide the logging up till the crash?

berrywhite96, 2020-04-22:
Don't know if the issue is similar, but the log lines looks the same.

For me zigbee2mqtt works fine for a few days, then nothing works.
Here are the last few lines:
`Apr 21 13:25:39 raspberrypi npm[472]: (node:874) UnhandledPromiseRejectionWarning: Error: SRSP - SYS - version after 6000ms
Apr 21 13:25:39 raspberrypi npm[472]:     at Timeout._onTimeout (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/utils/waitress.js:44:24)
Apr 21 13:25:39 raspberrypi npm[472]:     at listOnTimeout (internal/timers.js:531:17)
Apr 21 13:25:39 raspberrypi npm[472]:     at processTimers (internal/timers.js:475:7)
Apr 21 13:25:39 raspberrypi npm[472]: (node:874) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was
Apr 22 13:25:39 raspberrypi npm[472]: (node:874) UnhandledPromiseRejectionWarning: Error: SRSP - SYS - version after 6000ms
Apr 22 13:25:39 raspberrypi npm[472]:     at Timeout._onTimeout (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/utils/waitress.js:44:24)
Apr 22 13:25:39 raspberrypi npm[472]:     at listOnTimeout (internal/timers.js:531:17)
Apr 22 13:25:39 raspberrypi npm[472]:     at processTimers (internal/timers.js:475:7)
Apr 22 13:25:39 raspberrypi npm[472]: (node:874) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was`

Koenkk, 2020-04-22:
@berrywhite96 likely a stability issue of the stick, try the source routing firmware: https://github.com/Koenkk/Z-Stack-firmware/tree/master/coordinator/Z-Stack_Home_1.2/bin/source_routing which has increased stability.

berrywhite96, 2020-04-22:
> @berrywhite96 likely a stability issue of the stick, try the source routing firmware: https://github.com/Koenkk/Z-Stack-firmware/tree/master/coordinator/Z-Stack_Home_1.2/bin/source_routing which has increased stability.

But I want to use it as a coordinator, does this also work?

Koenkk, 2020-04-22:
yes, it's a coordinator firmware.

skvsn, 2020-05-01:
I just had the same error message in the log file, however in my case restarting zigbee2mqtt HA add-on solved the problem and I can see all my sensors are working fine (only have 2 at the moment). Is it worth trying source routing firmware?

berrywhite96, 2020-05-01:
> I just had the same error message in the log file, however in my case restarting zigbee2mqtt HA add-on solved the problem and I can see all my sensors are working fine (only have 2 at the moment). Is it worth trying source routing firmware?

Flashed my CC2530 a week ago, since then it works perfect!

closed 2020-07-10 by stale[bot]