Koenkk/zigbee2mqtt #3119: Failed to interview

ressof, 2020-03-13:
# Bug Report
## What happened
Failed to interview
## What did you expect to happen

## How to reproduce it (minimal and precise)

## Debug Info
zigbee2mqtt version: 1.11.0
CC253X firmware version:20190608

[https://pastebin.com/UknfTd72](url)

vaceslav, 2020-03-13:
please provide more information.

Which device did you try to connect? Manufacturer?  Model?

ressof, 2020-03-16:
It is a bulb from IKEA, LED1624G9

vaceslav, 2020-03-17:
Did you follow instruction from docu page?
https://www.zigbee2mqtt.io/devices/LED1624G9.html

> While pairing, keep the bulb close the the CC2531 USB sniffer.
>
> What works is to use (very) short “on’s” and a little bit longer “off’s”. Start with bulb on, then off, and then 6 “on’s”, where you kill the light as soon as the bulb shows signs of turning on.

ressof, 2020-03-18:
Yes

closed 2020-05-24 by stale[bot]