Koenkk/zigbee2mqtt #3120: Philips LivingColors Iris 7099960PH is not supported

uweng, 2020-03-13:
Hi,

I've checked the https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/devices.js for my  Philips LivingColors Iris **7099960PH**. But it is not in there. 
Please add support for this device. 

Best regards
Uwe

uweng, 2020-03-13:
Hardware reset to interview the device does not work and "touch link" as well. I got "Failed to factory reset device through Touchlink".

nldroid, 2020-03-13:
Hey Uwe, did you check this?

https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html

It's hard to add it when you don't have the device itself..

uweng, 2020-03-13:
I have the device and the CC2531 does not interview the lamp:

configuration.yaml:

<pre>
# Home Assistant integration (MQTT discovery)
homeassistant: false

# allow new devices to join
permit_join: true

# MQTT settings
mqtt:
  # MQTT base topic for zigbee2mqtt MQTT messages
  base_topic: zigbee2mqtt
  # MQTT server URL
  server: 'mqtt://localhost'
  # MQTT server authentication, uncomment if required:
  # user: my_user
  # password: my_password

# Serial settings
serial:
  # Location of CC2531 USB sniffer
  port: /dev/ttyACM0

advanced:
  log_level: debug
</pre>

here my logfile, no pearing information from lamp during reset procedure:
<pre>
info  2020-03-13 15:28:59: Logging to console and directory: '/opt/zigbee2mqtt/data/log/2020-03-13.15-28-59'
debug 2020-03-13 15:28:59: Removing old log directory '/opt/zigbee2mqtt/data/log/2020-03-13.15-27-46'
debug 2020-03-13 15:29:00: Loaded state from file /opt/zigbee2mqtt/data/state.json
info  2020-03-13 15:29:00: Starting zigbee2mqtt version 1.8.0 (commit #da4d26a)
info  2020-03-13 15:29:00: Starting zigbee-herdsman...
debug 2020-03-13 15:29:00: Using zigbee-herdsman with settings: '{"network":{"panID":6754,"extendedPanID":[221,221,221,221,221,221,221,221],"channelList":[11],"networkKey":"HIDDEN"},"databasePath":"/opt/zigbee2mqtt/data/database.db","databaseBackupPath":"/opt/zigbee2mqtt/data/database.db.backup","backupPath":"/opt/zigbee2mqtt/data/coordinator_backup.json","serialPort":{"baudRate":115200,"rtscts":true,"path":"/dev/ttyACM0"}}'
info  2020-03-13 15:29:04: zigbee-herdsman started
info  2020-03-13 15:29:04: Coordinator firmware version: '{"type":"zStack12","meta":{"transportrev":2,"product":0,"majorrel":2,"minorrel":6,"maintrel":3,"revision":20190608}}'
debug 2020-03-13 15:29:04: Zigbee network parameters: {"panID":6754,"extendedPanID":"0xdddddddddddddddd","channel":11}
info  2020-03-13 15:29:04: Currently 0 devices are joined:
warn  2020-03-13 15:29:04: `permit_join` set to  `true` in configuration.yaml.
warn  2020-03-13 15:29:04: Allowing new devices to join.
warn  2020-03-13 15:29:04: Set `permit_join` to `false` once you joined all devices.
info  2020-03-13 15:29:04: Zigbee: allowing new devices to join.
info  2020-03-13 15:29:04: Connecting to MQTT server at mqtt://localhost
info  2020-03-13 15:29:04: Connected to MQTT server
info  2020-03-13 15:29:04: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload 'online'
info  2020-03-13 15:29:04: MQTT publish: topic 'zigbee2mqtt/bridge/config', payload '{"version":"1.8.0","commit":"da4d26a","coordinator":{"type":"zStack12","meta":{"transportrev":2,"product":0,"majorrel":2,"minorrel":6,"maintrel":3,"revision":20190608}},"log_level":"debug","permit_join":true}'
debug 2020-03-13 15:34:00: Saving state to file /opt/zigbee2mqtt/data/state.json
</pre>

Koenkk, 2020-03-13:
I don't think that this is a Zigbee device? (https://www.lighting.philips.co.uk/consumer/p/livingcolors-table-lamp/7099960PH)

uweng, 2020-03-14:
Why not?
https://www.smartlights.de/licht/philips-hue/livingcolors-iris-bloom-ins-philips-hue-system-einbinden-und-steuern/

Koenkk, 2020-03-15:
Did you try to reset the device through touchlink? (https://www.zigbee2mqtt.io/information/touchlink.html)

uweng, 2020-03-15:
Yes sure:
<pre>
debug 2020-03-15 17:33:08: Received MQTT message on 'zigbee2mqtt/bridge/config/touchlink/factory_reset' with data ''
info  2020-03-15 17:33:08: Starting touchlink factory reset...
warn  2020-03-15 17:33:33: Failed to factory reset device through Touchlink
debug 2020-03-15 17:34:02: Saving state to file /opt/zigbee2mqtt/data/state.json
debug 2020-03-15 17:34:37: Received MQTT message on 'zigbee2mqtt/bridge/config/touchlink/factory_reset' with data ''
info  2020-03-15 17:34:37: Starting touchlink factory reset...
warn  2020-03-15 17:35:02: Failed to factory reset device through Touchlink
debug 2020-03-15 17:36:02: Received MQTT message on 'zigbee2mqtt/bridge/config/touchlink/factory_reset' with data ''
info  2020-03-15 17:36:02: Starting touchlink factory reset...
warn  2020-03-15 17:36:26: Failed to factory reset device through Touchlink
</pre>

As proposed on
https://www.zigbee2mqtt.io/information/touchlink.html

Koenkk, 2020-03-15:
I still doubt if this is a Zigbee device, it seems to be from 2012 (https://www.reddit.com/r/Hue/comments/3ywckr/does_the_7099960ph_livingcolors_iris_work_with_hue/)

> I hope you are right, haha. Thing is, there is this one which I have: 70999/60/PH (according to Philips it's from 2012) And there is this one: 71996/60/PH (according to Philips it's from 2014)
I already called Philips, but the person I got on the phone said it doesn't work, though she wasn't really confidence inspiring, which is why I'm asking it here too.

Besides the site you linked I cannot find any sources that it supports Zigbee.

uweng, 2020-03-31:
Maybe you are right. I've seen a comment in this blog:

https://www.alefo.de/forum/echo-plus-mit-philips-living-colors-verbinden-3957-10?sid=bbe31933cf09abac004fee4fe29410ae

Philips has built lamps with support for the hue bridge ( zigbee  protocol ). But it was not real intended that way. So they removed it via new firmware. It could be i have souch a device. Thats a pity. 

Thank you anyway!;-)

Best regards
Uwe

uweng, 2020-03-31:
Device may not support zigbee protocol.

closed 2020-03-31 by uweng

Eymerich01, 2021-01-21:
I'm added a philips 7099760ph with factory reset operation and it work .  You could add it as device supported

Papnaas, 2024-08-26:
I have running three 7099960PH for a while now. Device is perfectly supported, but factory reset is an issue. It came with a remote control though. If you have access to it, hold it near the bulb and simultaneously press the "on" key (labelled "I") and the button with one dot on it for a few seconds. If it works (sometimes you need to try 2-3 times) the bulb flashes and is ready to join a new network.

Novocaine85, 2024-11-10:
> 7099960PH

are you sure is the exact model? I've tried many times along those years but i never get it to be paired even using your mentioned method.

Papnaas, 2024-11-10:
Yes, I am surr. Just cross-checked it on the little Tag attached to the cable of the lamp. 
I updated the device several times by now, but I don‘t trink that made the difference, when switching from Hue bridge to zigbee2mqtt (or ZHA meanwhile).

Novocaine85, 2024-11-10:
Just retried, won't pair.
Light at reset flashes, than light is green cause is paired to the remote.
Unless the hub did something to your lamp like a firmware update, i don't know how to pair it.

Got the same sticker on the cable.

Papnaas, 2024-11-10:
Strange. Maybe you Shops try to Update over the Hue Bridge?
Nur there is one other thing: I don’t think that my lamp was connected to the remote after the factory reset! It seems to me that your lamp cannot connect to your network, because it is paired to your remote. Unfortunately though, I cannot 100% remember how to disconnect from the remote… I fear you need to connect to the Hue Bridge at least once for this. I think adding (with the code in the tag) and removing it from the bridge might do the trick!