Koenkk/zigbee2mqtt #3124: innr SP 222 plug not detected as known device

andyshilton, 2020-03-14:
# Bug Report
## What happened
Paired the device but it produces a lot of errors in the log and fails to set up correctly.
## What did you expect to happen
The device to pair correctly
## How to reproduce it (minimal and precise)
Put the coordinator into pairing mode
Plug in the innr SP 222 plug
View the logs and see the errors below.
## Debug Info
zigbee2mqtt version: 1.11.0
CC253X firmware version:
Using the CC135P-2 with firmware CC1352P_2_20191106.hex
## Log info
```zigbee2mqtt:warn  2020-03-14 13:29:22: Received message from unsupported device with Zigbee model 'undefined'
zigbee2mqtt:warn  2020-03-14 13:29:22: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
zigbee2mqtt:error 2020-03-14 13:29:22: Failed to call 'OTAUpdate' 'onZigbeeEvent' (TypeError: Cannot read property 'hasOwnProperty' of null
    at OTAUpdate.onZigbeeEvent (/zigbee2mqtt-1.11.0/lib/extension/otaUpdate.js:21:75)
    at Controller.callExtensionMethod (/zigbee2mqtt-1.11.0/lib/controller.js:329:44))
zigbee2mqtt:warn  2020-03-14 13:29:25: Received message from unsupported device with Zigbee model 'undefined'
zigbee2mqtt:warn  2020-03-14 13:29:25: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
zigbee2mqtt:error 2020-03-14 13:29:25: Failed to call 'OTAUpdate' 'onZigbeeEvent' (TypeError: Cannot read property 'hasOwnProperty' of null
    at OTAUpdate.onZigbeeEvent (/zigbee2mqtt-1.11.0/lib/extension/otaUpdate.js:21:75)
    at Controller.callExtensionMethod (/zigbee2mqtt-1.11.0/lib/controller.js:329:44))
zigbee2mqtt:warn  2020-03-14 13:29:28: Received message from unsupported device with Zigbee model 'undefined'
zigbee2mqtt:warn  2020-03-14 13:29:28: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
zigbee2mqtt:error 2020-03-14 13:29:28: Failed to call 'OTAUpdate' 'onZigbeeEvent' (TypeError: Cannot read property 'hasOwnProperty' of null
    at OTAUpdate.onZigbeeEvent (/zigbee2mqtt-1.11.0/lib/extension/otaUpdate.js:21:75)
    at Controller.callExtensionMethod (/zigbee2mqtt-1.11.0/lib/controller.js:329:44))
zigbee2mqtt:warn  2020-03-14 13:29:31: Received message from unsupported device with Zigbee model 'undefined'
zigbee2mqtt:warn  2020-03-14 13:29:31: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
zigbee2mqtt:error 2020-03-14 13:29:31: Failed to call 'OTAUpdate' 'onZigbeeEvent' (TypeError: Cannot read property 'hasOwnProperty' of null
    at OTAUpdate.onZigbeeEvent (/zigbee2mqtt-1.11.0/lib/extension/otaUpdate.js:21:75)
    at Controller.callExtensionMethod (/zigbee2mqtt-1.11.0/lib/controller.js:329:44))```

Koenkk, 2020-03-15:
There error in the log you've posted has already been fixed in the latest dev branch.

For the not supported part, please provide the log when pairing the device with debug log enabled.

To enable debug logging set in `configuration.yaml`:
```yaml
advanced:
  log_level: debug
```

andyshilton, 2020-03-15:
@Koenkk Thank you! I didn't know if it would help or not so I thought I'd post it in case. I'm not sure I will be able to replicate it again. I left the plug unplugged for a couple of hours. When I plugged it back in, it just paired in the normal way and is now working.

closed 2020-05-22 by stale[bot]