Koenkk/zigbee2mqtt #3127: IKEA TRADFRI control outlet (E1603/E1702) losing connect

turboproc, 2020-03-14:
Running a stable network for several weeks I suddenly have an IKEA control outlet that loses connections. Removing and repairing solves the issue for a while but after that 205 errors re-appear again. That is 100% connection loss.

Running a coordinatior based on a CC2531 with latest FW (20190608)  and Z2M on latest release as well. Network contains 19 devices so far.
The other 4 outlets have no issue at all.

```
Publish 'set' 'state' to '0x086bd7fffe065ca2' failed: 'Error: Command 0x086bd7fffe065ca2/1 genOnOff.on({}, {"timeout":6000,"manufacturerCode":null,"disableDefaultResponse":false}) failed (Error: Data request failed with error: 'No network route' (205))'
```
Any suggestions how to solve?

sjorge, 2020-03-14:
Are you on the current dev? The latest release has some issues with my ikea bulbs as wel, it's better on the current dev after another change to the route request stuff.

kloknibor, 2020-03-15:
I have 2 of these outlets and it happends with both of them. Has been running without problems for months... It comes online once every hour and other than that is unavailable all off a sudden...

closed 2020-05-22 by stale[bot]