Koenkk/zigbee2mqtt #3128: Ikea Tradfri E14 470lm 2700 Kelvin Device Request

amrap030, 2020-03-14:
Hello, could you please add the Ikea Tradfri E14 470lm 2700 Kelvin light bulb to the supported devices?

sjorge, 2020-03-14:
Do you have a database.db entry for one + a link to the ikea page? I can probably figure out the devices.js entry for that based on that.

amrap030, 2020-03-14:
I don't have that but on the packaging I can see Tradfri LED 470 lm 5.2W E14 LED1835C6 and under the barcode 704.243.12 19972.

sjorge, 2020-03-14:
Looks to be this one then? https://www.ikea.com/nl/en/p/tradfri-led-bulb-e14-470-lumen-white-spectrum-wireless-dimmable-opal-white-chandelier-70424312/

sjorge, 2020-03-14:
Just a complete stab in the dark, but yo ucan try adding this to devices.js

```
    {
        zigbeeModel: ['TRADFRI bulb E14 470lm'],
        model: 'LED1835C6',
        vendor: 'IKEA',
        description: 'TRADFRI bulb E14 470 lumen, dimmable, opal white',
        extend: generic.light_onoff_brightness,
        ota: ota.tradfri,
    },
```

JBarrskog, 2020-03-29:
I can confirm that the linked IKEA-page is the product corresponding to the model number LED1835C6. 

I don't understand where I am supposed to paste the suggested solution by @sjorge.

I have managed to pair the light - but after a zigbee2mqtt restart I receive

`zigbee2mqtt:info  2020-03-29 22:20:40: Vardagsrum 2 (0x680ae2fffe713903): Not supported (Router)`

The entry in the database.db-file

`{"id":12,"type":"Router","ieeeAddr":"0x680ae2fffe713903","nwkAddr":58221,"manufId":4476,"manufName":"IKEA of Sweden","powerSource":"Mains (single phase)","modelId":"TRADFRI bulb E14 WS 470lm","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":268,"inClusterList":[0,3,4,5,6,8,768,4096],"outClusterList":[5,25,32,4096],"clusters":{"genBasic":{"attributes":{"modelId":"TRADFRI bulb E14 WS 470lm","manufacturerName":"IKEA of Sweden","powerSource":1,"zclVersion":3,"appVersion":33,"stackVersion":98,"hwVersion":1,"dateCode":"20190312","swBuildId":"2.1.022"}}},"binds":[]},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[33],"outClusterList":[33],"clusters":{},"binds":[]}},"appVersion":33,"stackVersion":98,"hwVersion":1,"dateCode":"20190312","swBuildId":"2.1.022","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1585513112429}`

sjorge, 2020-03-29:
There is some info here https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html

Just use the entry from my previous post instead of the example and I think it should work.

mabachel, 2020-05-29:
The difference between the 
- already supported LED1649C5 ([IKEA Germany Product Link](https://www.ikea.com/de/de/p/tradfri-led-leuchtmittel-e14-400-lm-kabellos-dimmbar-warmes-tageslicht-kerzenfoermig-opalweiss-60365271/))

and the

- not yet supported LED1835C6 ([IKEA Germany Product Link](https://www.ikea.com/de/de/p/tradfri-led-leuchtmittel-e14-470-lm-kabellos-dimmbar-weissspektrum-weissspektrum-kabellos-dimmbar-kerzenfoermig-opalweiss-kerzenfoermig-opalweiss-70424312/))

is that the new model (LED1835C6) offers a colour temperature control (3 steps from warm white to cold white) in addition to brightness control and on/off control which the old  model (LED1649C5) already offered.

mabachel, 2020-06-02:
Is someone out there who got it (LED1835C6) working?

JBarrskog, 2020-06-02:
No, I have not. I run my home automation on a Raspberry Pi 4 using Docker and I’ve not been able to run “npm ci” in the bucket in order to get started. Have paused my efforts due to this struggle :/

mabachel, 2020-06-02:
I just found this enrty in [devices.js](https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/devices.js) from the dev branch.
```
{
        zigbeeModel: ['TRADFRI bulb E14 WS 470lm'],
        model: 'LED1903C5/LED1835C6',
        vendor: 'IKEA',
        description: 'TRADFRI bulb E14 WS 470 lumen, dimmable, white spectrum, opal white',
        extend: generic.light_onoff_brightness_colortemp,
        ota: ota.tradfri,
},
```

so there should be (some) support already in place..?

Koenkk, 2020-06-02:
@mabachel if the zigbeeModel is already there, the bulb should work just fine. The model (just a cosmetic thing) will be updated automatically on the next zigbee2mqtt release.

mabachel, 2020-06-02:
Thanks for the clarification.
@JBarrskog & @amrap030 this commit is from May 25, 2020. Thus LED1835C6 should be officially supported in the next release.
https://github.com/Koenkk/zigbee-herdsman-converters/commit/5b069f1892582c734ee31df005a8808afdc52f78

This issue can be closed.

closed 2020-06-02 by Koenkk

JBarrskog, 2020-08-05:
@mabachel @amrap030 Can confirm that 1.14.2 works with LED1835C6. I had to fully remove the light before it would pair. Thanks alot!