Koenkk/zigbee2mqtt #3129: Device (Embertec power meter) leave the network

Melvictsa, 2020-03-15:
Hello,
I have a Embertec 4 channel power meter and when I try to pair it, it initially pair according to the log but in the end it leave the network again.
The configuration if the devices,js script:
 //EmberMeter
    {
        zigbeeModel: ['EMAU-ZN04-HA-01'], 
        model: 'EMAU-ZN04-HA-01',
        vendor: 'Embertec',
        description: 'Embermeter', 
        supports: 'power measurement',
        fromZigbee: [fz.metering_power], 
        toZigbee: [],
    },

Debug log:
https://hastebin.com/wayokadowa.coffeescript

Any pointers would be most welcome.

Koenkk, 2020-03-15:
I've also seen this behaviour for other devices, after sniffing the traffic with the original gateway it appears that you need to send some "magic" payload in order for these devices to stay in the network. (https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/index.js#L39)

You would need to sniff the traffic when pairing it with the original gateway to figure this out.

closed 2020-05-22 by stale[bot]

Melvictsa, 2023-06-04:
I eventually came round to this again, it took a while to get the sniffer.

[Emberpulse trace 20230601.zip](https://github.com/Koenkk/zigbee2mqtt/files/11645095/Emberpulse.trace.20230601.zip)

The Rejoin Request is at line 405.

Is there perhaps one device that I can modify to get this working?

Any help is much appreciated.

samburney, 2026-11-15:
Hi @Melvictsa, in June 2023 you mentioned on the Home Assistant forum that you got the Embermeter (EMAU-ZN04-HA-01) pairing and staying connected with Zigbee2MQTT.  I'm hitting the same problem: it joins, the interview completes, then it leaves about 1.8 s later, on both ZHA and Z2M.

Are you able to please share how you managed to get this to stay connected to Zigbee2MQTT?

Apologies for the obscure communications channel, I've struggled to find any other information, or way to contact you!

Thanks.