Koenkk/zigbee2mqtt #3130: Expose light data for TRADFRI Motion Sensor

mariusmotea, 2020-03-15:
## What happened

Tradfri motion sensor has an integrated light level. Albeit this is just a binary level sensor, it will be very useful to know if the lights need to be turned on or not based on light level returned by this sensor. Currently i'm counting on a logical sensor that calculate the sunrise and sunset based on my location, but this is not efficient because on some cloudy days (aka today) the natural light level is very low, and my lights remain off despite is dark. I'm switching to zigbee2mqtt from Deconz and there the light sensor is exposed, albeit it report the measurement from previews motion detection (i know this because in the morning on it always trigger the lights on firs detection)

## What did you expect to happen

In the output json to have the parameter `"dark": true` or `"dark": false`, exactly like in the Hue API.

## Debug Info
zigbee2mqtt version: Docker, latest
CC253X firmware version: latest.

Koenkk, 2020-03-15:
I own an old TRADFRI motion sensor (E1525, `"dateCode":"20170302","swBuildId":"1.2.214"`) and sniffed the traffic. I couldn't find any commands that contain the light data information. 

Could you sniff the traffic when paired with Deconz when the dark value changes? (https://www.zigbee2mqtt.io/how_tos/how_to_sniff_zigbee_traffic.html)

mariusmotea, 2020-03-15:
I will try tomorrow. Thanks.

mariusmotea, 2020-03-15:
Are [these code lines](https://github.com/dresden-elektronik/deconz-rest-plugin/blob/master/sensor.cpp#L162) useful?

Koenkk, 2020-03-15:
- https://github.com/dresden-elektronik/deconz-rest-plugin/blob/master/sensor.cpp#L165 is already in 
- https://github.com/dresden-elektronik/deconz-rest-plugin/blob/master/sensor.cpp#L168 I have no clue what it means

Koenkk, 2020-03-15:
Thinking about it now, isn't dark: true just the same as occupancy: true? (as it will only send the occupancy event when it's dark enough)

In other words, do you also get dark: true when no occupancy is detected in Deconz?

baderas, 2020-03-16:
the new tradfri motion sensor supports only switching when its dark, therefore I think dark: ture, occupancy:false should work. 
We should separate the documentation for both devices, since the new one also has - sadly - a fixed 3 min intervall, whereas the old one was configurable.
See the new one: https://www.ikea.com/gb/en/p/tradfri-wireless-motion-sensor-white-70429913/
"Can be set for day or night mode."

mariusmotea, 2020-03-16:
I just pair a Tradfri Motion Sensor with latest Deconz and check his output. The `dark` parameter is no longer in the output. I think at the moment when i implement it in DiyHue it was also a copy of the logic daylight sensor from Deconz and i did not figure it.

closed 2020-03-16 by mariusmotea