Koenkk/zigbee2mqtt #3131: Zigbee2mqtt need a break between two requests.

mariusmotea, 2020-03-15:
## What happened

On every switch/sensor interaction the lights respond with ~1 second delay. I debug the problem and found that i receive the request from the switch/sensor instant and i push the mqtt message back right away to control one light, but the light state is changing after ~ 1 second. If i control the light manually it change instantly, so the issue manifest only when i try to control the light from another zigbee device. I made also a benchmark and set one zigbee switch to control two lights, one using wifi and another using zigbee. The light using zigbee start to turn on only after the wifi light is already at maximum brightness (transition take about 0.8s).

## What did you expect to happen
Small delay also on second request.

## How to reproduce it (minimal and precise)
set one zigbee switch to control one zigbee light and also a second device using different protocol (aka wifi).

## Debug Info
zigbee2mqtt version: docker (latest)
CC2531 firmware version: 1.2 latest

Koenkk, 2020-03-16:
Could you provide logging of zigbee2mqtt when this happens? (together with some description on what you did in this log).

mariusmotea, 2020-03-16:
I made [a video](https://www.youtube.com/watch?v=RjZkGG59oUY) that show exactly the delay. The light from left (near me) has a zigbee controller (from a Tradfri bulb), the light from the middle has an ESP8266 wifi controller. Both lights are in the same group and can be controller from a Hue app or from a switch ( Tradfri remote in this case). In the video we can see the zigbee light has a visible delay when i use the Tradfri remote, while on the phone it looks to respond even faster (probably default internal transition is less). I don't think the logs are very relevant but i upload them here: https://pastebin.com/Yjhagk6M. I will add that i have mixed lights in other rooms and the behavior is the same for the zigbee lights. I also notice that now when i enter in a room where i have a zigbee motion sensor the light is not turned on at the first step in that room but at the second or third step.

Koenkk, 2020-03-16:
- It's a bit hard to judge, but isn't this the default transition? 
- Could you provide the debug logging?

To enable debug logging set in `configuration.yaml`:
```yaml
advanced:
  log_level: debug
```

mariusmotea, 2020-03-16:
Here are the debug logs, bug again they miss the milliseconds value witch is important in these cases: https://pastebin.com/JS7qWqQV

Yes, from the app is working fine, also with two zigbee lights is fine ( i cannot test with more currently). From my side the issue occur only when i receive then i send. If i only send requests the lights play instantly. 

Regarding delay but outside of this issue, what is the maximum number of light requests i can send per second? Hue entertainment use 24 light frames per second, is this number possible with zigbee2mqtt? i can send rgb data or xy.

mariusmotea, 2020-03-17:
I made a new test with additional logger subscribed to zigbee2mqtt topic and saw there is a delay on lights messages received,  so i start to believe this is a limitation of mqtt or a problem in mosquito (or mosquito configuration).

#### Logger receive the Tradfri remote command
2020-03-17 12:46:**57.307** DEBUG:  Received PUBLISH (d0, q0, r0, m0), 'zigbee2mqtt/Telecomanda Baia Mica', ...  (74 bytes)
2020-03-17 12:46:**57.308** DEBUG:  Received PUBLISH (d0, q0, r0, m0), 'zigbee2mqtt/Telecomanda Baia Mica', ...  (68 bytes)

#### At the same millisecond the hue emulator receive the message
2020-03-17 12:46:**57.307** INFO:   MQTT: got state message on zigbee2mqtt/Telecomanda Baia Mica
.........
2020-03-17 12:46:**57.313** INFO:   rule 67 is triggered
2020-03-17 12:46:**57.320** DEBUG:  MQTT publish to zigbee2mqtt/Aplica Baia Mica/set {"transition": 0.3, "state": "ON"}
2020-03-17 12:46:**57.422** DEBUG:  MQTT publish to zigbee2mqtt/Oglinda Baia Mica/set {"transition": 0.3, "state": "ON"}

#### Logger receive the messages
2020-03-17 12:46:**57.786** DEBUG:  Received PUBLISH (d0, q0, r0, m0), 'zigbee2mqtt/Oglinda Baia Mica', ...  (54 bytes)
2020-03-17 12:46:**57.789** DEBUG:  Received PUBLISH (d0, q0, r0, m0), 'zigbee2mqtt/Aplica Baia Mica', ...  (54 bytes)

I notice there is also a delay of 100ms between the two lights requests that probably i can fix it, but the big delay of 350ms - 400ms is until de logger  receive the messages (and probably zigbee2mqtt receive them in the same time).

Koenkk, 2020-03-17:
You can get ms logging by setting `timestamp_format: "YYYY-MM-DD HH:mm:ss:SSS"` (https://www.zigbee2mqtt.io/information/configuration.html)

mariusmotea, 2020-03-18:
Future tests show the issue is caused by Paho MQTT library, it need some time before connection object can be reused. I will close this. Thanks.

closed 2020-03-18 by mariusmotea