Koenkk/zigbee2mqtt #3132: how to "analyze" health of network

bartplessers, 2020-03-15:
# Bug Report
## What happened
I have quite some devices, al lot of (battery powered) sensors.
Sometimes I get some unexpected behaviour, I suspect my zigbee mesh that is mes(h)sing things up.
The only way I get some grip on the network is to take a look at 
zigbee_networkmap
[image]

or
Z2MA map
[image]

The problem with both interfaces is 
- not searchable
- only for "nice" graphics, as @Koenkk mentioned somewhere

## What would you like to happen
some tool/interface/log where I could do some analysis

## Debug Info
RPI 3
CC2531 usb stick
hass.io

```
Hostname | HASSIO
System | HassOS 3.12
Deployment | production
```

Zigbee2mqtt Hass.io Add-on
```
info  2020-03-15 13:17:11: Logging to console and directory: '/share/zigbee2mqtt/log/2020-03-15.13-17-11' filename: log.txt
info  2020-03-15 13:17:13: Starting zigbee2mqtt version 1.11.0 (commit #unknown)
info  2020-03-15 13:17:13: Starting zigbee-herdsman...
info  2020-03-15 13:17:15: zigbee-herdsman started
info  2020-03-15 13:17:15: Coordinator firmware version: '{"type":"zStack12","meta":{"transportrev":2,"product":0,"majorrel":2,"minorrel":6,"maintrel":3,"revision":20190608}}'
```

carldebilly, 2020-03-18:
@bartplessers I'll be happy to implement a way to better represent the health of a network in Z2MA... :-)

You have more specific suggestions?

bartplessers, 2020-03-20:
Well, an easy one could be
https://github.com/yllibed/Zigbee2MqttAssistant/issues/219

a sortable and searchable table, like
[image]

Maybe a possibility to add/remove columns?
At least I would like to have
  - name
  - friendly name
  - brand
  - last seen
  - battery
  - linkquality

bartplessers, 2020-03-20:
Maybe it's possible to have a field to distinguish the "orphaned" items? 
[image]

bartplessers, 2020-03-20:
Maybe also a "clickable" map:
- You can click on an entity
- you will be redirected to the "datatables" view, filtered on the selected item
[image]

bartplessers, 2020-03-20:
a searchable map?
a dynamic search field where items will be highlighted if they match the search term?

bartplessers, 2020-03-20:
VERY usefull: a dynamic searchfilter on the logfile.
The zigbee2mqtt log contains a lot of lines.
It would be great to enter a searchterm (ID, "error", etc...) and to see only the lines that match the searchterm

closed 2020-05-30 by stale[bot]