Koenkk/zigbee2mqtt #3134: Paulmann ZigBee-Controller 939.99 not interviewed

julian2701, 2020-03-15:
# Bug Report
## What happened
Paulmann 939.99 (yet unsupported) is not being interviewed after factory reset (switch 5x on/off, lightbulb blinking afterwards correctly)

## What did you expect to happen
Starting interview with at least getting info of unsupported device to add it manually.

## How to reproduce it (minimal and precise)
See above (other lightbulbs are still working)

## Debug Info
zigbee2mqtt version: 1.11.0
CC253X firmware version: 20190608

Koenkk, 2020-03-16:
Make sure to check https://www.zigbee2mqtt.io/information/FAQ.html#why-does-my-device-not-or-fail-to-pair

julian2701, 2020-03-16:
Thank you very much. Working now, helpful was to shut down my other two devices, using an USB extension cable and stop-disconnect-connect-start zigbee2mqtt.

Additionally by adding the following Paulmann 939.99 is completely support by adding the following to devices.js:

   {
        zigbeeModel: ['H036-0001'],
        model: '93999',
        vendor: 'Paulmann',
        description: 'Plug Shine Zigbee Controller',
        extend: generic.light_onoff_brightness,
    },

Koenkk, 2020-03-16:
Thanks, I've added support for this device, will be supported in the latest dev branch in a few hours (or zigbee2mqtt 1.12.0 which will be released later this week).

closed 2020-03-16 by Koenkk