Koenkk/zigbee2mqtt #3136: Unable to pair Heiman HS1CG-M

magostinelli, 2020-03-16:
Hello, I use Hass.io with zigbee2mqtt.
zigbee2mqtt version: 1.1.11
CC253X firmware version: 2.6.3.20190608

I'm unable to pair the device, i push the little button inside the hole for 2 seconds (or more) led gree blink, but I can't see anything inside the zigbee2mqtt log.

I also tried to reboot, there are some tests that I can do?

Koenkk, 2020-03-16:
Make sure to check https://www.zigbee2mqtt.io/information/mqtt_topics_and_message_structure.html#zigbee2mqttbridgeconfigure

magostinelli, 2020-03-16:
How to check?
I don't have this topic.
[image]

Koenkk, 2020-03-16:
Sorry, wrong link: https://www.zigbee2mqtt.io/information/FAQ.html#why-does-my-device-not-or-fail-to-pair

magostinelli, 2020-03-17:
Hello,
I read the link, the device is very near to coordinator, and I have some other device connected without problem.

I enable the debug log as indicated in the link, but I can't see anything in the log.
I attach it.
[zigbee2mqtt.txt](https://github.com/Koenkk/zigbee2mqtt/files/4342450/zigbee2mqtt.txt)

magostinelli, 2020-03-19:
Hello,
I tryed all suggestetd here:
https://www.zigbee2mqtt.io/information/FAQ.html#why-does-my-device-not-or-fail-to-pair.
(I also try with a USB cable).

With two different new device i need to pair, without success.

Why I need to reflash the firmware if the device already paired work correctly?

Koenkk, 2020-03-19:
- Did you also try:

> Stop Zigbee2mqtt, unplug the coordinator, wait 10 seconds, plug the coordinator, start Zigbee2mqtt and try to pair the device again.

If that doesn't work we would need to sniff the traffic to see what is actually happening: https://www.zigbee2mqtt.io/how_tos/how_to_sniff_zigbee_traffic.html

magostinelli, 2020-03-20:
Yes, I have already tried the stop/disconnect method.

With sniff I must reflash the firmware, but I haven't a cc programmer, and I haven't a HUSBZB-1 stick.

Can I use more than one cc2531 on the same pi? I ordered another one.

I attach my actual zigbee devices connected and working perfectly.
[image]

magostinelli, 2020-03-20:
And this is my zigbee2mqtt conf:
`data_path: /share/zigbee2mqtt
devices: devices.yaml
groups: groups.yaml
homeassistant: true
permit_join: true
mqtt:
  base_topic: zigbee2mqtt
  server: 'mqtt://core-mosquitto'
  user: mqtt_michele
  password: NonMiTrovi12
serial:
  port: /dev/ttyACM0
advanced:
  pan_id: 6754
  channel: 11
  network_key:
    - 1
    - 3
    - 5
    - 7
    - 9
    - 11
    - 13
    - 15
    - 0
    - 2
    - 4
    - 6
    - 8
    - 10
    - 12
    - 13
  availability_blacklist: []
ban: []
whitelist: []
queue: {}
socat:
  enabled: false
  master: 'pty,raw,echo=0,link=/dev/ttyZ2M,mode=777'
  slave: 'tcp-listen:8485,keepalive,nodelay,reuseaddr,keepidle=1,keepintvl=1,keepcnt=5'
  restartdelay: 1
  initialdelay: 1
  options: '-d -d'
  log: true
`

Koenkk, 2020-03-20:
You can use multiple CC2531s on a Pi, lets wait until your sniffer arrives.

magostinelli, 2020-03-22:
Do you think i must try to reflash the firmware?

Koenkk, 2020-03-22:
Yes, please try it.

magostinelli, 2020-03-23:
Ok, a friends of mine flash the adapter for the first time (with cc debugger), I don't have a cc debugger, but I have the adapter to flash it via GPIO.

Unfortunately the utility cc_chipid doen's recognize the adapter. I have checked too many times the connection, but it doesn't work.

I used tje following guide to make the wiring:
https://lemariva.com/blog/2019/07/zigbee-flashing-cc2531-using-raspberry-pi-without-cc-debugger

magostinelli, 2020-03-24:
Yesterday, after reconnect the cc2531 and restarted zigbeetomqtt I lost three of my sensors, how is it possibile?
Maybe the problem is not the firmwware, but the zigbeetomqtt configuration files?
When i restart the service zigbee2mqttAssistant says that all the sensors was unresponsive since 6 week (instead an 1 hour), after a while some sensor sent the data and return ok.

What's appen when I pair a device, there is some check/modify that I can do on my zigbeetomqtt?

Regards.

[image]

Koenkk, 2020-03-24:
You already checked all things from https://www.zigbee2mqtt.io/information/FAQ.html#why-does-my-device-not-or-fail-to-pair right? So reflashing it the only thing left.

magostinelli, 2020-03-25:
Hello,
I managed to flah the firmware! I have used the option -m 90 and I reflashed the firmware.
And then pairing restart to work! I paired the HS1CG-E sensor, and now it stop to work, how it's possibile? Also some of my sensor doesn't work anymore..

closed 2020-05-31 by stale[bot]