Koenkk/zigbee2mqtt #3139: needed config to support for "plaform: device" (Device automations) in HA?

netweaver1970, 2020-03-17:
Hi,
I vaguely remember seeing an github issue passing by, to detail a config change to allow using device automation instead of the -previous- state automations in HA for Z2M.
I remember it was said it (by @KoenKK probably) that the support for Device automation was a breaking change so the default setting was still to genereate the old, state-compatible entities instead of the newer device automation entities.

But I can't find that post back anymore, not in the issues, not in the documentation. 
Did I dream it?

Thanks,
Geert

Koenkk, 2020-03-17:
Maybe https://github.com/Koenkk/zigbee2mqtt/issues/3033 ?

netweaver1970, 2020-03-17:
thanks. I didn't dream after all. I guess my search criteria were not specific enough.
I'll close the issue.

Keep up the good work !

closed 2020-03-17 by netweaver1970