Koenkk/zigbee2mqtt #3140: CPU locked in C2 package state

Bojan023, 2020-03-17:
# Bug Report
## What happened
**CPU package can't reach lower C-states than C2 by running a Z2M container. CPU Cores are able to reach C7. My 14 other Docker containers, like HA, don't prevent package C-state to stay in C2.**

## What did you expect to happen
**CPU package to reach lower C-state than C2 with Z2M container running.**
## How to reproduce it (minimal and precise)
Spin up Z2M Docker container and use Powertop to monitor CPU C-states

## Debug Info
zigbee2mqtt version: **Docker, 1.11.0**
**CC1352P-2** firmware version: **CC1352P_2_20191106**

closed 2020-05-23 by stale[bot]

Bojan023, 2024-09-18:
It has been a while that I’ve opened this ticket and investigated the issue. As far as I know, newer versions of Z2M have not influenced this behaviour significantly. However, my CPU Package does reach lower C States than C2 (not as much as I’d like to see). Meanwhile, my setup has changed so I cannot ensure that Z2M still contributes to the problem or is the root cause.

`PowerTOP 2.15     Overview   Idle stats   Frequency stats   Device stats   T

           Pkg(HW)  |            Core(HW) |            CPU(OS) 0
                    |                     | C0 active   1.3%
                    |                     | POLL        0.0%    0.0 ms
                    |                     | C1          0.1%    0.1 ms
C2 (pc2)   46.1%    |                     |
C3 (pc3)    0.6%    | C3 (cc3)    0.1%    | C3          0.2%    0.1 ms
C6 (pc6)   39.1%    | C6 (cc6)    2.6%    | C6          2.9%    0.6 ms
C7 (pc7)    0.0%    | C7 (cc7)   91.8%    | C7s         0.1%    1.4 ms
C8 (pc8)    0.0%    |                     | C8          5.4%    1.0 ms
C9 (pc9)    0.0%    |                     | C9          0.6%    6.7 ms
C10 (pc10)  0.0%    |                     |
                    |                     | C10        86.5%   11.7 ms
                    |                     | C1E         0.5%    0.1 ms

                    |            Core(HW) |            CPU(OS) 1
                    |                     | C0 active   1.2%
                    |                     | POLL        0.0%    0.0 ms
                    |                     | C1          0.1%    0.1 ms
                    |                     |
                    | C3 (cc3)    0.1%    | C3          0.2%    0.1 ms
                    | C6 (cc6)    4.0%    | C6          4.4%    0.6 ms
                    | C7 (cc7)   90.4%    | C7s         0.0%    1.2 ms
                    |                     | C8          7.5%    1.0 ms
                    |                     | C9          0.1%    3.6 ms
                    |                     |
                    |                     | C10        83.7%   10.3 ms
                    |                     | C1E         0.7%    0.2 ms

                    |            Core(HW) |            CPU(OS) 2
                    |                     | C0 active   0.9%
                    |                     | POLL        0.0%    0.0 ms
                    |                     | C1          0.1%    0.1 ms

<ESC> Exit | <TAB> / <Shift + TAB> Navigate |                               
`

popy2k14, 2024-09-18:
@Bojan023 thx for your answer.
I will also test it and give feedback here.

Than we can decide to reopen the issue or create a new one.

popy2k14, 2024-09-19:
@Bojan023 I have tested Z2M with an usb connected "SONOFF Zigbee 3.0 USB Dongle Plus,TI CC2652P + CP2102(N)" stick.
On unraid i have used the newest docker template and have passed the stick to the container.

After issues with the "channel" which brought my network down (because i am an idiot and forgot to merge the channel setting), the network was up and running.

Also in my case it just reached C2 with Z2M running in this configuration.
After stopping the Z2M container it immediately reached C8 again with a high percentage.

So i can confirm the issue!

How we proceed?
I have some ideas:

- open a new issue
- maybe it's an usb-serial issue that prevents reaching higher C states!?
  Do you have an ethernet zigbee stick to test?

It would be so nice to have also HA + Z2M dockerized to my i3 unraid and ditch my Dell Wyse HA box.
Let me know your thoughts.

popy2k14, 2024-09-19:
@Bojan023 I think that issue is not related to Z2M.
It seems an Intel issue, see here and the linked threads: https://community.intel.com/t5/Processors/High-power-consumption-when-using-USB1-1-serial-devices-under/td-p/1516435

It's so sad that that this issue is not fixed. 
It also happens up to gen 14.

Any ideas how to solve this

popy2k14, 2024-10-04:
So to sum it up.
I have migrated (cloned IEEE) my zigbee network (132 devices) from the sonoff to an SLZB-06 (which has the same zigbee frontend CC2652P and an PoE ethernet port) flawlessly.

The CPU now reaches C8 again!

So yes, this is a workaround and Intel has an serious USB issue there!

Here are the stats with running Z2M conrainer + SLZB-06 (PoE):

[image]

popy2k14, 2025-02-25:
Sadly i think thats an intel architecture issue which we cant solve.
The only workaround seems to be an network coordinator.