Koenkk/zigbee2mqtt #3143: Mijia presence sensor lost connection until I press the button

mr2c12, 2020-03-17:
Z2m v1.9
Coordinator: CC26X2R1

One of my motion sensor stopped working today:

> info  2020-03-17 18:33:14: MQTT publish: topic 'zigbee2mqtt/0x00158d00045161f4', payload '{"battery":91,"voltage":2985,"illuminance":21,"linkquality":66,"last_seen":"2020-03-17T17:33:14.421Z","occupancy":false,"illuminance_lux":21}'
> info  2020-03-17 18:33:14: MQTT publish: topic 'zigbee2mqtt/0x00158d00045161f4', payload '{"battery":91,"voltage":2985,"illuminance":21,"linkquality":63,"last_seen":"2020-03-17T17:33:14.486Z","occupancy":true,"illuminance_lux":21}'
> info  2020-03-17 18:34:44: MQTT publish: topic 'zigbee2mqtt/0x00158d00045161f4', payload '{"battery":91,"voltage":2985,"illuminance":21,"linkquality":63,"last_seen":"2020-03-17T17:33:14.486Z","occupancy":false,"illuminance_lux":21}'
> info  2020-03-17 18:45:39: MQTT publish: topic 'zigbee2mqtt/0x00158d00045161f4', payload '{"battery":91,"voltage":2985,"illuminance":0,"linkquality":99,"last_seen":"2020-03-17T17:45:39.193Z","occupancy":false,"illuminance_lux":0}'

At 21:00, I start sniffing and can't get any message from the sensor, even if the motion was triggered.
Any other devices nearby was working as expected.

Once I pressed the button (quickly, not triggering re-pairing) the sensor came back on the network.
I can provide the log from wireshark.
The strange part is the sequence:

- beacon request
- association request
- association response
- device announcement

While in the z2m log there is no log such "new device".

Why did it happens? This is the first time I was able to get a valid wireshark log but in the past happened already 2-3 times.
If I should upload the wireshark log, should I remove any sensitive data?

Thanks in advance!

Koenkk, 2020-03-17:
You can send the wireshark file directly to me via telegram (@koenkk)

mr2c12, 2020-03-18:
As per @Koenkk suggestions:

1. update the coordinator fw to latest versione (https://github.com/Koenkk/Z-Stack-firmware/blob/develop/coordinator/Z-Stack_3.x.0/bin/CC26X2R1_20200312.zip)
2. re-paired the sensor so that is associated to a different router (it used to be connected to a philips bulb)

Will report any update.

mr2c12, 2020-04-04:
After 16 days, no problem so far.
Given a new network graph, the sensor seems to be still connected to the same router.
I can't say if 1. or 2. helped but the problem never occurred again.

closed 2020-04-04 by mr2c12