Koenkk/zigbee2mqtt #3145: IKEA L1527 not publishing MQTT state

Dielee, 2020-03-18:
# Bug Report
## What happened

Paired Ikea Button not publishing mqtt state of paired IKEA L1527.
Lamp turns on, but in HA it stays 'off'.

Log:
```
zigbee2mqtt:info  2020-03-18 10:37:11: MQTT publish: topic 'zigbee2mqtt/KucheButton/action', payload 'toggle'
zigbee2mqtt:info  2020-03-18 10:37:11: MQTT publish: topic 'zigbee2mqtt/KucheButton', payload '{"linkquality":105,"action":"toggle"}'
zigbee2mqtt:info  2020-03-18 10:37:11: MQTT publish: topic 'zigbee2mqtt/KucheButton', payload '{"linkquality":105,"action":""}'
```

## What did you expect to happen
Publish mqtt message with payload 'on'
## How to reproduce it (minimal and precise)

## Debug Info
zigbee2mqtt version: Newest from master brach, also tested dev
CC253X firmware version: Newest

nldroid, 2020-03-18:
Do you have reporting on?

https://www.zigbee2mqtt.io/information/report.html

Dielee, 2020-03-18:
Sure, yes.. other bulbs and big panels working well.

nldroid, 2020-03-25:
When you restart Zigbee2mqtt, do you see it in the log?

```
Mar 25 18:44:38 nuc npm[30820]: zigbee2mqtt:info  2020-03-25 18:44:38: Successfully setup reporting for '0x00178801026e5c9d' - 11 - genOnOff
Mar 25 18:44:38 nuc npm[30820]: zigbee2mqtt:debug 2020-03-25 18:44:38: Setup reporting for '0x00178801026e5c9d' - 11 - genLevelCtrl

```

closed 2020-06-01 by stale[bot]