Koenkk/zigbee2mqtt #3146: No matching payload found for entity: Ikea Plug_update_available with state topic: zigbee2mqtt/Ikea Plug. Payload: , with value template Template("{{ value_json.update_available}}")

TASSDevon, 2020-03-18:
# Bug Report
Getting quite a few of these messages for my ikea plugs:
No matching payload found for entity: Ikea Plug_update_available with state topic: zigbee2mqtt/Ikea Plug. Payload: , with value template Template("{{ value_json.update_available}}")

## How to reproduce it (minimal and precise)
I just updated z2m to the latest version

## Debug Info
zigbee2mqtt version:  1.11.0

Koenkk, 2020-03-18:
The `update_available` property is available after the device checked for an OTA update once. This will probably happen in 24 hours from now. You can force it by executing the check yourself: https://www.zigbee2mqtt.io/information/ota_updates.html#check-if-an-update-is-available

CodeFinder2, 2020-03-19:
@Koenkk Would there be a way to avoid such warnings completely? I am also getting a lot of them at startup now and have seen them in the past for an Aqara Water Leak sensor as already mentioned [here](https://github.com/Koenkk/zigbee2mqtt/issues/1015). After the update to v1.12.0 (or maybe even already to v1.11.0 -- can't remember), there are quite a lot of them spamming my logs. 🙈

Koenkk, 2020-03-20:
@CodeFinder2 good idea, added in the latest dev branch, let me know if it works.

CodeFinder2, 2020-03-26:
@Koenkk great, thanks! Would it be possible to also assume `false` for the Xiaomi Aqara Water Leak sensor (for the same reason, see #1015 ) until a real value was provided, i.e., to have a default (false) there as well? Would be great!

I just paired<sup>1</sup> another water leak sensor and Z2M reports (via MQTT):
```
zigbee2mqtt:info  2020-03-26 21:37:51: MQTT publish: topic 'zigbee2mqtt/basement/ground', payload '{"battery":100,"voltage":3055,"linkquality":138,"last_seen":"2020-03-26T20:37:51.932Z"}'
```
As you can see, no "water_leak" attribute is present.  I think it's totaly safe to assume false unless the sensor provides an actual value.

1: Pairing Xiaomi devices is soooo great in Z2M now, reallly cool! Great work! :-)

Koenkk, 2020-03-26:
@CodeFinder2 good idea, implemented in the latest dev branch, can you check if it works?

CodeFinder2, 2020-03-27:
Will definitely test ASAP when I find a free time slot - sorry for the delay!

Koenkk, 2020-03-27:
It is, just remove and install edge addon.

CodeFinder2, 2020-03-27:
Ah good to know and it should be safe to:
- stop the non-edge addon version
- copy the entire config dir to a new dir (e.g., `z2m-edge/`)
- start z2m-edge addon with that new config dir and the exact same config like the non-edge addon
- after testing, stop z2m-edge and start z2m addon for normal operation again?

So, no changes in PAN ID or network key...safe to go this way?  Don't wanna mess up my setup :see_no_evil:

Koenkk, 2020-03-27:
yes!

CodeFinder2, 2020-03-27:
Hm, any way to check if it's really using your latest dev commits? It says:
```
info  2020-03-28 00:00:36: Starting zigbee2mqtt version 1.12.0-dev (commit #38cb519d3cdeb9503620f47e6a81e250a16e0765)
```

And that commit hash refers to z2m addon commit 9 days ago...

Koenkk, 2020-03-28:
@CodeFinder2 it was not indeed not updating anymore, fixed now, new build should be up in 4 hours from now.

CodeFinder2, 2020-03-28:
Thank you so much, will test when it's online!

Yeah, I had this feeling that it might not be updated anymore ... :see_no_evil:

CodeFinder2, 2020-03-28:
@Koenkk nonetheless, how can I check if its using your latest dev branch?

CodeFinder2, 2020-03-29:
> @CodeFinder2 good idea, implemented in the latest dev branch, can you check if it works?

Bingo:
```
info  2020-03-29 00:32:14: Starting zigbee2mqtt version 1.12.0-dev (commit #52d4193bc714a52a52e30801d2ab15d07695886b)
[...]
info  2020-03-29 00:32:50: MQTT publish: topic 'zigbee2mqtt/basement/ground', payload '{"battery":100,"voltage":3025,"linkquality":138,"last_seen":"2020-03-28T23:32:50.677Z","water_leak":false}'
```
And this sensor has never seen any water yet, so, according to my understanding, it must be due to your fix! Plus: it doesn't show the `"water_leak":false` if I am going back to v1.12.0. :sunglasses: 

My HA logs are clean (regarding the `update_available` *and* the `water_leak` payload warnings) as well now (also tested with v1.12.0 vs latest dev).

Again: great work, very appreciated! Thank you!

Koenkk, 2020-03-29:
Thanks, ill close this

closed 2020-03-29 by Koenkk