Koenkk/zigbee2mqtt #3147: Ikea LED1732G11 changing white balance tone on its own?!

fabsenet, 2020-03-18:
I have 2 Ikea bulbs: `IKEA TRADFRI LED bulb E27 1000 lumen, dimmable, white spectrum, opal white (LED1732G11)` which have a ghost behavior: I usually drive them on cold white but sometimes one of them switches to warm white and back. sometimes there are minutes between it, sometimes only seconds. I even observed this while the PI with Z2M, NodeRed and the Zigbee-Stick are powered down!

I recently switched to the dev version of Z2M and used the OTA update for the bulbs. The problem was there before and stayed that way.

I already tried repairing the bulbs through the 6x on/off thing, no change.

I really do not think this is an issue with Z2M but I guess you guys here are the ones who might know, which I should try next.

Could this be some kind of interference at any chance? should I change the zigbee auth key or channel?

can you give any advise what I should try next?

Koenkk, 2020-03-18:
In order to figure out what and why this is happening, you should sniff the traffic at the moment it changes color: https://www.zigbee2mqtt.io/how_tos/how_to_sniff_zigbee_traffic.html

fabsenet, 2020-06-07:
needed to get an additional receiver and it took very long this time.

I can now sniff the zigbee traffic and i can verify, that it looks okay to me. I have two identical lights, send them the identical msgs and one changes the color temp, one does nothing.

I have no idea how to proceed from here?
[image]

Flashing the stick was fun...turns out you do not need the ground pin if the usb is connected.
[image]

Koenkk, 2020-06-09:
Could you share the sniff here and provide the `No.` of the messages send?

fabsenet, 2020-06-11:
I did send 3 mqtt msgs to the first light changing the color_temp only:
commands:
```json
{"color_temp":450}
...
{"color_temp":350}
...
{"color_temp":250}
```

The first light was on a warm white and stayed that way the whole time, no reaction at all. then I sent basically the same 3 commands to another identical ikea light and that works as expected.

The trace for this 2x3 messages is: [light_no_reaction.pcapng.gz](https://github.com/Koenkk/zigbee2mqtt/files/4763660/light_no_reaction.pcapng.gz)

----

I tried disconnecting the first lamp from power for like a minute and it did not make it listen to color-temp commands either.

Later, it started changing its white balance on its own again. There were like 3 or 4 changes between msgs 11 and 13 in this trace (no direct msg relation): [reaction_without_commands.pcapng.gz](https://github.com/Koenkk/zigbee2mqtt/files/4763671/reaction_without_commands.pcapng.gz)
The second identical light stayed as it should the whole time again.

----

Sometimes the first light also works as expected, but currently it does not. I can make another capture if I can time it.

Koenkk, 2020-06-11:
- Can you share your network key so I can decrypt the traffic?
- Could it be that somehow the bulb is broken? Do you have another bridge to test it with (preferably the ikea bridge)

fabsenet, 2020-06-11:
"22:c9:50:76:d2:12:14:b3:9f:99:56:f9:e0:8d:6b:ee"

please do me a favor and delete this post if you have a copy of this

fabsenet, 2020-06-11:
My guess is really the bulb might be broken. The only thing I can think of might be me sending rubbish data and permanently overwriting some internal memory or something.

closed 2020-07-26 by stale[bot]