Koenkk/zigbee2mqtt #3148: Error: SRSP - SYS - ping after 6000ms until it eventually recovers itself?

copystring, 2020-03-18:
# Bug Report
I started to notice, especially recently, my CC2652R to either crash after a few days (which is not my main problem) and not starting up with the known "Error: Failed to connect to the adapter (Error: SRSP - SYS - ping after 6000ms)".

Eventually zigbee2mqtt stop trying to start the adapter so a lot of the time it stays that way for example for a whole night/day until I notice.

As far as I know this error means there is already an existing network with the same pan_id (right?).

Interestingly enough executing systemctl restart zigbee2mqtt and observing a lot tries it eventually starts and works just fine.

Not sure where to start to address this issue. Not sure if it is a bug, broken hardware or some other issue.
Here is the log: https://pastebin.com/wNuAxLZ8
Also: 
[image]

I looked here: https://www.zigbee2mqtt.io/information/FAQ.html#help-zigbee2mqtt-fails-to-start
Uninstalled modemmanager
Pressed reset next to the usb port.
Restarted ubuntu.
Unplugged and plugged the device back in.

I do not know where to go from here. What else can I try?
Looking at other issue with the same/similar issue it seems it cannot be solved?
Currently it does not start at all anymore even after 50 or so tries (maybe it recovers later).
I read somewhere that a router/bulb/plug can prevent starting the device? I have a lot of devices and 20 or more routers like bulbs etc.

## Debug Info
zigbee2mqtt version:1.12-0-dev (happens for ages though, I had the same with 1.11, 1.10, etc ...)
CC253X firmware version: CC26X2R1_20191106.zip (same as above, the previous version had this problem too)

Koenkk, 2020-03-18:
> As far as I know this error means there is already an existing network with the same pan_id (right?).

Definitely not in this case

Could you provide the logging up to the crash? (with `DEBUG=zigbee-herdsman:*`)
This would help to identify wether the CC2652R is the issue or if it's a connectivity issue.

copystring, 2020-03-18:
Yeah sure. It will probably take a few days. For now it is running again. I will report back once it crashed.

hughb8on, 2020-04-01:
I have seen failures with this message following server restarts. Following the restart, zigbee2mqtt fails to start until I unplug/replug the CC265R2, after which it is fine. 

This has happened on both of the two occasions my server has been restarted since I started using a CC265R2.

Here is the zigbee-herdsman debug log: -

DEBUG=zigbee-herdsman* npm start

> zigbee2mqtt@1.11.0 start /home/********/zigbee2mqtt
> node index.js

zigbee2mqtt:info  2020-04-01 13:29:33: Logging to console and directory: '/var/log/zigbee2mqtt' filename: log.txt
zigbee2mqtt:debug 2020-04-01 13:29:33: Loaded state from file /home/********/zigbee2mqtt/data/state.json
zigbee2mqtt:info  2020-04-01 13:29:33: Starting zigbee2mqtt version 1.11.0 (commit #unknown)
zigbee2mqtt:info  2020-04-01 13:29:33: Starting zigbee-herdsman...
zigbee2mqtt:debug 2020-04-01 13:29:33: Using zigbee-herdsman with settings: '{"network":{"panID":6758,"extendedPanID":[221,221,221,221,221,221,221,221],"channelList":[11],"networkKey":"HIDDEN"},"databasePath":"/home/********/zigbee2mqtt/data/database.db","databaseBackupPath":"/home/********/zigbee2mqtt/data/database.db.backup","backupPath":"/home/********/zigbee2mqtt/data/coordinator_backup.json","serialPort":{"baudRate":115200,"rtscts":true,"path":"/dev/ttyUSB-CC2652_00"}}'
  zigbee-herdsman:adapter Path '/dev/ttyUSB-CC2652_00' is valid for 'ZStackAdapter' +0ms
  zigbee-herdsman:controller:log Starting with options '{"network":{"networkKeyDistribute":false,"networkKey":[********],"panID":6758,"extendedPanID":[221,221,221,221,221,221,221,221],"channelList":[11]},"serialPort":{"baudRate":115200,"rtscts":true,"path":"/dev/ttyUSB-CC2652_00"},"databasePath":"/home/********/zigbee2mqtt/data/database.db","databaseBackupPath":"/home/********/zigbee2mqtt/data/database.db.backup","backupPath":"/home/********/zigbee2mqtt/data/coordinator_backup.json"}' +0ms
  zigbee-herdsman:adapter:zStack:znp:log Opening SerialPort with /dev/ttyUSB-CC2652_00 and {"baudRate":115200,"rtscts":true,"autoOpen":false} +0ms
  zigbee-herdsman:adapter:zStack:znp:log Serialport opened +1ms
  zigbee-herdsman:adapter:zStack:znp:log Writing skip bootloader payload +1ms
  zigbee-herdsman:adapter:zStack:unpi:writer --> buffer [239] +0ms
  zigbee-herdsman:adapter:zStack:znp:SREQ --> SYS - ping - {"capabilities":1} +0ms
  zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,0,33,1,32] +1s
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,28,68,129,0,0,0,4] +0ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,28,68,129,0,0,0,4] +0ms
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [62,114,1,1,0,114,0,51] +0ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,28,68,129,0,0,0,4,62,114,1,1,0,114,0,51] +0ms
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [183,0,0,0,8,24,76,10] +0ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,28,68,129,0,0,0,4,62,114,1,1,0,114,0,51,183,0,0,0,8,24,76,10] +1ms
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [0,0,33,64,0,222,135,28] +0ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,28,68,129,0,0,0,4,62,114,1,1,0,114,0,51,183,0,0,0,8,24,76,10,0,0,33,64,0,222,135,28] +0ms
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [21,254,27,68,129,0,0,6] +1ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,28,68,129,0,0,0,4,62,114,1,1,0,114,0,51,183,0,0,0,8,24,76,10,0,0,33,64,0,222,135,28,21,254,27,68,129,0,0,6] +0ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 28 - 2 - 4 - 129 - [0,0,0,4,62,114,1,1,0,114,0,51,183,0,0,0,8,24,76,10,0,0,33,64,0,222,135,28] - 21 +0ms
  zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - incomingMsg - {"groupid":0,"clusterid":1024,"srcaddr":29246,"srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"linkquality":114,"securityuse":0,"timestamp":46899,"transseqnumber":0,"len":8,"data":{"type":"Buffer","data":[24,76,10,0,0,33,64,0]}} +0ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,27,68,129,0,0,6] +2ms
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [4,62,114,1,1,0,114,0,65,185,0,0,0,7,24,77] +0ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,27,68,129,0,0,6,4,62,114,1,1,0,114,0,65,185,0,0,0,7,24,77] +0ms
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [10,0,0,24,1,222,135,28] +0ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,27,68,129,0,0,6,4,62,114,1,1,0,114,0,65,185,0,0,0,7,24,77,10,0,0,24,1,222,135,28] +0ms
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [30] +1ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,27,68,129,0,0,6,4,62,114,1,1,0,114,0,65,185,0,0,0,7,24,77,10,0,0,24,1,222,135,28,30] +0ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 27 - 2 - 4 - 129 - [0,0,6,4,62,114,1,1,0,114,0,65,185,0,0,0,7,24,77,10,0,0,24,1,222,135,28] - 30 +0ms
  zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - incomingMsg - {"groupid":0,"clusterid":1030,"srcaddr":29246,"srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"linkquality":114,"securityuse":0,"timestamp":47425,"transseqnumber":0,"len":7,"data":{"type":"Buffer","data":[24,77,10,0,0,24,1]}} +2ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +0ms
zigbee2mqtt:error 2020-04-01 13:29:41: Error while starting zigbee-herdsman
zigbee2mqtt:error 2020-04-01 13:29:41: Failed to start zigbee
zigbee2mqtt:error 2020-04-01 13:29:41: Exiting...
zigbee2mqtt:error 2020-04-01 13:29:41: Error: Failed to connect to the adapter (Error: SRSP - SYS - ping after 6000ms)
[... cut ...]
    at rejected (/home/********/zigbee2mqtt/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/zStackAdapter.js:6:65)
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! zigbee2mqtt@1.11.0 start: `node index.js`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the zigbee2mqtt@1.11.0 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/********/.npm/_logs/2020-04-01T12_29_41_133Z-debug.log

Koenkk, 2020-04-01:
@hughb8on this is a known issue, on some systems the reset button has to be pressed after restarting the system. Currently there is not solution for it.

copystring, 2020-04-01:
I finally managed to create the required log file which is almost 1.7 GB. Should I upload it to google drive and link it here?

copystring, 2020-04-01:
Here is the log: https://drive.google.com/file/d/1GUqCnsKrXku_6TidzhfSceU9vhdyMo52/view?usp=sharing

At the same time in dmesg it shows this:
https://pastebin.com/qaFGvp2n

So it seems the device actually reconnect/restarts

I noticed this when I came home from work about 14 hours later after it stopped working.
Starting zigbee2mqtt after that shows no error. However in the past that was not always the case. It eventually recovers itself. Other times it never recovered without me helping the device by re-plugging and resetting

Koenkk, 2020-04-01:
It indeed seems that the device is disconnected somehow

```
2020-04-01T05:46:43.824Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - dataConfirm - {"status":0,"endpoint":1,"transid":128}
2020-04-01T05:46:43.824Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2020-04-01T05:46:44.401Z zigbee-herdsman:adapter:zStack:znp:log Port closed
2020-04-01T05:46:44.402Z zigbee-herdsman:controller:log Adapter disconnected'
[31mzigbee2mqtt:error[39m 2020-04-01 07:46:44: Adapter disconnected, stopping
```

- What is your system setup?
- How is the CC26XR1 connected, through a hub? Long usb cable etc.
- Can you add the following udev rules: https://github.com/Koenkk/zigbee-herdsman/issues/72#issuecomment-599688536
    - To find out the correct `idVendor` and `idProduct` execute `lsusb`
    - Given the following example:

```
ubuntu@j3455:~$ lsusb
Bus 001 Device 002: ID 1a86:7523 QinHeng Electronics HL-340 USB-Serial adapter
```

It should be

```
ATTRS{idVendor}=="1a86", ATTRS{idProduct}=="7523", ENV{ID_MM_DEVICE_IGNORE}="1"
ATTRS{idVendor}=="1a86", ATTRS{idProduct}=="7523", ENV{MTP_NO_PROBE}="1"
```

copystring, 2020-04-02:
> * What is your system setup?

Ubuntu 18.04 with latest updates running a industry mini itx board (do you need more info? let me know)

> * How is the CC26XR1 connected, through a hub? Long usb cable etc.

I use the cable that came with the device. No hub. I will try with a different, shorter cable.

> * Can you add the following udev rules: [Koenkk/zigbee-herdsman#72 (comment)]

I have added this to the udev rules and double check it is applied:
[image]

I guess I will now just run this again to create a new log?

Koenkk, 2020-04-02:
Yes please try again

diogopms, 2020-04-06:
I'm having the same issue:

```
zigbee2mqtt:info  2020-04-06 22:25:48: Starting zigbee2mqtt version 1.12.0 (commit #840b9d9)
zigbee2mqtt:info  2020-04-06 22:25:48: Starting zigbee-herdsman...
zigbee2mqtt:debug 2020-04-06 22:25:48: Using zigbee-herdsman with settings: '{"network":{"panID":6754,"extendedPanID":[221,221,221,221,221,221,221,221],"channelList":[11],"networkKey":"HIDDEN"},"databasePath":"/opt/zigbee2mqtt/data/database.db","databaseBackupPath":"/opt/zigbee2mqtt/data/database.db.backup","backupPath":"/opt/zigbee2mqtt/data/coordinator_backup.json","serialPort":{"baudRate":115200,"rtscts":true,"path":"/dev/ttyACM0"}}'
  zigbee-herdsman:adapter Path '/dev/ttyACM0' is valid for 'ZStackAdapter' +0ms
  zigbee-herdsman:controller:log Starting with options '{"network":{"networkKeyDistribute":false,"networkKey":[1,3,5,7,9,11,13,15,0,2,4,6,8,10,12,13],"panID":6754,"extendedPanID":[221,221,221,221,221,221,221,221],"channelList":[11]},"serialPort":{"baudRate":115200,"rtscts":true,"path":"/dev/ttyACM0"},"databasePath":"/opt/zigbee2mqtt/data/database.db","databaseBackupPath":"/opt/zigbee2mqtt/data/database.db.backup","backupPath":"/opt/zigbee2mqtt/data/coordinator_backup.json"}' +0ms
  zigbee-herdsman:adapter:zStack:znp:log Opening SerialPort with /dev/ttyACM0 and {"baudRate":115200,"rtscts":true,"autoOpen":false} +0ms
  zigbee-herdsman:adapter:zStack:znp:log Serialport opened +10ms
  zigbee-herdsman:adapter:zStack:znp:log Writing skip bootloader payload +0ms
  zigbee-herdsman:adapter:zStack:unpi:writer --> buffer [239] +0ms
  zigbee-herdsman:adapter:zStack:znp:SREQ --> SYS - ping - {"capabilities":1} +0ms
  zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,0,33,1,32] +1s
zigbee2mqtt:error 2020-04-06 22:25:56: Error while starting zigbee-herdsman
zigbee2mqtt:error 2020-04-06 22:25:56: Failed to start zigbee
zigbee2mqtt:error 2020-04-06 22:25:56: Exiting...
zigbee2mqtt:error 2020-04-06 22:25:56: Error: Failed to connect to the adapter (Error: SRSP - SYS - ping after 6000ms)
    at ZStackAdapter.<anonymous> (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/zStackAdapter.js:71:23)
    at Generator.throw (<anonymous>)
    at rejected (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/zStackAdapter.js:6:65)
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! zigbee2mqtt@1.12.0 start: `node index.js`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the zigbee2mqtt@1.12.0 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/diogopms/.npm/_logs/2020-04-06T21_25_56_646Z-debug.log
diogopms@udoo:/opt/zigbee2mqtt$
```

Node: v12.16.1
NPM: 6.13.4
OS: Linux Ubuntu x86 - Ubuntu 18.04.4 LTS (GNU/Linux 5.3.0-45-generic x86_64)

copystring, 2020-04-12:
I have it running for 10 days straight now with a 2.2 GB log. So far no hiccups.

Is it OK if I close this for now and open it in case it comes back?

Koenkk, 2020-04-13:
yes

closed 2020-04-13 by copystring

arcstar7, 2020-04-22:
Just figured I'd add, currently experiencing similar issue with CC2531.  
However, same USB device running on a different RPi, with standalone mosquitto and zigbee2mqtt, works fine.  Homeassistant is the only place it seems to be having this issue.

ayasystems, 2020-10-07:
I have the same problem

Only works pressing the button but if restart need other time press the button.....

pi@domotica:/opt/zigbee2mqtt $ DEBUG=zigbee-herdsman* npm start

> zigbee2mqtt@1.15.0 start /opt/zigbee2mqtt
> node index.js

Zigbee2MQTT:info  2020-10-07 13:40:04: Logging to console and directory: '/opt/zigbee2mqtt/data/log/2020-10-07.13-40-03' filename: log.txt
Zigbee2MQTT:debug 2020-10-07 13:40:05: Can't load state from file /opt/zigbee2mqtt/data/state.json (doesn't exist)
Zigbee2MQTT:info  2020-10-07 13:40:05: Starting Zigbee2MQTT version 1.15.0 (commit #e30b257)
Zigbee2MQTT:info  2020-10-07 13:40:05: Starting zigbee-herdsman...
Zigbee2MQTT:debug 2020-10-07 13:40:05: Using zigbee-herdsman with settings: '{"adapter":{"concurrent":null},"backupPath":"/opt/zigbee2mqtt/data/coordinator_backup.json","databaseBackupPath":"/opt/zigbee2mqtt/data/database.db.backup","databasePath":"/opt/zigbee2mqtt/data/database.db","network":{"channelList":[11],"extendedPanID":[221,221,221,221,221,221,221,221],"networkKey":"HIDDEN","panID":6754},"serialPort":{"path":"/dev/ttyACM0"}}'
  zigbee-herdsman:adapter Path '/dev/ttyACM0' is valid for 'ZStackAdapter' +0ms
  zigbee-herdsman:controller:log Starting with options '{"network":{"networkKeyDistribute":false,"networkKey":[1,3,5,7,9,11,13,15,0,2,4,6,8,10,12,13],"panID":6754,"extendedPanID":[221,221,221,221,221,221,221,221],"channelList":[11]},"serialPort":{"path":"/dev/ttyACM0"},"databasePath":"/opt/zigbee2mqtt/data/database.db","databaseBackupPath":"/opt/zigbee2mqtt/data/database.db.backup","backupPath":"/opt/zigbee2mqtt/data/coordinator_backup.json","adapter":{"concurrent":null}}' +0ms
  zigbee-herdsman:adapter:zStack:znp:log Opening SerialPort with /dev/ttyACM0 and {"baudRate":115200,"rtscts":false,"autoOpen":false} +0ms
  zigbee-herdsman:adapter:zStack:znp:log Serialport opened +16ms
  zigbee-herdsman:adapter:zStack:znp:log Writing skip bootloader payload +462ms
  zigbee-herdsman:adapter:zStack:unpi:writer --> buffer [239] +0ms
  zigbee-herdsman:adapter:zStack:znp:SREQ --> SYS - ping - {"capabilities":1} +0ms
  zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,0,33,1,32] +1s
  zigbee-herdsman:adapter:zStack:znp:SREQ --> SYS - ping - {"capabilities":1} +6s
  zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,0,33,1,32] +6s
  zigbee-herdsman:adapter:zStack:znp:SREQ --> SYS - ping - {"capabilities":1} +6s
  zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,0,33,1,32] +6s
Zigbee2MQTT:error 2020-10-07 13:40:25: Error while starting zigbee-herdsman
Zigbee2MQTT:error 2020-10-07 13:40:25: Failed to start zigbee
Zigbee2MQTT:error 2020-10-07 13:40:25: Exiting...
Zigbee2MQTT:error 2020-10-07 13:40:25: Error: Failed to connect to the adapter (Error: SRSP - SYS - ping after 6000ms)
    at ZStackAdapter.<anonymous> (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/zStackAdapter.js:92:31)
    at Generator.throw (<anonymous>)
    at rejected (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/zStackAdapter.js:25:65)
npm ERR! code ELIFECYCLE

Any help?

pgaulon, 2020-10-13:
Hello, I also got the same problem after some reboot. However [using systemctl with `Restart=always`](https://www.zigbee2mqtt.io/getting_started/running_zigbee2mqtt.html#5-optional-running-as-a-daemon-with-systemctl) as described on the documentation is enough to make it recover. In my case systemctl tries for about 5 times, in around 2min span.