Koenkk/zigbee2mqtt #3150: 404 Issue on 1.12 update attempt in Home Assistant

Frankynov, 2020-03-19:
Hello !
I'm trying to install the brand new 1.12 update in Home Assistant, but unfortunately when I click on the upgrade button, I have this error in the logs : 

Can't install dwelch2101/zigbee2mqtt-amd64:1.12.0 -> 404 Client Error: Not Found ("manifest for dwelch2101/zigbee2mqtt-amd64:1.12.0 not found: manifest unknown: manifest unknown").

Something maybe missing ?
I'm running an Intel NUC, not a Raspberry Pi if that matters.

Thanks !

Koenkk, 2020-03-19:
The build is still running so this issue will probably resolve itself in a few hours, if after a few hours its still there please open an issue here: https://github.com/danielwelch/hassio-zigbee2mqtt/issues

Frankynov, 2020-03-19:
It is now available indeed :) Was too impatient to try it I guess !
I'm closing "issue" now. Thanks for this awesome addon which is really a joy to use !

closed 2020-03-19 by Frankynov