Koenkk/zigbee2mqtt #3151: IKEA E1524/E1810 battery not reporting

pplucky, 2020-03-19:
# Bug Report
## What happened
No battery is reported from IKEA remote E1524/E1810

## What did you expect to happen
Battery reported from IKEA remote E1524/E1810 every time the remote is used

## How to reproduce it (minimal and precise)
When zigbee2mqtt is started:
`zigbee2mqtt:info  2020-03-19 08:50:00: MQTT publish: topic 'zigbee2mqtt/ikea_comando_sala', payload '{"linkquality":52,"update_available":true,"last_seen":"2020-03-19T01:26:35.419Z"}'
zigbee2mqtt:info  2020-03-19 08:50:00: MQTT publish: topic 'zigbee2mqtt/ikea_comando_sala/availability', payload 'online'`

Clicking the buttons in the remote, to get the following MQTT publishing:
```
zigbee2mqtt:debug 2020-03-19 09:06:33: Received Zigbee message from 'ikea_comando_sala', type 'commandTradfriArrowSingle', cluster 'genScenes', data '{"value":257,"value2":13}' from endpoint 1 with groupID 45740
zigbee2mqtt:info  2020-03-19 09:06:33: MQTT publish: topic 'zigbee2mqtt/ikea_comando_sala', payload '{"linkquality":47,"update_available":true,"last_seen":"2020-03-19T09:06:33.861Z","action":"arrow_left_click"}'
zigbee2mqtt:debug 2020-03-19 09:06:35: Received Zigbee message from 'ikea_comando_sala', type 'commandTradfriArrowHold', cluster 'genScenes', data '{"value":3328}' from endpoint 1 with groupID 45740
zigbee2mqtt:info  2020-03-19 09:06:35: MQTT publish: topic 'zigbee2mqtt/ikea_comando_sala', payload '{"linkquality":44,"update_available":true,"last_seen":"2020-03-19T09:06:35.512Z","action":"arrow_right_hold"}'
zigbee2mqtt:debug 2020-03-19 09:06:36: Device 'ikea_comando_sala' announced itself
zigbee2mqtt:debug 2020-03-19 09:06:36: Device 'ikea_comando_sala' announced itself
zigbee2mqtt:debug 2020-03-19 09:06:41: Received Zigbee message from 'ikea_comando_sala', type 'commandStepWithOnOff', cluster 'genLevelCtrl', data '{"stepmode":0,"stepsize":43,"transtime":5}' from endpoint 1 with groupID 45740
zigbee2mqtt:info  2020-03-19 09:06:41: MQTT publish: topic 'zigbee2mqtt/ikea_comando_sala', payload '{"linkquality":60,"update_available":true,"last_seen":"2020-03-19T09:06:41.381Z","action":"brightness_up_click"}'
zigbee2mqtt:debug 2020-03-19 09:06:43: Received Zigbee message from 'ikea_comando_sala', type 'commandStep', cluster 'genLevelCtrl', data '{"stepmode":1,"stepsize":43,"transtime":5}' from endpoint 1 with groupID 45740
zigbee2mqtt:info  2020-03-19 09:06:43: MQTT publish: topic 'zigbee2mqtt/ikea_comando_sala', payload '{"linkquality":63,"update_available":true,"last_seen":"2020-03-19T09:06:43.744Z","action":"brightness_down_click"}'
zigbee2mqtt:debug 2020-03-19 09:06:44: Received Zigbee message from 'ikea_comando_sala', type 'commandToggle', cluster 'genOnOff', data '{}' from endpoint 1 with groupID 45740
zigbee2mqtt:info  2020-03-19 09:06:44: MQTT publish: topic 'zigbee2mqtt/ikea_comando_sala', payload '{"linkquality":60,"update_available":true,"last_seen":"2020-03-19T09:06:44.780Z","action":"toggle"}'
```
During OTA update checks, also noticed this for this remote:
```
zigbee2mqtt:debug 2020-03-19 09:08:03: Received Zigbee message from 'ikea_comando_sala', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":1,"manufacturerCode":4476,"imageType":4545,"fileVersion":304170354}' from endpoint 1 with groupID 0
zigbee2mqtt:debug 2020-03-19 09:08:03: No converter available for 'E1524/E1810' with cluster 'genOta' and type 'commandQueryNextImageRequest' and data '{"fieldControl":1,"manufacturerCode":4476,"imageType":4545,"fileVersion":304170354}'
zigbee2mqtt:debug 2020-03-19 09:08:03: Check if update available for '0x90fd9ffffe5c0c56' (TRADFRI remote control)
zigbee2mqtt:debug 2020-03-19 09:08:04: Is new image available for '0x90fd9ffffe5c0c56', current '{"fieldControl":1,"manufacturerCode":4476,"imageType":4545,"fileVersion":304170354}', latest meta '{"fileVersion":587286065,"fileSize":182590,"url":"https://fw.ota.homesmart.ikea.net/Tradfri_OTA_release_signed_2020_01_17_124857/bin/159699-5.1-TRADFRI-remote-control-2.3.014.ota.ota.signed"}'
zigbee2mqtt:debug 2020-03-19 09:08:04: Update available for '0x90fd9ffffe5c0c56': YES
```

## Debug Info
zigbee2mqtt version: 1.11.0
CC253X firmware version: `{"type":"zStack12","meta":{"transportrev":2,"product":0,"majorrel":2,"minorrel":6,"maintrel":3,"revision":20190425}}`
Using zigbee2mqtt via hass.io add-on

## Manual sensor configuration
```
  - platform: mqtt
    name: 'Comando IKEA Sala'
    state_topic: "zigbee2mqtt/ikea_comando_sala"
    availability_topic: "zigbee2mqtt/bridge/state"
    icon: "mdi:gesture-double-tap"
    value_template: "{{ value_json.action }}"
    force_update: true
    json_attributes_topic: "zigbee2mqtt/ikea_comando_sala"

  - platform: mqtt
    name: 'Battery Comando IKEA Sala'
    state_topic: "zigbee2mqtt/ikea_comando_sala"
    availability_topic: "zigbee2mqtt/bridge/state"
    unit_of_measurement: "%"
    device_class: "battery"
    value_template: "{{ value_json.battery }}"

  - platform: mqtt
    name: 'Link Comando IKEA Sala'
    state_topic: "zigbee2mqtt/ikea_comando_sala"
    availability_topic: "zigbee2mqtt/bridge/state"
    unit_of_measurement: "lqi"
    value_template: "{{ value_json.linkquality }}"
```

[image]

I think I remember seeing this working in the past, but I cannot be 100% sure anymore! 
I checked with a friend using same remote and he seems to recall it used to work, but it also doesn't work now for him.

Is there anything else I can provide to help troubleshoot, or should I try something else on my setup?

Kryzek, 2020-03-19:
Battery information will get reported periodically, usually within 24 hours. It should appear by it's own.

pplucky, 2020-03-19:
> Battery information will get reported periodically, usually within 24 hours. It should appear by it's own.
> For the "No converter available for ... with cluster 'genOta' " refer [Koenkk/zigbee-herdsman-converters#1015](https://github.com/Koenkk/zigbee-herdsman-converters/issues/1015)

It has been running for weeks without reporting battery

Kryzek, 2020-03-19:
You could try to reconfigure: https://www.zigbee2mqtt.io/information/mqtt_topics_and_message_structure.html#zigbee2mqttbridgeconfigure
Wake the device up by pressing buttons and trigger command.

pplucky, 2020-03-19:
> You could try to reconfigure: https://www.zigbee2mqtt.io/information/mqtt_topics_and_message_structure.html#zigbee2mqttbridgeconfigure
> Wake the device up by pressing buttons and trigger command.

Did it and it did not change a thing...

Should I remove it completely and add it again?

janb14, 2020-04-02:
I can confirm that something bricked the battery reporting after upgrading. I upgraded today on FreeBSD and using pm2. Im also using node-red and dumped the battery information into a influx-db to watch batterylevels. I specially needed to fix my parser as after the upgrade the battery property was missing. Ive upgraded from 1.6.0 to 1.12.0 by installing new zigbee2mqtt and copy over configuration and database.

janb14, 2020-04-03:
As @Kryzek already stated the 24h battery reporting still works so this change seems to only limit the battery readouts. So i guess this can be closed as long as @pplucky is ok with that.

pplucky, 2020-04-08:
> As @Kryzek already stated the 24h battery reporting still works (I can confirm this) so this change seems to only limit the battery readouts. So i guess this can be closed as long as @pplucky is ok with that.

I'd like to try totally removing the device from the network and re-adding it, to see if it starts working, because currently it definitely does not, EVER!

Koenkk, 2020-04-09:
It could be that the device needs to reconfigured, to confirm this, please try to reconfigure it manually via https://www.zigbee2mqtt.io/information/mqtt_topics_and_message_structure.html#zigbee2mqttbridgeconfigure (make sure to wakeup the device right before sending this command by clicking a button on it).

pplucky, 2020-04-10:
> It could be that the device needs to reconfigured, to confirm this, please try to reconfigure it manually via [...]

Finally could reconfigure the device successfully and after waiting 24h, the battery is back. 

Thanks everyone for your help and insights.

closed 2020-04-10 by pplucky

Anjerlaan, 2020-04-20:
> It could be that the device needs to reconfigured, to confirm this, please try to reconfigure it manually via [...]

I have 2 IKEA E1524/E1810 remotes, one of them is showing battery level and one is not?
They have equal firmwares:

```
info  2020-04-20 20:25:16: MQTT publish: topic 'zigbee2mqtt/Remote_E1524-E1810', payload '{"linkquality":44,"update_available":false}'
info  2020-04-20 20:25:16: MQTT publish: topic 'zigbee2mqtt/Remote_Tuinlicht', payload '{"update_available":false,"linkquality":34,"battery":74}'
```

If I re-configure this device, do I also have to re-bind it to the group it was bound to?

Morphy99, 2020-12-09:
I have one that went completely flat. Had to repair it. Now it reports 16% battery for over a week with a new battery and I've tried re configuring it two days ago and still the same. I think the only way around it is to remove and repair.

andrew01144, 2022-03-14:
@pplucky - I think I have the same problem with Ikea devices. What payload did you use for zigbee2mqtt/bridge/request/device/configure_reporting ? Many thanks.

pplucky, 2022-03-14:
> @pplucky - I think I have the same problem with Ikea devices. What payload did you use for zigbee2mqtt/bridge/request/device/configure_reporting ? Many thanks.

[https://www.zigbee2mqtt.io/guide/usage/mqtt_topics_and_messages.html#zigbee2mqtt-bridge-request](https://www.zigbee2mqtt.io/guide/usage/mqtt_topics_and_messages.html#zigbee2mqtt-bridge-request)

I probably used {"id": "device_friendly_name"}, but I don't recall anymore, sorry

andrew01144, 2022-03-18:
@pplucky many thanks for your reply. A memory that goes back 2 years would be impressive indeed!
I could not find any form of configure_reporting payload, including yours, that worked.
However, a remove and a re-join of the device seems to have done the trick.

peca89, 2022-04-10:
I can also confirm that now I'm having the same issue. Battery status is not reported on E1524.

MQTT publish: topic 'zigbee2mqtt/0xXXXX', payload '{"action":"arrow_left_click","battery":null,"linkquality":52,"update":{"state":"available"},"update_available":null}'

I tried repairing, reconfiguring and waiting for more than 24 hours and still no battery status.

andrew01144, 2022-04-11:
> I can also confirm that now I'm having the same issue. Battery status is not reported on E1524.
>
> MQTT publish: topic 'zigbee2mqtt/0xXXXX', payload [...]

@peca89 Did you remove before repairing? This is what I did:
remove battery from remote
mosquitto_pub -t zigbee2mqtt/bridge/request/device/remove -m '{"id":"ikea-Remote","force":true}'
mosquitto_pub -t zigbee2mqtt/bridge/config/permit_join -m true
replace battery in remote and press reset 5 times.
wait for remote to appear in the log.
mosquitto_pub -t zigbee2mqtt/bridge/config/rename_last -m ikea-Remote
mosquitto_pub -t zigbee2mqtt/bridge/config/permit_join -m false