Koenkk/zigbee2mqtt #3154: Unhandled promise rejection: Error: AREQ - AF - dataConfirm after 3000ms

runningman84, 2020-03-19:
# Bug Report
## What happened
After a 6 days of operations, zigbee2mqtt stopped working:
```
(node:18) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). To terminate the node process on unhandled promise rejection, use the CLI flag `--unhandled-rejections=strict` (see https://nodejs.org/api/cli.html#cli_unhandled_rejections_mode). (rejection id: 54592)
(node:18) UnhandledPromiseRejectionWarning: Error: AREQ - AF - dataConfirm after 3000ms
    at Timeout._onTimeout (/app/node_modules/zigbee-herdsman/dist/utils/waitress.js:44:24)
    at listOnTimeout (internal/timers.js:549:17)
    at processTimers (internal/timers.js:492:7)
(node:18) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). To terminate the node process on unhandled promise rejection, use the CLI flag `--unhandled-rejections=strict` (see https://nodejs.org/api/cli.html#cli_unhandled_rejections_mode). (rejection id: 54593)
(node:18) UnhandledPromiseRejectionWarning: Error: Timeout - 57964 - 3 - 228 - 0 - 1 after 6000ms
    at Timeout._onTimeout (/app/node_modules/zigbee-herdsman/dist/utils/waitress.js:44:24)
    at listOnTimeout (internal/timers.js:549:17)
    at processTimers (internal/timers.js:492:7)
(node:18) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). To terminate the node process on unhandled promise rejection, use the CLI flag `--unhandled-rejections=strict` (see https://nodejs.org/api/cli.html#cli_unhandled_rejections_mode). (rejection id: 54594)
```

## What did you expect to happen
It would be better to crash in case of such errors. This would allow docker/kubernetes to restart the container or at least indicate an error.
Unfortunately a manual container restart did not help. I had to disconnect the usb cable and use the reset button next to the usb port in order to make it work again.

## How to reproduce it (minimal and precise)
Use zigbee2mqtt and wait a few days.

## Debug Info
zigbee2mqtt version: 1.11.0
CC26X2R1 firmware version: CC26X2R1_20191106.zip

Koenkk, 2020-03-19:
Are these crashes still reproducible with the lates dev firmware? (https://github.com/Koenkk/Z-Stack-firmware/blob/develop/coordinator/Z-Stack_3.x.0/bin/CC26X2R1_20200312.zip)

runningman84, 2020-03-20:
Actually I was using CC26X2R1_20200229.hex before, sorry for the confusion.

Now I have updated it to CC26X2R1_20200312.hex

I will test it and report back. What do you think about the general error handling? Wouldn't it be better to just crash the application instead of writing an error every second?

Koenkk, 2020-03-20:
This would involve some heuristic, e.g. has to fail for X amount of times before crash.

Lets first focus on solving this problem itself, for now make sure your run with debug logging enabled so we get a good capture if it crashes again.

To enable debug logging set in `configuration.yaml`:
```yaml
advanced:
  log_level: debug
```

antonakv, 2020-04-01:
Hi @Koenkk 
I have the same issue, and there is output with log_level debug:

Apr 01 16:56:26 raspberrypi npm[18530]: (node:18545) UnhandledPromiseRejectionWarning: Error: Timeout - 61491 - 8 - 6 - 0 - 1 after 6000ms
Apr 01 16:56:26 raspberrypi npm[18530]:     at Timeout._onTimeout (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/utils/waitress.js:44:24)
Apr 01 16:56:26 raspberrypi npm[18530]:     at listOnTimeout (internal/timers.js:549:17)
Apr 01 16:56:26 raspberrypi npm[18530]:     at processTimers (internal/timers.js:492:7)
Apr 01 16:56:26 raspberrypi npm[18530]: (node:18545) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). To terminate the node process on unhandled promise rejection, use the CLI flag `--unhandled-rejections=strict` (see https://nodejs.org/api/cli.html#cli_unhandled_rejections_mode). (rejection id: 5)
Apr 01 16:56:26 raspberrypi npm[18530]: (node:18545) UnhandledPromiseRejectionWarning: Error: Timeout - 37653 - 1 - 7 - 0 - 1 after 6000ms
Apr 01 16:56:26 raspberrypi npm[18530]:     at Timeout._onTimeout (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/utils/waitress.js:44:24)
Apr 01 16:56:26 raspberrypi npm[18530]:     at listOnTimeout (internal/timers.js:549:17)
Apr 01 16:56:26 raspberrypi npm[18530]:     at processTimers (internal/timers.js:492:7)
Apr 01 16:56:26 raspberrypi npm[18530]: (node:18545) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). To terminate the node process on unhandled promise rejection, use the CLI flag `--unhandled-rejections=strict` (see https://nodejs.org/api/cli.html#cli_unhandled_rejections_mode). (rejection id: 6)
Apr 01 16:56:32 raspberrypi npm[18530]: zigbee2mqtt:debug 2020-04-01 16:56:32: Failed to ping 'fan_switch_bathroom'
Apr 01 16:56:32 raspberrypi npm[18530]: zigbee2mqtt:debug 2020-04-01 16:56:32: Failed to ping 'cc2531_router1'

Koenkk, 2020-04-01:
@antonakv what is your coordinator hardware and firmware? How many devices in your network?

antonakv, 2020-04-01:
@Koenkk 
CC2531 ZigBee USB-Sick 8dbi Antenna from smart-home-komponente.net
Coordinator firmware version: '{"type":"zStack12","meta":{"transportrev":2,"product":0,"majorrel":2,"minorrel":6,"maintrel":3,"revision":20190608}}'

Zigbee2mqtt:
commit 840b9d966c96ff18d2e08e95914cd1473e2611c9 
tag: 1.12.0, origin/master, origin/HEAD)
Date:   Wed Mar 18 19:01:45 2020 +0100

Koenkk, 2020-04-02:
@antonakv In your case I would recommend switching to the source routing firmware: https://github.com/Koenkk/Z-Stack-firmware/tree/master/coordinator/Z-Stack_Home_1.2/bin/source_routing

antonakv, 2020-04-02:
@Koenkk Thanks. Is it recommended for router one stick, right ?

antonakv, 2020-04-02:
I have more errors today
`Apr 02 10:23:02 raspberrypi npm[12804]: zigbee2mqtt:info  2020-04-02 10:23:02: MQTT publish: topic 'zigbee2mqtt/cc2531_router1', payload '{"state":true,"led_state":false,"linkquality":55,"last_seen":"2020-04-02T10:23:02+02:00","description":"14348/0x00158D000446EA94","type":"CHILD_RFD","rssi":87,"device":{"friendlyName":"cc2531_router1","model":"CC2530.ROUTER","ieeeAddr":"0x00124b000be87eaf","networkAddress":61491,"type":"Router","manufacturerID":4447,"manufacturerName":"LUMI","powerSource":"Mains (single phase)","zclVersion":1,"hardwareVersion":1,"dateCode":"2018-09-26"}}'
Apr 02 10:23:06 raspberrypi npm[12804]: (node:12816) UnhandledPromiseRejectionWarning: Error: Timeout - 37653 - 1 - 30 - 0 - 1 after 6000ms
Apr 02 10:23:06 raspberrypi npm[12804]:     at Timeout._onTimeout (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/utils/waitress.js:44:24)
Apr 02 10:23:06 raspberrypi npm[12804]:     at listOnTimeout (internal/timers.js:549:17)
Apr 02 10:23:06 raspberrypi npm[12804]:     at processTimers (internal/timers.js:492:7)
Apr 02 10:23:06 raspberrypi npm[12804]: (node:12816) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). To terminate the node process on unhandled promise rejection, use the CLI flag `--unhandled-rejections=strict` (see https://nodejs.org/api/cli.html#cli_unhandled_rejections_mode). (rejection id: 8)
Apr 02 10:23:10 raspberrypi npm[12804]: zigbee2mqtt:info  2020-04-02 10:23:10: MQTT publish: topic 'zigbee2mqtt/fan_switch_bathroom/availability', payload 'offline'
Apr 02 10:23:10 raspberrypi npm[12804]: zigbee2mqtt:error 2020-04-02 10:23:10: Failed to ping 'fan_switch_bathroom'
Apr 02 10:23:18 raspberrypi npm[12804]: zigbee2mqtt:info  2020-04-02 10:23:18: MQTT publish: topic 'zigbee2mqtt/thermostat_living1', payload '{"eurotronic_host_flags":{"mirror_display":false,"boost":false,"window_open":false,"child_protection":false},"system_mode":"auto","eurotronic_system_mode":1,"linkquality":57,"last_seen":"2020-04-02T10:23:18+02:00","local_temperature":17.5,"occupied_heating_setpoint":20,"unoccupied_heating_setpoint":16,"pi_heating_demand":97,"current_heating_setpoint":20,"eurotronic_error_status":0,"battery":100,"battery_low":0,"device":{"friendlyName":"thermostat_living1","model":"SPZB0001","ieeeAddr":"0x00158d0003c45e7d","networkAddress":60065,"type":"EndDevice","manufacturerID":4151,"manufacturerName":"Eurotronic","powerSource":"Battery","applicationVersion":22,"stackVersion":5,"zclVersion":2,"hardwareVersion":35,"dateCode":"20191014","softwareBuildID":"22190930"}}'
Apr 02 10:24:02 raspberrypi npm[12804]: zigbee2mqtt:info  2020-04-02 10:24:02: MQTT publish: topic 'zigbee2mqtt/thermostat_corridor0', payload '{"occupied_heating_setpoint":16,"unoccupied_heating_setpoint":16,"linkquality":57,"last_seen":"2020-04-02T10:24:02+02:00","current_heating_setpoint":16,"eurotronic_host_flags":{"mirror_display":false,"boost":false,"window_open":false,"child_protection":false},"system_mode":"auto","eurotronic_system_mode":1,"local_temperature":16,"pi_heating_demand":65,"eurotronic_error_status":0,"battery":100,"battery_low":0,"device":{"friendlyName":"thermostat_corridor0","model":"SPZB0001","ieeeAddr":"0x00158d00053d3dc3","networkAddress":28377,"type":"EndDevice","manufacturerID":4151,"manufacturerName":"Eurotronic","powerSource":"Battery","applicationVersion":22,"stackVersion":5,"zclVersion":2,"hardwareVersion":35,"dateCode":"20191014","softwareBuildID":"22190930"}}'
Apr 02 10:24:16 raspberrypi npm[12804]: (node:12816) UnhandledPromiseRejectionWarning: Error: Timeout - 37653 - 1 - 32 - 0 - 1 after 6000ms
Apr 02 10:24:16 raspberrypi npm[12804]:     at Timeout._onTimeout (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/utils/waitress.js:44:24)
Apr 02 10:24:16 raspberrypi npm[12804]:     at listOnTimeout (internal/timers.js:549:17)
Apr 02 10:24:16 raspberrypi npm[12804]:     at processTimers (internal/timers.js:492:7)
Apr 02 10:24:16 raspberrypi npm[12804]: (node:12816) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). To terminate the node process on unhandled promise rejection, use the CLI flag `--unhandled-rejections=strict` (see https://nodejs.org/api/cli.html#cli_unhandled_rejections_mode). (rejection id: 9)
Apr 02 10:24:20 raspberrypi npm[12804]: zigbee2mqtt:info  2020-04-02 10:24:20: MQTT publish: topic 'zigbee2mqtt/thp_sensor_bathroom', payload '{"battery":86,"voltage":2975,"temperature":20.45,"humidity":94.6,"pressure":1012.4,"linkquality":57,"last_seen":"2020-04-02T10:24:20+02:00","device":{"friendlyName":"thp_sensor_bathroom","model":"WSDCGQ11LM","ieeeAddr":"0x00158d000465f8be","networkAddress":31810,"type":"EndDevice","manufacturerID":4151,"manufacturerName":"LUMI","powerSource":"Battery","applicationVersion":3}}'
Apr 02 10:24:20 raspberrypi npm[12804]: zigbee2mqtt:info  2020-04-02 10:24:20: MQTT publish: topic 'zigbee2mqtt/thp_sensor_bathroom', payload '{"battery":86,"voltage":2975,"temperature":20.45,"humidity":95.09,"pressure":1012.4,"linkquality":60,"last_seen":"2020-04-02T10:24:20+02:00","device":{"friendlyName":"thp_sensor_bathroom","model":"WSDCGQ11LM","ieeeAddr":"0x00158d000465f8be","networkAddress":31810,"type":"EndDevice","manufacturerID":4151,"manufacturerName":"LUMI","powerSource":"Battery","applicationVersion":3}}'
`

closed 2020-06-09 by stale[bot]