Koenkk/zigbee2mqtt #3155: Osram Plug 01 OTA sensor causes HA errors

tomaae, 2020-03-19:
## Bug Report
## What happened
Error message:
2020-03-20 00:08:22 WARNING (MainThread) [homeassistant.components.mqtt.binary_sensor] No matching payload found for entity: USB Power_update_available with state topic: zigbee2mqtt/USB Power. Payload: , with value template Template("{{ value_json.update_available}}")

Entity "USB Power_update_available" exists:
[image]

## What did you expect to happen
To get no error message.

## How to reproduce it (minimal and precise)
Happens all the time, but only to this device.
This is only osram plug I have. Other devices with OTA sensor (ikea plugs, other ikea devices and hue devices) do not get this error.

## Debug Info
Device: 
```
{"id":23,"type":"Router","ieeeAddr":"0x0000000000000000","nwkAddr":15879,"manufId":48042,"manufName":"OSRAM","powerSource":"Mains (single phase)","modelId":"Plug 01","epList":[3],"endpoints":{"3":{"profId":49246,"epId":3,"devId":16,"inClusterList":[4096,0,3,4,5,6,2820,64527],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"modelId":"Plug 01","manufacturerName":"OSRAM","powerSource":1,"zclVersion":1,"appVersion":12,"stackVersion":2,"hwVersion":1,"dateCode":"20140331DEOS****","swBuildId":"V1.04.12"}},"genOnOff":{"attributes":{"onOff":1}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x0000000000000000","endpointID":1}]}},"appVersion":12,"stackVersion":2,"hwVersion":1,"dateCode":"20140331DEOS****","swBuildId":"V1.04.12","zclVersion":1,"interviewCompleted":true,"meta":{"configured":1},"lastSeen":1583530362310}
```
zigbee2mqtt version: 1.12.0 (this error is not new, it shows up since OTA sensors were introduced)
CC1352P_2 firmware version:  20191106 Z-Stack_3.x.0 firmware

tomaae, 2020-03-20:
Update:
OTA via z2m went without error for this plug and error did not reappeared after reboot. Not sure if it reappears later tho.

Koenkk, 2020-03-20:
Duplicate of https://github.com/Koenkk/zigbee2mqtt/issues/3146

closed 2020-03-20 by Koenkk