Koenkk/zigbee2mqtt #3156: Failed Interview: Sylvania Outdoor Accent RGB

addisonlynch, 2020-03-20:
# Bug Report
## What happened

Failed interview for Sylvania Outdoor Accent RGB lights. Could not pair unsupported device.

Thank you!

## What did you expect to happen

## How to reproduce it (minimal and precise)

## Debug Info
zigbee2mqtt version:

Latest

CC253X firmware version:

CC2531_DEFAULT_20190608.zip

Debugging info (inc. herdsman log):
https://pastebin.com/RWSpFwwK

Koenkk, 2020-03-20:
It appears that there is some race condition when a device leaves while it is being interviewed. I will look into that (causing the error).

Besides that it seems to pair fine so you can already add support for it.

addisonlynch, 2020-03-20:
Thanks, @Koenkk. 

However it does say that the interview failed...does the interview have to pass for the device to be paired? Here is the database (perhaps if if shows up in the database then it is paired? Not sure how it works)
```
{"id":1,"type":"Coordinator","ieeeAddr":"0x00124b000be96769","nwkAddr":0,"manufId":0,"epList":[13,12,110,11,8,6,5,4,3,2,1],"endpoints":{"1":{"profId":260,"epId":1,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[]},"2":{"profId":257,"epId":2,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[]},"3":{"profId":261,"epId":3,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[]},"4":{"profId":263,"epId":4,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[]},"5":{"profId":264,"epId":5,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[]},"6":{"profId":265,"epId":6,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[]},"8":{"profId":260,"epId":8,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[]},"11":{"profId":260,"epId":11,"devId":1024,"inClusterList":[],"outClusterList":[1280],"clusters":{},"binds":[]},"12":{"profId":49246,"epId":12,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[]},"13":{"profId":260,"epId":13,"devId":5,"inClusterList":[25],"outClusterList":[],"clusters":{},"binds":[]},"110":{"profId":260,"epId":110,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[]}},"interviewCompleted":false,"meta":{},"lastSeen":null}
{"id":2,"type":"Router","ieeeAddr":"0xf0d1b8000005be24","nwkAddr":33561,"manufId":4489,"manufName":"LEDVANCE","modelId":"Outdoor Accent RGB","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":258,"inClusterList":[0,3,4,5,6,8,768,2821,64513],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"modelId":"Outdoor Accent RGB","manufacturerName":"LEDVANCE"}}},"binds":[]}},"interviewCompleted":false,"meta":{},"lastSeen":1584663729041}
```

Koenkk, 2020-03-20:
If you see the `"modelId":"Outdoor Accent RGB"` you can add support for it (it is in database so this can be done).

jrc1883, 2020-03-27:
I am getting the same errors, looking into it now.
`Mar 26 21:48:06 openhab npm[11173]: zigbee2mqtt:debug 2020-03-26 21:48:06: Received Zigbee message from '0xf0d1b8000005b92a', type 'readResponse', cluster 'genBasic', data '{"modelId":"Outdoor Accent RGB","manufacturerName":"LEDVANCE"}' from endpoint 1 with groupID 0
`

`Mar 26 21:48:06 openhab npm[11173]: zigbee2mqtt:debug 2020-03-26 21:48:06: No converter available for '75541' with cluster 'genBasic' and type 'readResponse' and data '{"modelId":"Outdoor Accent RGB","manufacturerName":"LEDVANCE"}'`

closed 2020-06-06 by stale[bot]