Koenkk/zigbee2mqtt #3158: Received message from unsupported device with Zigbee model 'undefined'

sjorge, 2020-03-20:
Some devices will always trigger this warning while pairing, e.g. lumi.* devices and also the Schneider Electric iTRV... perhaps we can defer the warning until after we try to lookup the missing data?

Seems like 1 second later we get the device announcement and ~10sec later we have the data we need from a attReport from genBasic?

This is on current dev branch, but also present in the last 3 (atleast) releases

```
zigbee2mqtt:warn  2020-03-20 12:02:33: Received message from unsupported device with Zigbee model 'undefined'
zigbee2mqtt:warn  2020-03-20 12:02:33: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
zigbee2mqtt:debug 2020-03-20 12:02:34: Device '0x00158d00044faf3e' announced itself
zigbee2mqtt:info  2020-03-20 12:02:34: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_announced","message":"announce","meta":{"friendly_name":"0x00158d00044faf3e"}}'
zigbee2mqtt:debug 2020-03-20 12:02:44: Received Zigbee message from '0x00158d00044faf3e', type 'attributeReport', cluster 'genBasic', data '{"65281":{"1":3055,"3":30,"4":424,"5":42,"6":[0,0],"10":27321,"100":0},"modelId":"lumi.sensor_magnet.aq2"}' from endpoint 1 with groupID 0
zigbee2mqtt:info  2020-03-20 12:02:45: MQTT publish: topic 'zigbee2mqtt/0x00158d00044faf3e', payload '{"battery":100,"voltage":3055,"contact":true,"linkquality":47}'
zigbee2mqtt:debug 2020-03-20 12:02:49: Received Zigbee message from '0x00158d00044faf3e', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 1 with groupID 0
zigbee2mqtt:debug 2020-03-20 12:02:50: Received Zigbee message from '0x00158d00044faf3e', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 1 with groupID 0
zigbee2mqtt:info  2020-03-20 12:02:51: MQTT publish: topic 'zigbee2mqtt/0x00158d00044faf3e', payload '{"battery":100,"voltage":3055,"contact":true,"linkquality":47}'
zigbee2mqtt:debug 2020-03-20 12:02:59: Received Zigbee message from '0x00158d00044faf3e', type 'readResponse', cluster 'genBasic', data '{"modelId":"lumi.sensor_magnet.aq2","manufacturerName":"LUMI"}' from endpoint 1 with groupID 0
zigbee2mqtt:info  2020-03-20 12:03:08: Successfully interviewed '0x00158d00044faf3e', device has successfully been paired
zigbee2mqtt:info  2020-03-20 12:03:08: Device '0x00158d00044faf3e' is supported, identified as: Xiaomi Aqara door & window contact sensor (MCCGQ11LM)
zigbee2mqtt:info  2020-03-20 12:03:08: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_successful","meta":{"friendly_name":"0x00158d00044faf3e","model":"MCCGQ11LM","vendor":"Xiaomi","description":"Aqara door & window contact sensor","supported":true}}'
```

Koenkk, 2020-03-20:
Should be fixed in latest dev branch, could you please confirm?

sjorge, 2020-03-20:
Will try to test tomorrow, will update with results.

sjorge, 2020-03-21:
```
zigbee2mqtt:info  2020-03-21 20:27:34: Device '0x00158d00044faf3e' joined
zigbee2mqtt:info  2020-03-21 20:27:34: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_connected","message":{"friendly_name":"0x00158d00044faf3e"}}'
zigbee2mqtt:info  2020-03-21 20:27:34: Starting interview of '0x00158d00044faf3e'
zigbee2mqtt:info  2020-03-21 20:27:34: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_started","meta":{"friendly_name":"0x00158d00044faf3e"}}'
zigbee2mqtt:info  2020-03-21 20:27:34: MQTT publish: topic 'zigbee2mqtt/0x00158d00044faf3e/availability', payload 'online'
zigbee2mqtt:debug 2020-03-21 20:27:35: Device '0x00158d00044faf3e' announced itself
zigbee2mqtt:info  2020-03-21 20:27:35: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_announced","message":"announce","meta":{"friendly_name":"0x00158d00044faf3e"}}'
zigbee2mqtt:debug 2020-03-21 20:27:35: Received Zigbee message from '0x00158d00044faf3e', type 'attributeReport', cluster 'genBasic', data '{"modelId":"lumi.sensor_magnet.aq2","appVersion":3}' from endpoint 1 with groupID 0
zigbee2mqtt:debug 2020-03-21 20:27:35: Received Zigbee message from '0x00158d00044faf3e', type 'attributeReport', cluster 'genBasic', data '{"65281":{"1":3055,"3":28,"4":424,"5":47,"6":[0,1],"10":27321,"100":0}}' from endpoint 1 with groupID 0
zigbee2mqtt:info  2020-03-21 20:27:36: MQTT publish: topic 'zigbee2mqtt/0x00158d00044faf3e', payload '{"battery":100,"voltage":3055,"contact":true,"linkquality":42}'
zigbee2mqtt:debug 2020-03-21 20:27:42: Received Zigbee message from '0x00158d00044faf3e', type 'attributeReport', cluster 'genBasic', data '{"65281":{"1":3055,"3":28,"4":424,"5":47,"6":[0,0],"10":27321,"100":0},"modelId":"lumi.sensor_magnet.aq2"}' from endpoint 1 with groupID 0
zigbee2mqtt:info  2020-03-21 20:27:43: MQTT publish: topic 'zigbee2mqtt/0x00158d00044faf3e', payload '{"battery":100,"voltage":3055,"contact":true,"linkquality":42}'
zigbee2mqtt:info  2020-03-21 20:27:44: Successfully interviewed '0x00158d00044faf3e', device has successfully been paired
zigbee2mqtt:info  2020-03-21 20:27:44: Device '0x00158d00044faf3e' is supported, identified as: Xiaomi Aqara door & window contact sensor (MCCGQ11LM)
zigbee2mqtt:info  2020-03-21 20:27:44: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_successful","meta":{"friendly_name":"0x00158d00044faf3e","model":"MCCGQ11LM","vendor":"Xiaomi","description":"Aqara door & window contact sensor","supported":true}}'
```

Looking good now!

closed 2020-03-21 by sjorge

caffeineflo, 2020-05-31:
I'm seeing the same issue, has this fix made it's way into master or should I still switch to dev?

HydrelioxGitHub, 2020-07-08:
I have the same issue here.